hoylabs/OpenDTU-OnBattery #1599: [Bug] DPL does not stop full solar-passthrough (SoC threshold)

schlimmchen, 2025-02-02:
### Discussed in https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1567

<div type='discussions-op-text'>

<sup>Originally posted by **Manos1966** January 21, 2025</sup>
Eine Frage

hat jemand Probleme bemerkt wenn:
- nach einem sonnigen Tag (Hurrah!) der die Batterien voll geladen hat
- dadurch das Full Solar Passthrough aktiviert wurde
- Nachts, die Batterie ist voll,
-  **aber die Wechselrichter bleiben auf STANDBY**

Mein Setup:
- Chip-Modell  ESP32-D0WD-V3 4MB 😢 
- generic_esp32_4mb_no_ota
- Firmwareversion / git Hash  2024.11.20 (letzte Version noch nicht installiert)
- 1 x HM1500, 1 x HM800 (beide von DPL gesteuert)
- 2 x VE.Direct (1 x MPPT250/60, 1x MPPT150/45) zusammen produzieren sie mehr als 3000W, also mehr als die Full Solar Passthrough Leistung von 1500+800 = 2300W die die Wechselrichter erreichen koennen
- "Batterie nachts sogar teilweise geladen nutzen" ist eingeschaltet

Der Tag war sonnig, die Batterien wurde schnell geladen und danach "Full Solar Passthrough" wurde aktiviert.

Jetzt ist Nacht die Batterien sind bei 85%, aber beide Wechselrichter sind auf standby.

Ich habe bemerkt, der HM1500 hat viele Kommunikationsprobleme (die OpenDTU ist 15cm entfernt und ich habe vergessen die NRF24+ Leistung auf Minimum zu drosseln). Daran aber koennte es nicht liegen (hoffe ich).

Console Log: https://pastebin.com/mtj0x4bw
 
[image]
[image]
[image]
[image]

</div>

schlimmchen, 2025-02-02:
Das Log sagt ganz klar, dass der DPL im full solar-passthrough ist, obwohl es nach 21 Uhr ist. Scheinbar wurde die Batterie auf 85% SoC geladen, dann war full solar-passthrough an, und zwar gut ausbalanciert, d.h. die Batterie wurde nicht (so stark) entladen, dass die auf 84% gesunken wäre. Die Inverter passen ihre Leistung dann der solaren Leistung an. Es wird dunkel, die WR gehen aus, und die Batterie bleibt bei 85%.

Weil Full Solar-Passthrough so anders funktioniert, als der reguläre DPL Betrieb, auch (normal) Solar-Passthrough, habe ich das ausgelagert beim Implementieren des Supports für mehrere Inverter. Es fehlt eine geeignete Abbruchbedingung...

Manos1966, 2025-02-02:
Danke Bernhard

AndreasBoehm, 2025-02-10:
Das Problem ist in neueren Versionen bereits gelöst.

In der von dir verwendeten Version 2024.11.20 war es nötig das sich der SoC bzw die Batteriespannung ändert um Full Solar Passthrough abzubrechen. (siehe https://github.com/hoylabs/OpenDTU-OnBattery/blob/2024.11.20/src/PowerLimiter.cpp#L184)

In development ist zu sehen das hier bereits geprüft wird ob neben dem SoC bzw Batteriespannungsthreshold überhaupt genug Solarleistung verfügbar ist für Full Solar Passthrough. (siehe https://github.com/hoylabs/OpenDTU-OnBattery/blob/development/src/PowerLimiter.cpp#L672)

Das Problem das bleibt ist das `_fullSolarPassThroughEnabled` weiter auf `true` steht bis die Batterie weit genug entladen wurde, das ist aber kein funktionales Problem, es wird nur über MQTT veröffentlicht das Full Solar Passthrough aktiv ist obwohl er es nicht mehr ist.

schlimmchen, 2025-02-10:
Der Mann hat recht. Danke, Andreas.

closed 2025-02-10 by schlimmchen