hoylabs/OpenDTU-OnBattery #1603: [Request] DPL without "Charge Cycles"

boennemann, 2025-02-03:
### Is your feature request related to a problem? Please describe.

My problem is that I want my battery output to match my house's consumption without being bound to "charge cycles".

Right now the DPL is hardcoded to believe that power can only come from the sun – however with an AC Charger and dynamic energy and grid prices it becomes relevant to also charge the battery from the grid.

E.g. Sun charges the battery from 10AM-4PM.
4PM discharge cycles begins.
Energy is expensive from 6-8PM.
Battery is still discharged after 8PM when energy is cheap again. Meanwhile tomorrow morning is super expensive, while all my energy was discharged at night when it was cheap anyway.

Additionally, energy is cheapest at 4AM so I charge my battery from grid.
I want to discharge at 7AM-9AM when it's expensive.
I want to stop discharging and get more sun in. I want to discharge again at 6PM.

### Describe the solution you'd like

I'd like a DPL mode that's "manual" in the sense that I can say: "Charge-only" or "Discharge allowed" via MQTT/HA.

I'd automate the rest on Home Assistant side.

### Describe alternatives you've considered

I'm currently playing around with manually setting the discharge treshold to what I need it to be (higher or lower than current SoC), but that's not really predictable.

schlimmchen, 2025-02-03:
The DPL knows the concept of battery allowance. So you want to be able to force the battery allowance to 0 W.

I need to check whether there is actually a battery allowance, of it it's a power-bus allowance, where the solar charger's output contribute to the power bus allowance (if solar-passthrough is on).

Would that be sufficient?

boennemann, 2025-02-04:
@schlimmchen I just realized that there is also the "Maximum Total Output" settings, but I can't set that via MQTT.

I'd then just need to enable solar passthrough and set this value to either 0 or my actual inverters limit and I'd be good already, right?

schlimmchen, 2025-02-04:
> POST api/battery/config totally works 👌

Until we break it. The web API is explicitly declared to be **not** stable.

You already can set the max total output, see `powerlimiter/cmd/upper_power_limit` in the [docs](https://opendtu-onbattery.net/firmware/mqtt_topics/#power-limiter-topics) (sorry, I see the description is outdated... this topic does control the "maximum total output" setting since support for multiple inverters arrived.)

https://github.com/hoylabs/OpenDTU-OnBattery/blob/ab95aa39323eb481df0a6d2d0a08a9b3e90d43b9/src/MqttHandlePowerLimiter.cpp#L197-L201

> and I'd be good already, right?

Can't tell you, sorry. Please figure that out for yourself. Skimming from your original post, yes that should work for you.

gitisgreat2023, 2025-02-05:
@boennemann 
It's already available via MQTT
Topic: OpenDTU-onBattery/powerlimiter/cmd/mode

`//0 - Normal operation: The Power Limiter works as configured through the web application.
//1 - Fully disable with inverter shut down: The inverter is shut down and afterwards the Power Limiter stops operating, as if it was disabled in the web application.Note that this means that the inverter can start producing power if the web application or an MQTT topic is used to control it.
//2 - Unconditional Full Solar - Passthrough: The power limit is set such that all available solar power is fed into the home, irrespective of grid consumption.Essentially, the inverter mimics the behavior of a traditional, non - smart inverter.Depending on your configuration, the inverter's limit is set to the following:
//Inverter is powered by a battery: The inverter's limit is set to the solar power output (VE.Direct interface), adjusted for efficiency, such that no energy from the battery is consumed. Note that if VE.Direct is disabled or the data is outdated, the inverter is shut down instead. This mode can be particularly useful in scenarios where solar power is better stored elsewhere, such as in an electric car.
//Inverter is powered by solar modules: The inverter's limit is set to the upper limit configured in the DPL settings (starting from release 2024.05.03).
`

closed 2025-03-06 by github-actions[bot]