hoylabs/OpenDTU-OnBattery #1609: [Bug] Huawei R4850 startet das Laden nicht

schlimmchen, 2025-02-05:
### Discussed in https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1607

<div type='discussions-op-text'>

<sup>Originally posted by **tstrempel1980** February  5, 2025</sup>
Hallo zusammen,
ich nutze das CAN Joly Board, Pin mapping habe ich angepasst. Jumper gesetzt.
Der Laderegler soll den Marstek B2500 laden.
Die Entladung abends erfolgt über den HM-600.
OpenDTU on Battery Version: 2025.01.23

warum wird der Laderegler nicht aktiviert?
Ich habe über einen längeren Zeitraum -300 Watt.
Und was bedeutet "disable PSU"

[DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 5 W)
11:23:22.616 > [DPL inverter 114184518511]:
11:23:22.682 >     solar-powered, producing 43 W
11:23:22.743 >     lower/current/upper limit: 50/49/600 W, output capability: 600 W
11:23:22.810 >     sending commands enabled, reachable, eligible
11:23:22.869 >     max reduction production/standby: 0/0 W, max increase: 0 W
11:23:22.936 >     target limit/output/state: -1 W (unchanged)/43 W/unchanged, 0 update timeouts
11:23:22.999 >     MPPTs AC power: a: 21 W b: 23 W
11:23:23.058 > Fetch inverter: 114184518511
11:23:23.123 > Queue size - NRF: 1 CMT: 0
11:23:23.186 > Last missing
11:23:23.248 > Request retransmit: 3
11:23:23.313 > Middle missing
11:23:23.479 > Request retransmit: 1
11:23:23.533 > [Huawei::Controller] Inverter is active, disable PSU
11:23:23.685 > [Huawei::HwIfc] [60.452] InputTemperature: 24.00°C
11:23:23.735 > [Huawei::HwIfc] [60.451] OutputTemperature: 23.00°C
11:23:23.788 > [Huawei::HwIfc] [60.450] InputVoltage: 231.34V
11:23:23.845 > [Huawei::HwIfc] [60.448] OutputCurrentMax: 0.00A
11:23:23.894 > [Huawei::HwIfc] [60.442] InputFrequency: 50.03Hz
11:23:23.964 > [Huawei::HwIfc] [60.453] OutputCurrent: 0.00A
11:23:24.025 > [Huawei::HwIfc] [60.441] InputPower: 0.00W
11:23:24.083 > [Huawei::HwIfc] [60.443] InputCurrent: 0.00A
11:23:24.146 > [Huawei::HwIfc] [60.445] OutputPower: 0.00W
11:23:24.217 > [Huawei::HwIfc] [60.446] Efficiency: 0.00
11:23:24.272 > [Huawei::HwIfc] [60.447] OutputVoltage: 1.77V
11:23:24.335 > Middle missing
11:23:24.399 > Request retransmit: 1
11:23:24.460 > Middle missing
11:23:24.540 > Request retransmit: 1
11:23:24.603 > Fetch inverter: 114184518511
11:23:24.665 > Queue size - NRF: 1 CMT: 0
11:23:24.744 > [DPL] targeting 0 W, base load is 200 W, power meter reads -167.9 W (valid)
11:23:24.881 > [DPL] inverter 114184518511 is behind power meter producing 43 W
11:23:24.948 > [DPL] requesting 0 W from 1 solar-powered inverter currently producing 43 W (diff -43 W, hysteresis 5 W)
11:23:24.999 > [DPL] will cover 43 W using 1 solar-powered inverter
11:23:25.064 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
11:23:25.126 > [DPL] requesting 0 W from 0 battery-powered inverters currently producing 0 W (diff 0 W, hysteresis 5 W)
11:23:25.188 > [DPL inverter 114184518511]:
11:23:25.254 >     solar-powered, producing 43 W
11:23:25.313 >     lower/current/upper limit: 50/49/600 W, output capability: 600 W
11:23:25.377 >     sending commands enabled, reachable, eligible
11:23:25.442 >     max reduction production/standby: 0/0 W, max increase: 0 W
11:23:25.505 >     target limit/output/state: -1 W (unchanged)/43 W/unchanged, 0 update timeouts
11:23:25.565 >     MPPTs AC power: a: 21 W b: 23 W
11:23:25.732 > Success

Vielen Dank für Eure Hilfe :)</div>

schlimmchen, 2025-02-05:
@tstrempel1980 Kannst du bitte die Firmware aus dem [PR Build Run](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/13166437902?pr=1610) testen? Ich bin sehr zuversichtlich, dass das dein Problem behebt.

tstrempel1980, 2025-02-08:
Hi, hier mein Feedback,
sieht soweit gut aus :)

[Huawei::HwIfc] [13590.616] InputTemperature: 17.00°C
11:04:43.585 > [Huawei::HwIfc] [13590.615] OutputTemperature: 21.00°C
11:04:43.685 > [Huawei::HwIfc] [13590.614] InputVoltage: 230.34V
11:04:43.743 > [Huawei::HwIfc] [13590.613] OutputCurrentMax: 3.25A
11:04:43.897 > [Huawei::HwIfc] [13590.607] InputFrequency: 50.00Hz
11:04:43.992 > [Huawei::HwIfc] [13590.618] OutputCurrent: 3.01A
11:04:44.092 > [Huawei::HwIfc] [13590.606] InputPower: 97.40W
11:04:44.189 > [Huawei::HwIfc] [13590.608] InputCurrent: 0.42A
11:04:44.295 > [Huawei::HwIfc] [13590.609] OutputPower: 85.70W
11:04:44.484 > [Huawei::HwIfc] [13590.611] Efficiency: 0.88
11:04:44.665 > [Huawei::HwIfc] [13590.612] OutputVoltage: 28.26V
11:04:44.777 > Websocket: [/livedata][1] disconnect
11:04:44.959 > Websocket: [/huaweilivedata][1] disconnect
11:04:45.381 > All missing
11:04:45.526 > Nothing received, resend whole request
11:04:45.634 > Fetch inverter: 114184518511
11:04:45.743 > Request SystemConfigPara
11:04:45.838 > Queue size - NRF: 3 CMT: 0
11:04:45.929 > [DPL] targeting 0 W, base load is 200 W, power meter reads 10.6 W (valid)
11:04:46.047 > [DPL] inverter 114184518511 is behind power meter producing 0 W
11:04:46.151 > [DPL] skipping 114184518511 as it is not reachable
11:04:46.266 > [DPL] DC power bus usage blocked by HuaweiCan auto power
11:04:46.340 > [DPL inverter 114184518511]:
11:04:46.444 >     solar-powered, standing by at 0 W
11:04:46.558 >     lower/current/upper limit: 50/0/600 W, output capability: 0 W
11:04:46.668 >     sending commands enabled, offline, disqualified (unreachable)
11:04:46.775 >     max reduction production/standby: 0/0 W, max increase: 0 W
11:04:47.170 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
11:04:47.286 >     MPPTs AC power: a: 0 W b: 0 W
11:04:47.368 > All missing
11:04:47.478 > Nothing received, resend count exeeded
11:04:47.585 > Fetch inverter: 114184518511
11:04:47.683 > Request SystemConfigPara
11:04:47.788 > Queue size - NRF: 3 CMT: 0
11:04:47.989 > All missing
11:04:48.102 > Nothing received, resend whole request
11:04:48.190 > All missing
11:04:48.337 > Nothing received, resend whole request
11:04:48.507 > [PowerMeterHttpJson] New total: -14.54
11:04:48.594 > All missing
11:04:48.705 > Nothing received, resend whole request
11:04:48.814 > [DPL] targeting 0 W, base load is 200 W, power meter reads -14.5 W (valid)
11:04:48.910 > [DPL] inverter 114184518511 is behind power meter producing 0 W
11:04:49.026 > [DPL] skipping 114184518511 as it is not reachable
11:04:49.133 > [DPL] DC power bus usage blocked by HuaweiCan auto power
11:04:49.221 > [DPL inverter 114184518511]:
11:04:49.327 >     solar-powered, standing by at 0 W
11:04:49.425 >     lower/current/upper limit: 50/0/600 W, output capability: 0 W
11:04:49.529 >     sending commands enabled, offline, disqualified (unreachable)
11:04:49.634 >     max reduction production/standby: 0/0 W, max increase: 0 W
11:04:49.761 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
11:04:49.953 >     MPPTs AC power: a: 0 W b: 0 W
11:04:50.102 > All missing
11:04:50.247 > Nothing received, resend whole request
11:04:50.355 > Fetch inverter: 114184518511
11:04:50.556 > Request SystemConfigPara
11:04:50.656 > Queue size - NRF: 3 CMT: 0
11:04:50.766 > All missing
11:04:50.931 > Nothing received, resend count exeeded
11:04:51.009 > [Huawei::Controller] newPowerLimit: 101, output_power: 85.7
11:04:51.167 > [Huawei::Controller] Setting output current to 3.13A. This is the lower value of calculated 3.13A and BMS permissable 340282346638528859811704183484516925440.00A currents
11:04:51.270 > [Huawei::HwIfc] [13593.111] InputTemperature: 17.00°C
11:04:51.374 > [Huawei::HwIfc] [13593.110] OutputTemperature: 21.00°C
11:04:51.460 > [Huawei::HwIfc] [13593.109] InputVoltage: 229.94V
11:04:51.565 > [Huawei::HwIfc] [13593.108] OutputCurrentMax: 3.25A
11:04:51.695 > [Huawei::HwIfc] [13593.102] InputFrequency: 50.00Hz
11:04:51.771 > [Huawei::HwIfc] [13593.113] OutputCurrent: 0.00A
[... 8 lines cut ...]
11:04:53.069 > [DPL] targeting 0 W, base load is 200 W, power meter reads -14.5 W (valid)
11:04:53.318 > [DPL] inverter 114184518511 is behind power meter producing 0 W
11:04:53.449 > [DPL] skipping 114184518511 as it is not reachable
11:04:53.535 > [DPL] DC power bus usage blocked by HuaweiCan auto power
11:04:53.621 > [DPL inverter 114184518511]:
11:04:53.727 >     solar-powered, standing by at 0 W
11:04:53.822 >     lower/current/upper limit: 50/0/600 W, output capability: 0 W
11:04:53.885 >     sending commands enabled, offline, disqualified (unreachable)
11:04:54.071 >     max reduction production/standby: 0/0 W, max increase: 0 W
11:04:54.251 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
11:04:54.330 >     MPPTs AC power: a: 0 W b: 0 W
11:04:54.439 > Fetch inverter: 114184518511
11:04:54.547 > Request SystemConfigPara
11:04:54.615 > Queue size - NRF: 3 CMT: 0
11:04:54.699 > All missing
11:04:54.825 > Nothing received, resend whole request
11:04:55.301 > [PowerMeterHttpJson] New total: -120.68
11:04:55.481 > All missing
11:04:55.560 > Nothing received, resend whole request
11:04:55.679 > [DPL] targeting 0 W, base load is 200 W, power meter reads -120.7 W (valid)
11:04:55.770 > [DPL] inverter 114184518511 is behind power meter producing 0 W
11:04:55.889 > [DPL] skipping 114184518511 as it is not reachable
11:04:55.970 > [DPL] DC power bus usage blocked by HuaweiCan auto power
11:04:56.077 > [DPL inverter 114184518511]:
11:04:56.174 >     solar-powered, standing by at 0 W

tstrempel1980, 2025-02-08:
Ich habe versucht das SN56HVD230 (VP230) direkt an meinen ESP32 S3 anzuschließen.
Hier kommt aber keine Kommunikation zustande?
Habe es auch in der Konfig  GPIO 7 und 17 angesprochen und das Setting geladen und auch die Einstellung unter AC Ladegerät aktiviert. Oder funktioniert das nur mit dem Fusion Shield?

LG Thomas

schlimmchen, 2025-02-08:
Das funktioniert auch mit einem sonstigen VP230, das Fusion ist transparent für den ESP32.

Mach mal eine neue Diskussion auf, mit Pinmapping, Fotos, Screenshots, etc.

Hast du einen 120Ohm Abschlusswiderstand an deinem VP230? Ist der auf einer Breakoutplatine, nehme ich an?

Danke für die Rückmeldung zum Fix. Andreas und ich sind da noch nicht fertig, siehe pull request, aber das Problem ist verstanden und wir werden den Fix im nächsten Release drin haben.

PR #1610 (2025-02-08): Fix: R48xx: only interlock against battery-powered inverters
Closes #1609.
closed 2025-02-08 by AndreasBoehm