hoylabs/OpenDTU-OnBattery #1626: DPL für einzelne Inverter über MQTT ein- und ausschalten

kutzera, 2025-02-11:
### Is your feature request related to a problem? Please describe.

Derzeit kann die Nulleispeisung der einzelnen Inverter nur über das webinterface geändert werden. 
Ich habe 5 Inverter. 4 speisen aus der Sonne direkt ins Netz ein. Einer hängt hinter einem Speicher.  Die Batterie lädt über Solar aber auch über AC bei Überschuss. Wenn Jetzt aber morgens alle Inverter bereits runterregeln, erhält der Speicher nie den Überschusswert, um über AC zu laden, obwohl genug Überschuss theoretisch vorhanden ist. Also muss ich morgen immer für 4 Inverter DPL ausschalten, bis die Akkus geladen sind und dann manuell über das webinterface wieder zu DPL hinzufügen. 

### Describe the solution you'd like

Auswählen der Inverter für DPL über MQTT

schlimmchen, 2025-02-11:
Deinen Anwendungsfall habe ich noch nicht ganz durchstiegen auf die Schnelle, aber es klingt mir ganz schwer danach als würde der Titel bzw. dein Lösungswunsch nicht zum Problem passen.

Hast du einmal nachgedacht, was stattdessen in der AC charger Logik geändert werden müsste, damit dein Anwendungsfall klappt? Eine ernste Frage ist das, ich hab noch nicht nachgedacht...

Macht es Sinn, dass ich die Lösung eher im AC Charger suche?

spcqike, 2025-02-11:
> Macht es Sinn, dass ich die Lösung eher im AC Charger suche?

eine weitere Möglichkeit wäre ein "DPL mode 3". unconditional, so dass die solar Inverter auf Maximal gehen, die Batterieinverter aber aus bleiben. So könnte der Charger überschüsse aufnehmen.

schlimmchen, 2025-02-11:
@spcqike meint Mode 2, siehe [Doku](https://opendtu-onbattery.net/firmware/mqtt_topics/#mode). Und ich denke er hat recht: Das ist doch genau das richtige für dich. Sogar die Erklärung passt, man muss nur "eletric vehicle" mit "local battery" ersetzen. Passt das aus deiner Sicht?

Ist deine Batterie dann eine "klassische"? Oder ist das eine smart-buffer Batterie, wir jüngst implementiert? In letzterem Fall haben wir nämlich erst gestern entschieden, dass Inverter an solchen smart-buffer Batterien auch voll aufgedreht werden.

kutzera, 2025-02-11:
> [@spcqike](https://github.com/spcqike) meint Mode 2, siehe [Doku](https://opendtu-onbattery.net/firmware/mqtt_topics/#mode). Und ich denke er hat recht: Das ist doch genau das richtige für dich. [...]

ach mensch mein Problem hat sich im Grunde genommen erledigt. @schlimmchen Du hast vollkommen recht. der Mode 2 in verbindung mit Inverter ohne solar macht genau das, was ich brauche.
Vielen Dank euch

closed 2025-02-11 by kutzera