hoylabs/OpenDTU-OnBattery #1629: Problem mit dem senden von MQTT commands

PhilJaro, 2025-02-11:
### What happened?

Wenn ein MQTT-Befehl gesendet wird, friert das System ein (keine neue abfrage von Wechselrichter/Batterie/Solarcharger) und es erfolgt keine Ausgabe mehr in der Web-Konsole – über USB wird lediglich die Powermeter-Ausgabe angezeigt. Die Web-Oberfläche funktioniert zwar weiterhin normal, jedoch ist das Speichern nicht mehr möglich. Dies führt dazu, dass das System vollständig abstürzt und neu startet. Bei  aktiviertem „retain“-Flag läuft das System nach dem Neustart dann wieder in den gleichen Fehler rein und wird somit gar nicht mehr stabil.

### To Reproduce Bug

Ab Version 2024.11.20 (und allen nachfolgenden Versionen) kann der Fehler durch das Senden eines MQTT-Kommandos ausgelöst werden. Auffallend ist, dass der Fehler nicht immer konsistent auftritt – manchmal setzt er schneller ein, manchmal verzögert.

Da ich automatisierte Befehle etwa einmal pro Minute an das System sende – hauptsächlich zur Steuerung des Betriebsmodus – gelingt es ab dieser Version nicht mehr, das System länger als fünf Minuten aktiv zu halten. Insbesondere MQTT-Befehle mit aktiviertem „retain“-Flag führen, wie bereits erwähnt, zu einem vollständigen Systemversagen, bei dem auch ein Neustart des Systems keine Abhilfe schafft.

Der einzige bisher zuverlässige Lösungsansatz besteht in einem Downgrade auf Version 2024.10.22, bei der sämtliche Funktionen einwandfrei arbeiten.

### Expected Behavior

MQTT Befehle sollten das System nicht zum abstürzen bringen denke ich ;) 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2024.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
[PowerMeterHttpJson] New total: 1.77
20:47:39.785 > Fetch inverter: 116491132886
20:47:39.787 > Queue size - NRF: 0 CMT: 1
20:47:40.186 > Success
20:47:40.263 > Websocket: [/livedata][1] disconnect
20:47:40.265 > Websocket: [/solarchargerlivedata][1] disconnect
20:47:40.268 > Websocket: [/batterylivedata][1] disconnect
20:47:40.416 > [PowerMeterHttpJson] New total: 2.56
20:47:40.418 > [DPL] the system is stable, the last power limit is still valid
20:47:40.643 > Fetch inverter: 116491132886
20:47:40.644 > Queue size - NRF: 0 CMT: 1
20:47:41.219 > Success
20:47:41.406 > [PowerMeterHttpJson] New total: 1.01
20:47:41.644 > Fetch inverter: 116491132886
20:47:41.647 > Queue size - NRF: 0 CMT: 1
20:47:42.191 > Success
20:47:42.414 > [PowerMeterHttpJson] New total: 1.60
20:47:42.647 > Fetch inverter: 116491132886
20:47:42.649 > Queue size - NRF: 0 CMT: 1
20:47:43.193 > Success
20:47:43.489 > [PowerMeterHttpJson] New total: -1.50
20:47:43.658 > Fetch inverter: 116491132886
20:47:43.660 > Queue size - NRF: 0 CMT: 1
20:47:43.751 > Received MQTT message on topic: solar/powerlimiter/cmd/threshold/soc/stop

Ab hier ist schluss...
```

### Anything else?

Verwendet wird ein ESP32-S3 (open-DTU-Fusion) mit POE. 

Verwendete MQTT Commands: 

powerlimiter/cmd/mode 
powerlimiter/cmd/threshold/soc/stop
powerlimiter/cmd/threshold/soc/start
powerlimiter/cmd/threshold/soc/full_solar_passthrough

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-02-11:
Jaaa, verdammt, dann hab ich es wohl verschlimmbessert: b1edb13b3c32c0e8f0e0196eed14de1c0ee8c923

Du bist nicht der erste, der diese Auffälligkeit beobachtet und davon berichtet. Dann muss ich da nochmal ganz genau hinschauen.

Und du musst den ESP32 dann stromlos machen? Der Watchdog hilft auch nicht (Gerät startet nach kurzer Zeit von selbst neu)? Das wäre ja ein Deadlock während die Tasks aber munter weiterlaufen... Merkwürdig...

PhilJaro, 2025-02-12:
Stromlos ist nicht unbedingt notwendig, es genügt, wenn ich irgendetwas Speichern will in den Einstellungen, dann startet das System neu und geht wieder (vorausgesetzt die commands waren nicht mit dem retain-Flag versehen).

Wenn ich hier nicht manuell einschreite, passiert lange zeit erstmal nichts, irgendwann startet dann das System neu, habe ich aber nicht gemessen, wie lange das gedauert hat aber sicherlich 30-60min.

Ja genau wenn ich mich per USB dran hänge sehe ich die HTTPS+JSON Meter werte im eingestellten Rhythmus (1sec) munter weiterlaufen, aber nichts anderes.

ali-baba-de, 2025-02-15:
Bei mir gab es ein ähnliches Problem, siehe #1494 . Vor ca. drei Wochen hab ich dann ein 3s-Delay zwischen den MQTT-Kommandos eingebaut und seit dem ist das Problem nicht mehr aufgetreten. Firmware ist 2025.01.23, die neueste hab ich noch nicht getestet.

PhilJaro, 2025-09-17:
Ich hatte das Problem "behoben", indem ich wie oben von @ali-baba-de beschrieben ein 3s-Delay zwischen den MQTT-Kommandos eingebaut habe. 

Zum Testen wurde nun die neueste Version installiert und dieses Delay wieder deaktiviert, Problem ist aber nach wie vor noch da, somit habe ich das Delay für mich wieder aktiviert, trotzdem danke für die anderen Fehlerbehebungen und Features.

SW-Niko, 2026-08-31:
Hallo @PhilJaro,
magst du mal einen Blick auf den Fix werfen?

PhilJaro, 2026-09-08:
Hallo @SW-Niko,
ich nutze das Senden mehrerer MQTT commands aktuell nicht mehr, aber zum testen habe ich mir den fix mal aufgespielt und mehrere Kommandos gleichzeitig abgesetzt, ohne Problem. Sieht für mich daher so aus als würde es den Fehler beheben, danke.

SW-Niko, 2026-09-08:
Hallo @PhilJaro ,
danke fürs testen. Ich verwende ja kein MQTT und bin mehr durch Zufall über den Bug gestolpert.

Dann werde ich in den kommenden Tagen im Upstream Projekt einen PR platzieren.

rudolfp1979, 2026-09-08:
Das erklärt meinen Absturz heute Nacht seit dem ich mir einen websocket exporter für die victron Daten und die pylontech Batterie gebaut hab

AndreasBoehm, 2026-09-08:
@rudolfp1979 Du bist auf ein anderes Problem gestoßen, Websocket hat mit MQTT nichts zu tun. Du hast wohl zu viele Websocket Clients erzwungen in der DTU und dadurch ist der heap voll gelaufen.

AndreasBoehm, 2026-10-01:
This has been fixed upstream in tbnobody/OpenDTU#3217.

The cause was a race in the configuration `WriteGuard`. If a second config write arrived (from an MQTT command, a Home Assistant entity or the web UI) while the first one was still running, the configuration loop and the second writer could end up waiting on each other forever. The DTU then stopped polling the inverter, battery and charger, and the web UI stayed reachable but couldn't save anything. That is also why spacing the MQTT commands a few seconds apart made the freeze rarer but didn't stop it. Thanks to @SW-Niko for the analysis in #2702.

The fix will land in OpenDTU-OnBattery once #2731 (the merge of upstream `v26.9.28`) is merged. I'm keeping this issue open until then.

AndreasBoehm, 2026-11-05:
https://github.com/hoylabs/OpenDTU-OnBattery/pull/2731 wurde gemerged, damit sollte das Problem erledigt sein.

closed 2026-11-05 by AndreasBoehm