hoylabs/OpenDTU-OnBattery #1650: Add Battery BMS Valence U-Charge U27-36XP via RS485

SigmaSirio, 2025-02-18:
### Is your feature request related to a problem? Please describe.

The Valence U-Charge U27-36XP battery communicates via RS485, but there is currently no integration for its BMS in OpenDTU. I have a lot of material, link and there is a developer who seems to have decoded the serial string, it would be useful to have native support for this battery's BMS, avoiding the need for external shunt measurements.
Can I paste here all the link?

### Describe the solution you'd like

I would like to see the Valence U-Charge U27-36XP BMS integrated into OpenDTU, allowing direct communication via RS485. This would enable real-time monitoring of battery parameters without requiring a shunt. (I think) I am available for testing and can contribute with some basic programming and electronic creations if needed.

AndreasBoehm, 2025-02-18:
Thanks for your request.

To integrate this BMS we will need a) someone who has the hardware and is willing to implement it and b) a protocol documentation.

I added the 'contribution needed' label as i won't be implementing a protocol blindly.

SigmaSirio, 2025-02-18:
a) I have the battery and I can contribute.
b) I found these devs who have already developed something, I searched for "Valence Battery". I'll paste the clearest ones for you
the first link describes part of the protocol but be careful because it talks about a different model: RT instead of XP, I think it's the same, but you never know.

https://github.com/Canyoneer/ValenceRT-DecodingTool
https://github.com/t3chN0Mad/OpenXPBMS
https://github.com/seb303/Arduino-XP-BMS

Thanks a lot

AndreasBoehm, 2025-02-18:
Perfect!
You can take at the structure for battery providers in this PR https://github.com/hoylabs/OpenDTU-OnBattery/pull/1451 which will be merged soon.

Happy coding!