hoylabs/OpenDTU-OnBattery #1656: DPL change Target Grid Consumption on Tablet

michael1413, 2025-02-19:
### What happened?

It is not possible to set a negativ value  in Target Grid Consumption if you use a tablet.
The keyboard has no sign and copy and paste a negativ value is also not possible.

Samsung Tab S8

Michael 

### To Reproduce Bug

Try to change the Target Grid Consumption value to -5W

### Expected Behavior

The value should be -5W
I can set it on my PC but not with my tablet.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-02-19:
I don't think that this is an issue with the web app... Much more likely with your device. There is nothing there which tells the browser that the value must be greater than zero.

[image]

Please provide a screenshot and triple-check your tablet.

michael1413, 2025-02-19:
[image]

Test with Opera and Brave.
Language German and English.

spcqike, 2025-02-19:
Still an OS problem. Not opendtu. 

What hides behind this small keyboard icon? 

[image]

Don’t you have a way to switch to another board with more buttons?

michael1413, 2025-02-19:
I can change only the mode off the keyboard. 
Flowting or default on the bottom position.
I found no other keyboard layout for this value entry.

Android 14 and Samsung UI 6.1.1

schlimmchen, 2025-02-20:
This is a Samsung-specific issue, even more so, specific to the Samsung default keyboard. See [The Internet](https://stackoverflow.com/questions/21677695/input-type-number-not-possible-to-insert-negative-numbers-on-mobile-phone).

We are not working around a vendor-specific bug -- and this is certainly a bug in their design.

Please install another keyboard, like [Gboard](https://play.google.com/store/apps/details?id=com.google.android.inputmethod.latin&hl=en).

closed 2025-02-20 by schlimmchen