hoylabs/OpenDTU-OnBattery #1657: Battery discharge configuration: voltage for limit is used, even when SoC is valid + enabled

hhoebel, 2025-02-20:
### What happened?

Die Kommentare in der Konfiguration für das Entladelimit der Batterie beschreiben, das das Voltage-Limit nur verwendet wird, wenn der SoC ignoriert werden soll. Tatsächlich wird aber auch die Spannungsgrenze zusätzlich zum SoC verwendet, auch wenn dieser aktiv ist.

### To Reproduce Bug

Voltage und SoC so einstellen, das die SoC-Grenze zur Limitierung erfüllt ist, aber die Voltage-Grenze nicht --> es erfolgt keine Limitierung, obwohl der SoC unterschritten ist.

### Expected Behavior

Siehe oben. Entweder die Beschreibung ist falsch oder die Implementierung - ich vermute letzteres.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025-02-10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

Das Problem findet sich in der Battery.cpp in getDischargeCurrentLimit():
`if (statsSoC > dischargeCurrentLimitBelowSoc && statsVoltage > dischargeCurrentLimitBelowVoltage) {
// Above SoC and Voltage thresholds, ignore custom limit.
// Battery-provided limit will still be applied.
        dischargeCurrentLimitValid = false;
    }`

Stattdessen sollte das eher so aussehen:
`    if (statsSoCValid)
    {
        if (statsSoC > dischargeCurrentLimitBelowSoc) {
            // Above SoC and Voltage thresholds, ignore custom limit.
            // Battery-provided limit will still be applied.
            dischargeCurrentLimitValid = false;
        }
    } else {
        if (statsVoltage > dischargeCurrentLimitBelowVoltage) {
            // Above SoC and Voltage thresholds, ignore custom limit.
            // Battery-provided limit will still be applied.
            dischargeCurrentLimitValid = false;
        }
    }`

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

hhoebel, 2025-02-20:
Der Bugfix ist übrigens auch in meinem Pull-Request enthalten:
https://github.com/hoylabs/OpenDTU-OnBattery/pull/1658

Manos1966, 2025-02-21:
Good to know!

Just in case anyone needs it, below is the voltage calculation related to SOC % for Pylontech 48V (14 cells) batteries and other 51V (15 cells)

[image]

and one more Graph for the Pylontech batteries

[image]

PR #1737 (2025-03-19): Fix: make getDischargeCurrentLimit() readable and return expected value
this change re-implements said function, using lambda functions inside it to allow for usage of return statements, which greatly improve readability.

closes #1657.
closed 2025-03-19 by schlimmchen