hoylabs/OpenDTU-OnBattery #1662: [Request] Extend excess charging support to trucki T2MG stick (and MeanWell NPB adapter series)

swingstate, 2025-02-20:
### Is your feature request related to a problem? Please describe.

No, however the excess charging based on the Huawei RG4850G2 poses challenges to users (cabling, configuration) and the # of support requests show that a more straightforward solution could benefit many users. 

### Describe the solution you'd like

### Feature Request: Extend Excess Charging Support to trucki T2MG Stick and MeanWell NPP Adapter Series

#### Overview

I propose adding support for the [Trucki T2MG stick](https://trucki.de/t2mg/), and by extension, the MeanWell NPB power adapter/charger series, to enhance the excess charging capabilities of OpenDTU-onBattery. 

#### Proposed Changes, based on the Input from Trucki himself. 

**1. UDP Communication:**  
- **Transmission:** OpenDTU will transmit a UDP packet approximately every second to the T2MG stick’s IP address on port 4211.  
- **Payload:** The packet will contain the desired charging power, where the value (in watts) is multiplied by 10 to encode the data.

**2. Response Handling:**  
- **Format:** The T2MG stick is expected to respond with a UDP packet structured as follows:  
  `AC OUT; Powerlimit; Battery State`  
- **Integration:** The system will parse this response to update the charging status and other relevant system metrics.

**3. Device Discovery:**  
- **mDNS Integration:** The T2MG stick will be automatically discovered using mDNS with the call:  
  `MDNS.addService("t2mglink", "tcp", 80);`

#### Rationale

The proposed changes follow the integration guidelines provided by Trucky, ensuring that the implementation is straightforward while significantly enhancing the functionality of OpenDTU-onBattery. This feature will simplify the process of installing and using the excess charging capability. 

### Describe alternatives you've considered

-

### Additional context

-

AndreasBoehm, 2025-02-20:
Those adapters are expensive… and the meanwell chargers as well.
Is there any official documentation available for what you proposed?
Is the procedure the same for the T2HG?
What about supporting the meanwell chargers directly as implemented in the fork of @skippermeister ? Would save a lot of money.

schlimmchen, 2025-02-20:
> What about supporting the meanwell chargers directly as implemented in the fork of @skippermeister ? Would save a lot of money.

Good point, didn't think of that.

Also, as I told Christian already on discord, the premise for integration of such a feature is an abstraction for multiple AC charger providers, similar to the abstraction we have for power meters and batteries, which is quite the feature in and on itself.

swingstate, 2025-02-20:
> Is there any official documentation available for what you proposed?
> Is the procedure the same for the T2HG?
> What about supporting the meanwell chargers directly as implemented in the fork of [@skippermeister](https://github.com/skippermeister) ? Would save a lot of money.

Regarding support/documentation - I’m not entirely sure and shared what I know above, but I had an email exchange with Christian (Trucki), and he’s willing to support the idea. However, the extent of his support remains to be clarified—perhaps it would be beneficial to reach out to him directly.
From what I understand, the Meanwell PSUs require some interfacing, which is what the T2MG stick accomplishes in an easy plug n play way. For instance, the NPB 1700-48 uses a 14-pin Hirose port for this purpose.

[image]

If there were a simple, plug-and-play method to connect Meanwell PSUs to an ESP, I would be all for it. Since I’m not aware of any such solution at this time, I recommend integrating the existing solution via software. Additionally, the capability to wirelessly use multiple chargers—as schlimmchen mentioned—would be a significant advantage.
Looking ahead, I could consider adding a port for the Meanwell PSU to one of my boards later this year. I also agree that both the stick and the PSUs are relatively expensive compared to the Huawei option.

rainer667, 2025-02-22:
Hello,
After optimizing my system, I needed to put excess solar power into the battery to charge the car in the evening. This works really well with the Trucki Stick, but the OpenDTUonBattery tends to get into each other's way and get into a tizzy. It works best when the Trucki Stick keeps the inverter output <= -100W. But the best thing would certainly be communication between the two to avoid a tizzy.
Please excuse my bad English. Automatic Google translation
LG Rainer

tobi0171, 2025-03-03:
Hello. 
I would like it. 
I asked about it a very long time ago. There are obviously still a few 24V users. The Trucki stick's has another advantage apart from the oDTUoB use. You can maybe use a 12V, 24V or 48V MW Charger which is in a different place but within the same WiFi, for example, if the 48V storage of the openDTUonBattery-System is full, you can operate another system with 12V or 24V or 48V (power station, 2nd (stand alone) battery or just use a 12V and use/charge it based on the available excess solar power. You don't have to be near the ESP/DTU for this. 

That's why I use this system in parallel occasionally. Just as a further advantage, in case if someone is looking for arguments and can take the time for it.

AndreasBoehm, 2025-03-14:
I updated the issue with the details that i received from Trucki and created a new issue for the direct integration of meanwell chargers without the T2MG: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1725

AndreasBoehm, 2025-07-02:
If anyone is interested you can try the first alpha version of the Trucki T2HG/T2MG implementation in OpenDTU using a test-version from https://github.com/hoylabs/OpenDTU-OnBattery/pull/2092#issuecomment-3025053657

rainer667, 2025-07-10:
@AndreasBoehm The test version works well with excess power. < 100W
Thanks Rainer

rainer667, 2025-07-11:
@AndreasBoehm Will T2MG/T2HG support be included in the next version?

AndreasBoehm, 2025-07-12:
I cant tell yet.

rainer667, 2025-07-18:
I've continued testing, and it works great (20W) without having to restart the controller on the Trucki board.
This extension definitely needs to be integrated into the software as an enhancement and makes it easier to add a charging option to OpenDTU without any major tinkering. I wouldn't want to be without this extension anymore. 
@AndreasBoehm vielen Dank für die Arbeit.

Cartman586, 2025-08-27:
@AndreasBoehm Would it be helpful if Trucki provided a parameter that allows 6000 watts to be entered as a separate value so that the T2HG Master knows what to do?

AndreasBoehm, 2025-08-27:
I think that Trucki would need to re-think his whole local/remote control mechanism to make that possible.
Because we are using the remote control feature to control the T2xG it is switched to remote mode, which basically makes it dumb and only work based on values provided by external control. The master is OpenDTU.

Support for more than one ac charger will not be part of the alpha or even the first release version of the T2xG implementation.

Cartman586, 2025-08-27:
@AndreasBoehm That's a shame. I can't continue testing at the moment, but I'm still very interested.

Cartman586, 2025-08-27:
@AndreasBoehm Trucki is thinking about a possible slave service. Maybe you contact him by email and something nice will come out of it. Developers and developers :-) https://trucki.de/impressum/

rainer667, 2025-08-27:
I have a question about this alpha version. If the SOC full solar passthrough start threshold is 85%, it shuts down the power supply even though the Trucki power supply is at 95%. Because if that's what's required, I don't understand the parameter that shuts down the power supply. Please don't misunderstand me, I'm extremely happy with how it works. I haven't been able to test the emergency charging, but winter is sure to come.

AndreasBoehm, 2025-08-28:
The ac charger is only ON as long as NO battery powered inverter is ON, because we don't want to circulate the power from battery to grid and then from the grid to the battery again.

@rainer667 That is expected, because as soon as full-solar-passthrough starts the battery powered inverter will be turned on, which leads to the ac charger being turned off. For a meaning full config your ac charger SoC setting needs to be equal or lower than the one set for full-solar-passthrough.

@Cartman586 I hope what i wrote above answers your question. This interlocking feature is actually the only reason to have OpenDTU control the ac charger.
I am in contact with trucki, he was guiding me for the initial T2xG implementation. 
The limiting factor to get T2xG from alpha to release and to get multiple ac chargers integrated is time, not technology.

rainer667, 2025-08-28:
Hello @AndreasBoehm ,
I understand that, but wouldn't it make sense to let the charger charge up to the set SoC? I switch off at 80%. If I want to slowly charge the battery, I have to turn off the charger at the moment.

Cartman586, 2025-08-28:
@AndreasBoehm Hello Andreas, thank you very much for your feedback. I find this topic really exciting. I have already spoken to Trucki, and he is considering something related to slave power, which would only require a T2HG to pass on the slave power to the second T2HG. I would be really happy if you could stay on top of this topic and its development. I am very grateful to you for taking up this issue. I assume you use a T2xG yourself? We can also write in German if you prefer, it doesn't matter to me.