hoylabs/OpenDTU-OnBattery #1665: smartbuffer DPL hängt bei zu hohem Limit fest

Martin1275, 2025-02-21:
### What happened?

DPL regelt nicht runter wohl der Zielwert weit überschritten ist (es wird zu viel eingespeist)

### To Reproduce Bug

Vermutung: DPL regelt nicht runter wril die Batterie die erwartete Leistung/MPPT nicht liefern kann.

### Expected Behavior

DPL sollte jederzeit runterregeln können.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
Success
17:11:08.349 > Websocket: [/livedata][27] disconnect
17:11:14.675 > Fetch inverter: 1164A0099655
17:11:14.859 > Queue size - NRF: 0 CMT: 1
17:11:15.152 > Success
17:11:15.250 > [DPL] waiting for sufficiently recent power meter reading
17:11:17.266 > [DPL] up 36220 s, next inverter restart at 60788 s (set to 0)
17:11:17.285 > [DPL] targeting -15 W, base load is 40 W, power meter reads -162.0 W (valid)
17:11:17.311 > [DPL] inverter 1164a0099655 is behind power meter producing 410 W
17:11:17.335 > [DPL] requesting 263 W from 1 smart-buffer-powered inverter currently producing 410 W (diff -147 W, hysteresis 10 W)
17:11:17.379 > [DPL inverter 1164a0099655]:
17:11:17.412 >     expected AC power per MPPT 490 W
17:11:17.433 >     MPPT-a AC power 0 W
17:11:17.464 >     MPPT-b AC power 21 W
17:11:17.490 >     MPPT-c AC power 26 W
17:11:17.505 >     MPPT-d AC power 362 W
17:11:17.528 >     all mppts are shaded, keeping the current limit of 2000 W
17:11:17.552 > [DPL] will cover 1400 W using 1 smart-buffer-powered inverter
17:11:17.576 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
17:11:17.609 > [DPL inverter 1164a0099655]:
17:11:17.631 >     smart-buffer-powered, producing 410 W
17:11:17.658 >     lower/current/upper limit: 20/2000/1400 W, output capability: 2000 W
17:11:17.678 >     sending commands enabled, reachable, eligible
17:11:17.713 >     max reduction production/standby: 390/410 W, max increase: 990 W
17:11:18.136 >     target limit/output/state: 2000 W (update)/1400 W/production, 0 update timeouts
17:11:18.192 >     MPPTs AC power: a: 0 W b: 21 W c: 26 W d: 362 W
17:11:18.213 > [DPL inverter 1164a0099655]: sending limit of 100.0 % (2000 W respectively), max output is 2000 W
17:11:18.231 > [DPL] waiting for a start/stop/restart/limit command to complete
17:11:19.362 > Success
17:11:19.463 > [DPL inverter 1164a0099655]: actual limit is 100.0 % (2000 W respectively), effective 2045 ms after update started, requested were 100.0 %
17:11:19.486 > [DPL] waiting for sufficiently recent inverter data
17:11:21.660 > Fetch inverter: 1164A009A1B6
17:11:21.839 > Queue size - NRF: 0 CMT: 1
17:11:22.274 > Success
17:11:28.731 > Fetch inverter: 1164A0099655
17:11:28.817 > Queue size - NRF: 0 CMT: 1
17:11:29.194 > Success
17:11:29.212 > [DPL] waiting for sufficiently recent power meter reading
17:11:31.346 > [DPL] up 36234 s, next inverter restart at 60788 s (set to 0)
17:11:31.550 > [DPL] targeting -15 W, base load is 40 W, power meter reads -160.0 W (valid)
17:11:31.753 > [DPL] inverter 1164a0099655 is behind power meter producing 409 W
17:11:32.165 > [DPL] requesting 264 W from 1 smart-buffer-powered inverter currently producing 409 W (diff -145 W, hysteresis 10 W)
17:11:32.265 > [DPL inverter 1164a0099655]:
[... 102 lines cut ...]
17:12:15.067 > [DPL] will cover 1400 W using 1 smart-buffer-powered inverter
17:12:15.103 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
17:12:15.128 > [DPL inverter 1164a0099655]:
17:12:15.151 >     smart-buffer-powered, producing 409 W
17:12:15.178 >     lower/current/upper limit: 20/2000/1400 W, output capability: 2000 W
17:12:15.199 >     sending commands enabled, reachable, eligible
17:12:15.219 >     max reduction production/standby: 389/409 W, max increase: 991 W
17:12:15.241 >     target limit/output/state: 2000 W (update)/1400 W/production, 0 update timeouts
17:12:15.261 >     MPPTs AC power: a: 0 W b: 21 W c: 25 W d: 363 W
17:12:15.280 > [DPL inverter 1164a0099655]: sending limit of 100.0 % (2000 W respectively), max output is 2000 W
17:12:15.308 > [DPL] waiting for a start/stop/restart/limit command to complete
```

### Anything else?

Der max Wert von 1400W für den Inverter wird anscheinend auch nicht eingehalten... 

Wechselrichter ist HMS 2000 4t

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-02-21:
Vielen Dank für die ausführliche Beschreibung inklusive Log.

Ich denke ich hab das Problem gefunden, du kannst dir hier eine Test-Version herunterladen: https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/13462326933

Bitte gib Bescheid obs damit läuft wie erwartet.

> Der max Wert von 1400W für den Inverter wird anscheinend auch nicht eingehalten...

Ja und nein. Da du das ausgleichen von verschatteten Eingängen aktiviert hast kann das Limit bis auf das maximum des Inverters skaliert werden, die maximale Ausgangsleistung wird aber entsprechend deiner Einstellung auf 1400 W begrenzt.

Martin1275, 2025-02-21:
Habs installiert und sieht bisher gut aus....

Martin1275, 2025-02-21:
Also bisher ist der Fehler nicht mehr aufgetreten :-)

Was aber noch nicht ganz rund läuft.... unter 15% produziert der Wechselrichter zu wenig 
Gab es da nicht mal Pläne unter 15% den Wert um Faktor X zu erhöhen? bzw eine Tabelle zum anpassen. 

Zweitens: wenn nur 2 von 4 Eingängen belegt sind und die Last im Haus ändert sich, scheint OpenDTUoB "zu überereagieren".

Dann wird zwischen den Extremen -53 und +35Watt hin und her gesprungen.(soll -15Watt)
Manchmal wird der WR sogar abgeschalten (gelber Status)

Aber hier nicht zu viel Kritik...  Es hat sich mit den letzten Versionen einiges getan, vor allem die Verbindung zum WR ist stabiler. 👍

AndreasBoehm, 2025-02-22:
Sehr gut 😃 

Da mit zu geringer Leistung einfach der Akku länger hält und somit unterm Strich gleich viel durch PV Strom abgedeckt wird haben wir das nicht weiter verfolgt, noch nicht.

Kannst du zu dem schwingenden Verhalten auch noch einen Log machen? Das wäre super.

Martin1275, 2025-02-23:
Hallo Andreas, das Schwingen hab ich seitdem nicht mehr beobachtet...  hatte noch ein wenig an der Hysterese des DPL rumprobiert und seitdem scheint es stabil zu laufen.

Wenn das Problem  mit der zu niedrigen Erzeugung im unteren Lastbereich gelöst würde wäre noch schön.

Ich hatte mir mal selber eine FW kompiliert in der nach der DPL Berechnung  Werte bis einschließlich 14 um 1% erhöht wurden.
Aus 6% wurde dann z.B 7%. Das hat eigentlich im groben recht gut funktioniert.

spcqike, 2025-02-23:
> Ich hatte mir mal selber eine FW kompiliert in der nach der DPL Berechnung Werte bis einschließlich 14 um 1% erhöht wurden.
Aus 6% wurde dann z.B 7%. Das hat eigentlich im groben recht gut funktioniert.

Das mag für deinen speziellen inverter gepasst haben. Leider kann es sein, dass die inverter da unterschiedliche Abweichungen haben. Aufgrund von Toleranzen der Bauteile in der Fertigung? Dann sieht es zumindest bei meinen so aus, als wäre das auch temperaturabhängig. Dann waren die „unterdeckungen“ auch abhängig von der Anzahl der belegten Eingänge, wenn ich mich recht erinnere. Und zu guter letzt mag es mit unterschiedlichen firmwares das Verhalten unterschiedlich ist. 

Kurz um:
Das ist nicht trivial. Wenn du die Möglichkeit hast ne eigene Firmware zu compilieren und du den Bedarf siehst den einen Prozent noch rauszuholen, würd ich es an deiner Stelle weiterhin selber machen 😄

Martin1275, 2025-02-23:
Vielleicht könnte man ja eine Logik zum "feintunig" einbauen... unter 15% kann das Limit z.B. mehrmals erhöht werden, bis zu 15% . Ab hier sollte die Regelung ja wieder zuverlässig funktionieren und die tatsächliche verfügbare Leistung erkennen. 

Bin gerne für Betatests bereit.

AndreasBoehm, 2025-02-23:
Ich habe für die Problematik mit niedrigen Limits ein neuen Issue angelegt: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1675

Lasst uns hier bitte beim Thema bleiben.

Martin1275, 2025-02-23:
Okay.

Also von mir aus wäre das Problem mit den hängenbleiben in zu hohem Limit gelöst. 

Soll ich das schließen?  Oder kannst du das auch machen @AndreasBoehm

AndreasBoehm, 2025-02-23:
Danke für die Info @Martin1275  dann mache ich hier zu.

closed 2025-02-23 by AndreasBoehm

Gururom, 2025-03-11:
Bei mir scheint das überscalieren auch weg mit dieser Version.
Habe einen HMS16oo und zwei Marstek , wenn ein Akku leer war ists sonst immer hochgerannt und hat sich nicht ans Limit gehalten. Ich hatte deswegen das "verschattete Eingänge ausgleichen" rausgenommen.
Jetzt ist es drin, steht auf 50%, da ja sowieso entweder der Akku liefert oder nicht.
Es macht ein guten Eindruck.
Fließt das dann in die nächste offizielle Version ein?

Grüße und besten Dank
Stephan

Larvy, 2025-03-17:
Hallo ich teste die hier zur Verfügung gestellten Version mit 2 HMS-2000-4T. Die Regelung des ersten WR funktioniert schon soweit ganz gut nur der zweite scheint immer noch das Problem des hängen bleibens zu haben. Ich habe noch eine 2. OpenDTU und werde beide mal getrennt laufen lassen. Beide Wechselrichter werden mit BluePalm und Marstek Speicher auf allen Eingängen versorgt.
Wenn beim ersten WR ein Speicher leer ist wird der Mangel mit dem anderen über den anderen beiden Eingängen ausgeglichen.
Bricht aber der 1. WR komplett weg wegen leerer Speicher gleicht das der 2. WR nicht aus.

Wenn dann mal von dem 2. WR volle Leistung gefordert wird bleibt der eine Weile auf dieser Leistung hängen und speist unnötig viel Energien ins Netz.

Leider habe ich noch  nicht rausbekommen wie ich eine Logdatei anfertige. Sonst hätte ich sie hier mit angehangen.

AndreasBoehm, 2025-03-17:
@Larvy nimm mal die Test-Version von diesem Kommentar https://github.com/hoylabs/OpenDTU-OnBattery/pull/1714#issuecomment-2708798020

Da sollten die Probleme die du beschreibst behoben sein.

Larvy, 2025-03-18:
Ich habe die version jetzt installiert und mal paar std. laufen lassen. Sie schwingt mir leider zu stark hin und her. Was dazu führt das die Energie teilweise verschwendet wird. Da fand ich die Version hier besser. Dann lasse ich den eimen WR erstmal ungeregelt aus grundbedarf laufen und den Anderen WR dann auf Nulleinspeisung.

Trotzdem danke Tip

[image]

Larvy, 2025-03-20:
> [@Larvy](https://github.com/Larvy) nimm mal die Test-Version von diesem Kommentar [#1714 (comment)](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1714#issuecomment-2708798020)
> 
> Da sollten die Probleme die du beschreibst behoben sein.

Vielen Dank für den Tip ich hab meine Aussage gelöscht was die hier verlinkte Version angeht. Sie funktioniert perfekt. Ich habe den DPL nochmal fein eingestellt.  Nun läuft es mit 2 HMS-2000-4T und jeweils 2 Mastek und Bluepalm Speicher wie erwartet. Jetzt bloß nichts mehr anfassen. ;)

Vielen Dank nochmal.

PR #1666 (2025-02-24): fix: DPL use current AC output for smart buffer reduction
With the scaled limit we are not able to reduce the power as expected.
Closes #1665.