hoylabs/OpenDTU-OnBattery #1672: DPL ändert Limit nicht und Inverter produziert nur etwa 46% des Limits

steeed, 2025-02-22:
### What happened?

Der DPL behält das Limit längere Zeit bei obwohl der Netzbezug steigt (selbst bei 2-3 kW über dem gesetzten Limit.

Und seit Release 2025.01.23 (betrifft auch 2025.02.10) produziert der Wechselrichter nur etwa 46% des gesetzten Limits. Dies ist mit früheren Releases nicht der Fall.

### To Reproduce Bug

Passiert regelmässig im laufenden Betrieb.

### Expected Behavior

Limit wird entsprechend dem Verbrauch gesetzt und Wechselrichter produziert entsprechend dem gesetzten Limit.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
Fetch inverter: 116494503602
14:14:16.503 > Queue size - NRF: 0 CMT: 1
14:14:16.537 > [PowerMeterHttpJson] New total: 137.36
14:14:16.615 > Websocket: [/livedata][28] disconnect
14:14:16.826 > [DPL] targeting 5 W, base load is 38 W, power meter reads 137.4 W (valid)
14:14:16.968 > [DPL] inverter 116494503602 is behind power meter producing 49 W
14:14:17.061 > [DPL] requesting 181 W from 1 solar-powered inverter currently producing 49 W (diff 132 W, hysteresis 5 W)
14:14:17.076 > [DPL] will cover 49 W using 1 solar-powered inverter
14:14:17.497 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 132 W AC
14:14:17.573 > [DPL inverter 116494503602]:
14:14:17.702 >     solar-powered, producing 49 W
14:14:17.781 >     lower/current/upper limit: 35/105/800 W, output capability: 1600 W
14:14:17.807 >     sending commands enabled, reachable, eligible
14:14:17.838 >     max reduction production/standby: 14/14 W, max increase: 0 W
14:14:17.870 >     target limit/output/state: -1 W (unchanged)/49 W/unchanged, 0 update timeouts
14:14:17.918 >     MPPTs AC power: a: 25 W b: 0 W c: 25 W d: 0 W
14:14:17.952 > Success
14:14:17.975 > [PowerMeterHttpJson] New total: 233.40
14:14:17.993 > [DPL] targeting 5 W, base load is 38 W, power meter reads 233.4 W (valid)
14:14:18.062 > [DPL] inverter 116494503602 is behind power meter producing 49 W
14:14:18.444 > [DPL] requesting 277 W from 1 solar-powered inverter currently producing 49 W (diff 228 W, hysteresis 5 W)
14:14:18.523 > [DPL] will cover 49 W using 1 solar-powered inverter
14:14:18.680 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 228 W AC
14:14:18.818 > [DPL inverter 116494503602]:
14:14:19.179 >     solar-powered, producing 49 W
14:14:19.334 >     lower/current/upper limit: 35/105/800 W, output capability: 1600 W
14:14:19.360 >     sending commands enabled, reachable, eligible
14:14:19.377 >     max reduction production/standby: 14/14 W, max increase: 0 W
14:14:19.400 >     target limit/output/state: -1 W (unchanged)/49 W/unchanged, 0 update timeouts
14:14:23.754 >     MPPTs AC power: a: 25 W b: 0 W c: 25 W d: 0 W
14:14:23.754 > Fetch inverter: 116494503602
14:14:23.754 > Queue size - NRF: 0 CMT: 1
14:14:23.755 > [PowerMeterHttpJson] New total: 1166.87
14:14:23.755 > Success
14:14:23.755 > [DPL] targeting 5 W, base load is 38 W, power meter reads 1166.9 W (valid)
14:14:23.755 > [DPL] inverter 116494503602 is behind power meter producing 49 W
14:14:23.755 > [DPL] requesting 800 W from 1 solar-powered inverter currently producing 49 W (diff 751 W, hysteresis 5 W)
14:14:23.755 > [DPL] will cover 49 W using 1 solar-powered inverter
14:14:23.755 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 751 W AC
14:14:23.755 > [DPL inverter 116494503602]:
14:14:23.755 >     solar-powered, producing 49 W
[... 815 lines cut ...]
14:16:26.680 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 119 W AC
14:16:26.680 > [DPL inverter 116494503602]:
14:16:26.680 >     solar-powered, producing 84 W
14:16:26.680 >     lower/current/upper limit: 35/185/800 W, output capability: 1600 W
14:16:26.680 >     sending commands enabled, reachable, eligible
14:16:26.680 >     max reduction production/standby: 49/49 W, max increase: 0 W
14:16:26.680 >     target limit/output/state: -1 W (unchanged)/84 W/unchanged, 0 update timeouts
14:16:26.680 >     MPPTs AC power: a: 42 W b: 0 W c: 42 W d: 0 W
14:16:26.681 > [PowerMeterHttpJson] New total: 93.67
14:16:26.681 > Fetch inverter: 116494503602
14:16:26.681 > Queue size - NRF: 0 CMT: 1
14:16:26.681 > [DPL] targeting 5 W, base load is 38 W, power meter reads 93.7 W (valid)
14:16:26.681 > [DPL] inverter 116494503602 is behind power meter producing 84 W
14:16:26.681 > [PowerMeterHttpJson] New total: 85.09
14:16:26.681 > [DPL] requesting 173 W from 1 solar-powered inverter currently producing 84 W (diff 89 W, hysteresis 5 W)
14:16:26.681 > [DPL] will cover 84 W using 1 solar-powered inverter
14:16:26.681 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 89 W AC
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-02-23:
Nutzt du einen Akku oder hast du an den zwei Eingängen deines WRs Solarmodule angeschlossen?

Ich denke generell gibt es hier ein missverständnis. Bei Solarbetrieb setzen wir das Limit nicht einfach entsprechend dem Netzbezug, sondern prüfen ob der WR mehr liefern könnte bzw mehr liefert als der Netzbezug und passen das Limit nur dann an wenn sich durch die Änderungen die Ausgangsleistung des WRs ändern würde.

> Der DPL behält das Limit längere Zeit bei obwohl der Netzbezug steigt

Durch den Support von mehreren WRs im DPL prüfen wir ob der WR in der Lage ist mehr Leistung zu liefern bevor wir das Limit erhöhen. In deinem Log ist zu sehen das du mit 25 W an zwei Eingängen vermutlich knapp an den geforderten 25,46 W (97% des gesetzten Limits pro Eingang) vorbei geschrammt bist, doch dadurch wird verhindert das wir das Limit erhöhen. Später im Log sieht man aber das der DPL das Limit auf 185 W erhöht hat.

Da du nur zwei Eingänge nutzt würde ich dir Empfehlen "Ausgleichen von verschatetten/ungenutzten Eingängen" zu aktivieren und dann etwas mit dem Verschattungs Threshold spielst bis du eine Einstellungen gefunden hast die bei dir verhindert das du nicht mehr von einem niedrigen Limit wegkommst.

> Manchmal wird das Limit auch nicht angepasst, wenn der Verbrauch sinkt.

Nur die Ausgangsleistung ist relevant, die liegt trotz hohem Limit so niedrig das der WR nicht zu viel produziert -> der DPL regelt nicht nach. Das spart Zeit wenn wir mehr als einen WR steuern müssen, bei dir nicht relevant, aber damit du verstehst warum das so ist.

steeed, 2025-02-23:
Danke für die ausführliche Erklärung. Hab ich soweit verstanden und werde ich heute Abend ausprobieren. 
Ich denke allerdings, dass ich nochmal neu Flashen muss, weil da was schief gelaufen ist. Die Einstellungen für Verschattung / Skalierung sind verschwunden.

AndreasBoehm, 2025-02-23:
Jetzt sag doch mal ob du einen Akku nutzt oder nicht?

Wenn dein WR en HMS-1600/1800/2000 ist und du eine Firmwareversion >= 01.01.12 verwendest gibt es die Option nicht, da der WR dann selbst skaliert bzw bei einem Limit von 100W nicht stumpf jeden Kanal auf 25W limitiert, sondern über jeden Kanal so viel bezieht das er das Limit erreicht. Sorry, das hatte ich vorhin vergessen. Unabhängig davon erreichen die WRs unter 15% das Limit nur selten.

steeed, 2025-02-23:
Ach sorry, überlesen. Ja ich hab einen Akku davor. 

Ja es ist ein HMS-1600 mit der aktuellen Firmware, die den Ausgang limitiert und nicht die Eingänge.  Beim ersten Release mit den Einstellungen für die Skalierung war der Schalter noch zu sehen, deswegen hab ich noch danach gesucht. 
Ich hab das jetzt nochmal beobachtet und mit verschiedenen Verbrauchern getestet. 
Der Hinweis mit den 15% ist gut. Darüber funktioniert es jetzt ziemlich gut und ich bin immer nah am angestrebten Netzbezug. Ist das Limit unter 15% wird das Limit nur zur Hälfte erreicht. Ist mir mit den früheren Releases nie aufgefallen.

closed 2025-02-23 by AndreasBoehm