hoylabs/OpenDTU-OnBattery #1681: DPL und Verbindungsprobleme, wenn Teil der Wechselrichter nicht mehr erreichbar

kutzera, 2025-02-26:
### What happened?

Zuerst einmal, einen herzlichen Dank an die super Entwickler hier. Es ist schon erstaunlich, was ihr hier so programmiert.
Ich habe in den letzten Wochen alle möglichen Firmware-Versionen ausprobiert und stoße seltsamerweise immer wieder auf eine Unregelmäßigkeit.
Wenn alle verfügbaren Wechselrichter online sind, funktioniert das System wunderbar. DPL wird auf allen Wechselrichtern super ausgeführt und die Aktualisierungsrate ist auch topp! Sobald aber die Sonne weg ist und damit die Solar-Wechselrichter offline gehen, dann fängt es bei mir mit Verbindungsabbrüchen zum Batterie-Wechselrichter an. Ich kann einstellen was ich will ( DTU -> Funkfrequenz, Sendeleistung, Aktualiserungsrate, DTU Neustart, Batterie-Inverter offline nehmen und wieder mit Energie versorgen, Andere Netzteile und Kabel) . Nichts hilft. Die Aktualisierungsrate fällt dann manchmal auf einmal die Minute bis manchmal 3 Minuten. Gestern habe ich den Inverter gar nicht mehr verbunden bekommen. Bei der Funkstatistik wird mir kein Fehler angezeigt.
Ich verstehe einfach den Unterschied zwischen Tag und Nacht nicht. 
Den aktuellen Log von heute Abend würde ich heute Abend hinzufügen.

### To Reproduce Bug

bei mir reicht es, wenn die Sonne untergeht und die Inverter offline gehen.

### Expected Behavior

Die Verbindung und damit DPL sollte auch in der Nacht stabil bleiben 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10 aber auch alle anderen ab 2024.11.20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Martin1275, 2025-02-26:
Hast du bei den Wechselrichter Einstellungen auch Daten nachts abfragen / Befehle nachts senden eingeschaltet? 

Welche Farbe hat das Fenster des Wechselrichters wenn er nicht erreichbar ist?

kutzera, 2025-02-26:
> Hast du bei den Wechselrichter Einstellungen auch Daten nachts abfragen / Befehle nachts senden eingeschaltet?
> 
> Welche Farbe hat das Fenster des Wechselrichters wenn er nicht erreichbar ist?

Wo kann ich so etwas einstellen? die Wechselrichter sind gestern alle gelb geblieben

kutzera, 2025-02-26:
> > Bei allen Wechselrichtern ist aktiviert, dass die Daten und Befehle auch Nachts abgerufen und gesendet werden
> 
> das solltest du für Solar-Wechselrichter mal deaktivieren. wie soll die openDTU denn Daten abrufen, wenn das Gerät nicht erreichbar ist weil die Sonne untergeht?

Vielen Dank für die schnelle Rückmeldung. Ich habe die Daten Gleich bei den Solarwechselrichtern deaktiviert und berichte spätestens morgen :)

schlimmchen, 2025-02-26:
Wie sicher kannst du uns denn versprechen, dass nichts anderes dem DPL dazwischenfunkt? Also nichts das ein neues Limit an einen der Inverter schickt oder einen der Inverter aus-/einschaltet, weder per MQTT noch per Web UI/API?

> Gestern habe ich den Inverter gar nicht mehr verbunden bekommen.

Heißt das, dass die "letzte Aktualisierung" bis zu 180 Sekunden her ist bei deinem batteriebetriebenen Wechselrichter?

> das solltest du für Solar-Wechselrichter mal deaktivieren

Hm... Ich glaube @spcqike hat Recht. Wenn die auch "nachts" abgefragt werden, dann vergeblich, und das hält die Hoymiles Loop dann auf, weil sie vergeblich versucht Statistiken abzurufen für die Inverter, die gar nicht mehr erreichbar sind.

Interessant aber, dass dieses Problem bisher scheinbar niemand geschildert hat. Sind die anderen User alle so geschickt das nächtliche Polling abzustellen?

AndreasBoehm, 2025-02-26:
> das solltest du für Solar-Wechselrichter mal deaktivieren. 

Klingt so als ob es sinnvoll wäre das „Powersource“ setting aus dem DPL in die Inverter settings umzuziehen?

schlimmchen, 2025-02-26:
Darüber habe ich schon öfter nachgedacht und mich jedes Mal dagegen entschieden. Insbesondere habe ich Bauchschmerzen dabei, diese Settings dann auch in das entsprechende struct zu verschieben, dass dann Konflikte mit dem Upstream herbeiruft...

spcqike, 2025-02-26:
Ob man das ganze im upstream vorstellen kann, dort jedoch nur mit dem Standard „solar powered“? 
Ggf auch mit einer Abfrage im webUI, dass das Feld nur angezeigt wird, wenn es eben mehr als „Solar powered“ gibt? Also, hier. 

Wäre für die reine OpenDTU sicher …. Unnötig und zu viel. Da es nur ne Vorbereitung für den onBattery fork ist. Dafür wäre es halt tiefer integriert und würde halt weniger Konflikte verursachen. 

¯\_(ツ)_/¯

kutzera, 2025-02-26:
So, bin ch habe die Solar-Wechselrichter mal nachts ignorieren lassen und die Punkte deaktiviert. Trotzdem ist wieder das Problem aufgetaucht. Sobald die anderen Wechselrichter nichts mehr produzieren hat die openDTU Probleme den Batteriewechselrichter konstant zu steuern oder zu erreichen. Tagsüber ist die Verbindung 1A. 
Der log ist vom Zeitpunkt, als er wieder zickte. Aktuell hat er sich wieder gefangen. 

Interrupt received
19:13:33.900 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 91 00 92 02 96 02 9C 00 06 F9 | -53 dBm
19:13:33.928 > Interrupt received
19:13:33.936 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 91 00 06 F0 3D 00 49 00 49 01 C8 01 C7 00 91 04 | -52 dBm
19:13:33.985 > Interrupt received
19:13:33.993 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 03 00 8E 02 95 02 88 00 06 09 AE 00 05 38 6C 00 49 DE | -52 dBm
19:13:34.028 > Interrupt received
19:13:34.033 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 04 00 48 09 3F 13 8B 09 CC 00 02 00 69 03 E7 01 3C 62 | -53 dBm
19:13:34.068 > Interrupt received
19:13:34.074 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 85 00 3C 6A AD 89 | -52 dBm
19:13:34.398 > RX Period End
19:13:34.406 > Success
19:13:34.902 > Fetch inverter: 116481313573
19:13:34.935 > Queue size - NRF: 0 CMT: 1
19:13:34.941 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 59 CE 00 00 00 00 00 00 00 00 5C 7D 92 
19:13:34.948 > Interrupt received
19:13:34.957 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 94 00 95 02 A4 02 AA 00 06 FF | -53 dBm
19:13:34.963 > Interrupt received
19:13:34.969 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 91 00 06 F0 3D 00 49 00 49 01 C8 01 C7 00 94 01 | -53 dBm
19:13:34.984 > Interrupt received
19:13:34.990 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 03 00 91 02 A4 02 97 00 06 09 AE 00 05 38 6C 00 49 EF | -53 dBm
19:13:35.033 > Interrupt received
19:13:35.040 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 04 00 48 09 43 13 8B 0A 03 00 00 00 6C 03 E7 01 3C D5 | -53 dBm
19:13:35.073 > Interrupt received
19:13:35.081 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 85 00 3C 32 40 3C | -53 dBm
19:13:35.347 > RX Period End
19:13:35.354 > Success
19:13:35.823 > Fetch inverter: 116481313573
19:13:35.832 > Queue size - NRF: 0 CMT: 1
19:13:35.844 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 59 CF 00 00 00 00 00 00 00 00 CC 70 0E 
19:13:35.890 > Interrupt received
19:13:35.898 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 97 00 98 02 AF 02 B5 00 06 E5 | -53 dBm
19:13:35.939 > Interrupt received
19:13:35.946 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 91 00 06 F0 3D 00 49 00 49 01 C8 01 C7 00 96 03 | -52 dBm
19:13:35.990 > Interrupt received
19:13:36.000 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 03 00 94 02 AF 02 A2 00 06 09 AE 00 05 38 6C 00 49 D4 | -52 dBm
19:13:36.038 > Interrupt received
19:13:36.047 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 04 00 48 09 43 13 8B 0A 2D 00 00 00 6D 03 E8 01 3C F5 | -52 dBm
19:13:36.080 > Interrupt received
19:13:36.086 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 85 00 3C F4 10 AA | -52 dBm
19:13:36.350 > RX Period End
19:13:36.364 > Success
19:13:36.847 > [PowerMeterHttpJson] New total: -13.98
19:13:36.855 > Fetch inverter: 116481313573
19:13:36.861 > Queue size - NRF: 0 CMT: 1
19:13:36.871 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 59 D0 00 00 00 00 00 00 00 00 FC FC AD 
19:13:36.900 > Interrupt received
19:13:36.906 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 97 00 98 02 AF 02 B5 00 06 E5 | -53 dBm
19:13:36.951 > Interrupt received
19:13:36.956 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 91 00 06 F0 3D 00 49 00 49 01 C8 01 C7 00 96 03 | -52 dBm
19:13:36.999 > Interrupt received
19:13:37.006 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 03 00 94 02 AF 02 A2 00 06 09 AE 00 05 38 6C 00 49 D4 | -52 dBm
19:13:37.048 > Interrupt received
19:13:37.054 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 04 00 48 09 43 13 8B 0A 2D 00 00 00 6D 03 E8 01 3C F5 | -52 dBm
19:13:37.088 > Interrupt received
19:13:37.094 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 85 00 3C F4 10 AA | -52 dBm
19:13:37.359 > RX Period End
19:13:37.368 > Success
19:13:37.871 > Fetch inverter: 116481313573
19:13:38.179 > Queue size - NRF: 0 CMT: 1
19:13:38.225 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 59 D1 00 00 00 00 00 00 00 00 6C F1 31 
19:13:38.232 > Interrupt received
19:13:38.240 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 97 00 98 02 AF 02 B5 00 06 E5 | -53 dBm
19:13:38.253 > Interrupt received
19:13:38.260 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 91 00 06 F0 3D 00 49 00 49 01 C8 01 C7 00 96 03 | -53 dBm
19:13:38.266 > Interrupt received
19:13:38.275 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 03 00 94 02 AF 02 A2 00 06 09 AE 00 05 38 6C 00 49 D4 | -53 dBm
[... 2869 lines cut ...]
19:17:03.901 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 85 00 3C 7F CD FC | -53 dBm
19:17:04.208 > RX Period End
19:17:04.240 > Success
19:17:04.618 > Fetch inverter: 116481313573
19:17:04.653 > Queue size - NRF: 0 CMT: 1
19:17:04.670 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 5A A0 00 00 00 00 00 00 00 00 31 69 86 
19:17:04.698 > Interrupt received
[... cut ...]
19:17:05.193 > Success
19:17:05.616 > Fetch inverter: 116481313573
19:17:05.648 > Queue size - NRF: 0 CMT: 1
19:17:05.663 > TX RealTimeRunData 866.00 MHz --> 15 81 31 35 73 80 15 81 80 80 0B 00 67 BF 5A A1 00 00 00 00 00 00 00 00 A1 64 1A 
19:17:05.709 > Interrupt received
19:17:05.722 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 01 00 01 01 C6 01 C7 00 91 00 92 02 96 02 9C 00 06 F9 | -53 dBm
19:17:05.757 > Interrupt received
19:17:05.767 > RX 866.00 MHz --> 95 81 31 35 73 80 15 81 80 02 35 95 00 06 F0 4

kutzera, 2025-02-26:
So, bin ch habe die Solar-Wechselrichter mal nachts ignorieren lassen und die Punkte deaktiviert. Trotzdem ist wieder das Problem aufgetaucht. Sobald die anderen Wechselrichter nichts mehr produzieren hat die openDTU Probleme den Batteriewechselrichter konstant zu steuern oder zu erreichen. Tagsüber ist die Verbindung 1A.
Aktuell hat er sich wieder gefangen. Ich muss den log mal zufügen, wenn er wieder rumzickt.

spcqike, 2025-02-26:
> Ich muss den log mal zufügen, wenn er wieder rumzickt.

Und dann bitte entsprechend formatieren und verstecken (: nicht so wie eben, bevor du ihn wieder raus editiert hast ;) 

Das geht hier oben in der Leiste unter den „Slash commands“ als „Details“ bspw ganz gut.

kutzera, 2025-02-26:
> > Ich muss den log mal zufügen, wenn er wieder rumzickt.
> 
> Und dann bitte entsprechend formatieren und verstecken (: nicht so wie eben, bevor du ihn wieder raus editiert hast ;)
> 
> Das geht hier oben in der Leiste unter den „Slash commands“ als „Details“ bspw ganz gut.

Sorry, ich wollte den bockmist noch schnell entfernen, bevor jemand die Zeilen bis nach ganz unten scrollen muss 😂

kutzera, 2025-02-27:
Heute mal mit dem log. Irgendwie scheint der batteriewechselrichter nicht das Go zu bekommen, anzuspringen.

/[PowerMeterHttpJson] New total: 271.43
17:38:06.432 > Websocket: [/livedata][21] disconnect
17:38:06.437 > Websocket: [/batterylivedata][21] disconnect
17:38:08.465 > [PowerMeterHttpJson] New total: 270.36
17:38:09.798 > Fetch inverter: 114485010664
17:38:09.805 > Request SystemConfigPara
17:38:09.812 > Queue size - NRF: 0 CMT: 2
17:38:09.821 > TX RealTimeRunData 866.00 MHz --> 15 85 01 06 64 80 15 81 80 80 0B 00 67 C0 94 F1 00 00 00 00 00 00 00 00 BE 66 F6 
17:38:09.828 > Interrupt received
17:38:09.835 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 01 00 01 01 90 01 90 00 03 00 02 00 0D 00 08 00 06 E5 | -57 dBm
17:38:09.842 > Interrupt received
17:38:09.861 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 02 EB 20 00 06 91 91 02 58 02 5B 09 18 13 85 00 00 AC | -57 dBm
17:38:09.867 > Interrupt received
17:38:09.874 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 83 00 00 00 00 00 00 00 5A 00 06 31 99 90 | -57 dBm
17:38:09.881 > RX Period End
17:38:09.887 > Success
17:38:09.899 > TX SystemConfigPara 866.00 MHz --> 15 85 01 06 64 80 15 81 80 80 05 00 67 C0 94 F1 00 00 00 00 00 00 00 00 70 69 39 
17:38:09.909 > Interrupt received
17:38:09.914 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 81 00 01 00 46 00 00 03 E8 00 00 00 00 00 10 1E CB 0F | -57 dBm
17:38:09.990 > RX Period End
17:38:09.998 > Success
17:38:11.642 > [PowerMeterHttpJson] New total: 266.79
17:38:12.809 > [DPL] waiting for sufficiently recent inverter data
17:38:14.303 > Fetch inverter: 116491958590
17:38:14.310 > Queue size - NRF: 0 CMT: 1
17:38:14.316 > TX RealTimeRunData 866.00 MHz --> 15 91 95 85 90 80 15 81 80 80 0B 00 67 C0 94 F6 00 00 00 00 00 00 00 00 8E 40 10 
17:38:14.321 > Interrupt received
17:38:14.326 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 01 00 01 01 0F 01 13 00 02 00 02 00 06 00 08 00 06 04 | -73 dBm
17:38:14.338 > Interrupt received
17:38:14.344 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 02 04 0D 00 05 31 5A 00 00 00 00 01 10 01 11 00 02 76 | -74 dBm
17:38:14.427 > Interrupt received
17:38:14.436 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 03 00 02 00 06 00 06 00 05 C0 B5 00 04 6B 93 00 00 9D | -74 dBm
17:38:14.479 > Interrupt received
17:38:14.488 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 85 00 02 AC B6 8D | -74 dBm
17:38:14.504 > [PowerMeterHttpJson] New total: 269.80
17:38:14.773 > RX Period End
17:38:14.779 > Middle missing
17:38:14.786 > Request retransmit: 4
17:38:14.791 > TX RequestFrame 866.00 MHz --> 15 91 95 85 90 80 15 81 80 84 14 
17:38:14.817 > Interrupt received
17:38:14.825 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 04 00 00 09 2A 13 85 00 19 00 00 00 01 03 E8 00 6B 39 | -74 dBm
17:38:14.874 > RX Period End
17:38:14.880 > Success
17:38:17.887 > [PowerMeterHttpJson] New total: 268.64
17:38:19.220 > Fetch inverter: 116491959108
17:38:19.225 > Queue size - NRF: 0 CMT: 1
17:38:19.242 > TX RealTimeRunData 866.00 MHz --> 15 91 95 91 08 80 15 81 80 80 0B 00 67 C0 94 FB 00 00 00 00 00 00 00 00 1E 18 59 
17:38:19.291 > Interrupt received
17:38:19.297 > RX 866.00 MHz --> 95 91 95 91 08 80 15 81 80 01 00 01 01 9A 01 99 00 00 00 00 00 01 00 00 00 05 9B | -56 dBm
17:38:19.342 > Interrupt received
17:38:19.348 > RX 866.00 MHz --> 95 91 95 91 08 80 15 81 80 02 78 C2 00 05 44 B3 00 CB 00 AD 01 99 01 99 00 00 B0 | -56 dBm
17:38:19.425 > Interrupt received
17:38:19.434 > RX 866.00 MHz --> 95 91 95 91 08 80 15 81 80 03 00 00 00 00 00 00 00 07 A7 E3 00 07 BC 50 02 2F 1A | -56 dBm
17:38:19.483 > Interrupt received
17:38:19.496 > RX 866.00 MHz --> 95 91 95 91 08 80 15 81 80 85 00 06 8D 43 D1 | -56 dBm
17:38:19.774 > RX Period End
17:38:19.783 > Middle missing
17:38:19.785 > Request retransmit: 4
17:38:19.794 > TX RequestFrame 866.00 MHz --> 15 91 95 91 08 80 15 81 80 84 98 
17:38:19.817 > Interrupt received
17:38:19.822 > RX 866.00 MHz --> 95 91 95 91 08 80 15 81 80 04 02 3E 09 17 13 85 00 00 00 00 00 00 00 00 00 50 7C | -56 dBm
17:38:19.875 > RX Period End
17:38:19.881 > Success
17:38:19.891 > [DPL] waiting for sufficiently recent power meter reading
17:38:20.857 > [PowerMeterHttpJson] New total: 268.03
17:38:24.235 > [PowerMeterHttpJson] Value 1: HTTP Error: read Timeout
17:38:25.055 > Fetch inverter: 116491954386
17:38:25.062 > Queue size - NRF: 0 CMT: 1
[... 376 lines cut ...]
17:39:24.522 > Interrupt received
17:39:24.530 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 01 00 01 01 8C 01 8C 00 03 00 02 00 0D 00 08 00 06 E5 | -57 dBm
17:39:24.575 > Interrupt received
17:39:24.583 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 02 EB 20 00 06 91 91 02 58 02 5B 09 0A 13 85 00 00 BE | -57 dBm
17:39:24.618 > Interrupt received
17:39:24.627 > RX 866.00 MHz --> 95 85 01 06 64 80 15 81 80 83 00 00 00 00 00 00 00 5A 00 06 6D BD E8 | -57 dBm
17:39:25.060 > RX Period End
17:39:25.574 > Success
17:39:26.495 > [PowerMeterHttpJson] New total: 259.96
17:39:29.464 > [PowerMeterHttpJson] New total: 260.06
17:39:29.479 > Fetch inverter: 116491958590
17:39:29.479 > Queue size - NRF: 0 CMT: 1
17:39:29.485 > TX RealTimeRunData 866.00 MHz --> 15 91 95 85 90 80 15 81 80 80 0B 00 67 C0 95 41 00 00 00 00 00 00 00 00 BC 02 D6 
17:39:29.525 > Interrupt received
17:39:29.532 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 01 00 01 01 0F 01 14 00 02 00 02 00 06 00 07 00 06 0C | -73 dBm
17:39:29.578 > Interrupt received
[... cut ...]  | -73 dBm
17:39:29.624 > Interrupt received
17:39:29.636 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 03 00 02 00 05 00 05 00 05 C0 B5 00 04 6B 93 00 00 9D | -73 dBm
17:39:29.677 > Interrupt received
17:39:29.685 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 04 00 00 09 2F 13 84 00 18 00 00 00 01 03 E8 00 6B 3C | -74 dBm
17:39:29.715 > Interrupt received
17:39:29.724 > RX 866.00 MHz --> 95 91 95 85 90 80 15 81 80 85 00 02 E7 4A 3A | -73 dBm
17:39:29.987 > RX Period End
17:39:30.001 > Success

Sorbat, 2025-02-27:
`17:38:26.605 > [DPL] battery interface enabled, SoC 98.5 % (ignored), age 6 s (valid)
17:38:26.612 > [DPL] BMS: -1.00 V, MPPT: -1.00 V, inverter 116481313573: 46.80
17:38:26.624 > [DPL] battery voltage 46.80 V, load-corrected voltage 46.80 V @ 0 W, factor 0.00100 1/A
17:38:26.632 > [DPL] battery discharge allowed, start 50.00 V or 80 %, stop 16.00 V or 10 %
17:38:26.638 > [DPL] start NOT reached, stop NOT reached, solar-passthrough disabled, use at night enabled and dormant`

AndreasBoehm, 2025-03-05:
Wie @Sorbat schon in deinem Log gesehen hat wird
a) keine Batterie Spannung vom BMS bereitgestellt obwohl du in deiner Konfiguration SoC ignorieren aktiviert hast 
und 
b) die Spannung am WR liegt mit 46,8V unterhalb der von dir Konfigurierten Startspannung von 50V

kutzera, 2025-03-05:
Vielen Dank euch Schonmal für den vielen Input. Ich habe tatsächlich eine ganz andere Lösung gefunden, die einen kuriosen Auslöser hatte. Scheinbar hat der Empfang nicht gestimmt, wodurch es die ganze Zeit funkabbrüche zu den ganzen invertern gekommen ist. Danach hatte ich versucht, im System den Fehler zu eliminieren, dabei musste ich nur die DTU anders positionieren. durch die funkabbrüche ist der batterieinverter nicht angesprungen, nachdem die sonne untergegangen war, bzw. Hat nicht richtig geregelt, ein Drittel der Daten nicht übermittelt wurde.  Vielen Dank euch nochmal für die tolle Arbeit, die ihr am System vollbringt

schlimmchen, 2025-03-06:
Ich lese, dass dein Problem behoben ist und erlaube mir daher, das Issue zu schließen. Schreibe gerne wieder, falls ich das falsch verstanden habe und es hier noch etwas zu tun gibt.

closed 2025-03-06 by schlimmchen