hoylabs/OpenDTU-OnBattery #1686: Solarbetriebener Inverter regeltz nicht auf 100%, sondern Leistung wird von batteriebetriebenen Inverter bereit gestellt

RealNBB, 2025-02-27:
### What happened?

Hallo,

ist das Verhalten gewollt, dass der solarbetriebene Inverter (in meinem Fall der HMS-1600) auf 8% rum dümpelt, während der batteriebetriebene Inverter (HM-1500) volles Rohr läuft? Macht doch imho keinen Sinn?! :)

[image]

### To Reproduce Bug

wenn die sonne scheint

### Expected Behavior

der sonnenbetriebene Inverter soll auf 100% laufen

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

latest

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2025-02-27:
> der sonnenbetriebene Inverter soll auf 100% laufen

passt nicht zur Aufgabe des DPL. Wenn du das willst, nimm den WR aus dem DPL und setze ihn manuell und dauerhaft auf 100%.

zu deinem Problem.
Ich vermute, da das Limit so gering ist, ist es das typische "der WR schafft das Limit nicht und wird als verschattet angenommen, daher wird gar nicht erst versucht das Limit zu erhöhen"-Problem.

Du hast ein Limit von 117W. nur 2 von 4 Eingängen belegt. also könnte er 58,5W erzeugen. er erzeugt nur 52,8W, also 90.2% des Solls. Wahrscheinlich reicht das nicht, um als "nicht verschattet" zu gelten.

AndreasBoehm, 2025-02-27:
Ausgleich von ungenuzten/verschatteten Eingängen hast du ja bereits aktiv? Dann setze doch mal den Threshold von 98% auf einen niedrigeren Wert.

Manos1966, 2025-02-28:
> der sonnenbetriebene Inverter soll auf 100% laufen

Eine Verständnis-Frage meinerseits: Warum laesst du den "sonnenbetriebenen Inverter" vom DPL steuern?
Er sollte gar nicht gesteuert werden und immer auf 100% Leistung stehen.

RealNBB, 2025-02-28:
> > der sonnenbetriebene Inverter soll auf 100% laufen
> 
> Eine Verständnis-Frage meinerseits: Warum laesst du den "sonnenbetriebenen Inverter" vom DPL steuern? Er sollte gar nicht gesteuert werden und immer auf 100% Leistung stehen (wie im Bild)
> 
> [image]
> 
> [image]

Nein, "sollte" er nicht, da ich eine Nulleinspeisung realisieren möchte. :)

spcqike, 2025-02-28:
> Nein, "sollte" er nicht, da ich eine Nulleinspeisung realisieren möchte. :)

> To Reproduce Bug
>
>wenn die sonne scheint
>Expected Behavior
>
>der sonnenbetriebene Inverter soll auf 100% laufen

🤔

was denn nun?

Manos1966, 2025-02-28:
Danke @spcqike ich habe lange nicht so gut gelacht mit einem einfachen emojy 👍 

OK @RealNBB ich verstehe jetzt was du meinst.... und erkenne die Abweichung von:
To Reproduce Bug
wenn die sonne scheint

Expected Behavior
der sonnenbetriebene Inverter soll auf 100% laufen

Also, doch kein "Bug", sondern "Made by Design" 😄

RealNBB, 2025-02-28:
Kann sein, dass es tatsächlich kein Bug ist, sondern ein gewolltes Verhalten. Was imho aber ziemlich unsinnig ist.
Genauso wie ein anderer User hier auch geschrieben hat, dass es keine priorisierten Inverter gibt, sondern alle immer geregelt werden. Da freut sich die DTU über so viel Arbeit. :)

@spcqike 

Kann es sein, dass du mein Anliegen (auch) nicht verstanden hast?

AndreasBoehm, 2025-02-28:
Nicht gleich den Kopf in den Sand stecken.

Ich denke das ist eine valide Anfrage, die wir uns anschauen sollten.

spcqike, 2025-02-28:
ich denke, dass ich dein Anliegen verstanden habe. Allerdings hast du auf @AndreasBoehm Vorschlag nicht reagiert. Von daher kann ich dir auch nicht weiter helfen.

Ich stecke nicht zu sehr im Detail der Regelung, meine aber, dass Solarbetriebe Inverter bevorzugt werden. Aber eben nur, wenn sie nicht verschattet sind. 
Andernfalls versucht der DPL mit möglichst wenig Übertragungen die Inverter gleich zu belasten. 

Wie gesagt, wird ein Inverter als verschattet erkannt, wird nicht versucht, das Limit zu erhöhen.

AndreasBoehm, 2025-02-28:
> Ich stecke nicht zu sehr im Detail der Regelung, meine aber, dass Solarbetriebe Inverter bevorzugt werden. Aber eben nur, wenn sie nicht verschattet sind. 

Korrekt! Wir versuchen immer zuerst alles über Solar Inverter abzudecken.

> Wie gesagt, wird ein Inverter als verschattet erkannt, wird nicht versucht, das Limit zu erhöhen.

Auch korrekt. Bitte mit dem skalierungs threshold spielen, der beeinflusst auch die erkennung von verschattung.

RealNBB, 2025-02-28:
ich werds mir anschauen.

wollte hier auch nicht dumm kommen, nur hab ich mich gerade bisschen blöd angemacht gefühlt. ;)

spcqike, 2025-02-28:
> nur hab ich mich gerade bisschen blöd angemacht gefühlt

Meine Antwort war auch durchaus spitz/schnippich. Du musst zugeben, dass deine Aussagen einerseits widersprüchlich sind und du auch nicht auf die Lösungsvorschläge eingegangen bist. 😄

schlimmchen, 2025-03-01:
@AndreasBoehm Fürs Protokoll: Auch hier würde ein Regler das Problem lösen. Zielgröße ist (bei solarbetriebenen Invertern) der Haushaltsbedarf. Wenn die solarbetriebenen Inverter den nicht liefern können, dann treibt der integrative Anteil des Reglers das Limit nach oben, weil die Differenz immer bleibt, bis auch 100% Limit nicht ausreichen, was dann okay ist.

AndreasBoehm, 2025-03-01:
@schlimmchen da stimme ich dir zu. Allerdings muss hierfür zuerst die methode zur erkennung von maxIncrease bzw verschattung optimiert werden.

closed 2025-03-20 by AndreasBoehm