hoylabs/OpenDTU-OnBattery #1688: [Request] Mehrere Inverter priorisieren bei DPL

HA-IDIPS, 2025-02-28:
### Is your feature request related to a problem? Please describe.

Erstmal vielen Dank für die tolle Lösung und der Dynamik, mit der sie sich weiter entwickelt.

DPL funktioniert gut, habe ich jetzt zum ersten Mal getestet und war wirklich erstaunt.

Ich habe allerdings zwei Inverter mit Akkubetrieb im Einsatz (HM1500, HM800). Dafür wäre es gut, wenn man für DPL eine Priorisierung der Inverter vorgeben könnte, also zum Beispiel der HM1500 wird erst zugeschaltet, wenn mehr Leistung benötigt wird, als der HM800 liefern kann. Es ist effizienter, wenn bei kleineren Strombezügen nur der HM800 alleine liefert. Derzeit liefern bei geringen Leistungsstufen jeweils beide Inverter mit entsprechend geringen Effizienzwerten.

### Describe the solution you'd like

Bei dem von mir ebenfalls hochgeschätztem Skript HoymilesZeroExport kann man eine Priorisierung bei mehreren Invertern setzen. Wenn man etwas ähnliches hier realisieren könnte, wäre das aus meiner Sicht eine optimale Ergänzung.

Manos1966, 2025-02-28:
Hallo @HA-IDIPS 

> Derzeit liefern bei **geringen Leistungsstufen** jeweils beide Inverter mit entsprechend **geringen Effizienzwerten**.

Definiere bitte:
- geringe Leistungsstufen
- geringe Effizienzwerten

Zur Info: Hauptgrund der gleichzeitigen Nutzung beide Wechselrichter ist dass so, beide Wechselrichter "Aktiv" (bereit) sind.
Ein ausgeschalteter Wechselrichter braucht bis zu 30 Sekunden um die vorgegebene Leistung zu erreichen. 
Dies wäre "katastrophal" für die schnelle Anpassung variabler Verbraucher wie Wasserkocher, Herd usw.

Fazit: Abhängig wie schwerwiegend die Ineffizienz aufgrund geringen Leistungsstufen zu betrachten ist. Der Vorteil alle Wechselrichter ständig "Online" zu haben ist viel wichtiger.

PS. Die bedeutet selbstverständlich NICHT, dass es keine schlechte Idee wäre, diese Option als Enhancement zu haben. 😉

spcqike, 2025-02-28:
> Ein ausgeschalteter Wechselrichter braucht bis zu 30 Sekunden um die vorgegebene Leistung zu erreichen.

die WR sind jedoch nicht "aus", sie speisen nur nicht ein.

Ich meine, wenn der WR nur gestoppt ist, braucht er keine 30s zum hochregeln. oder?

HA-IDIPS, 2025-02-28:
Geringe Leistungsstufen wäre bei mir alles < 300 Watt. Bis zu 800 W könnte der HM800 ja bringen.

Ich habe gelesen, alles unter 10% Leistung bezogen auf die Wechselrichter Leistung hat sehr schlechten Wirkungsgrad. Bei mir könnte der HM800 über Nacht komplett alles abdecken und der HM1500 könnte "schlafen". Das wäre dann effizienter als wenn der HM1500 zum Beispiel mit weniger als 150 Watt beteiligt ist. Soweit ich das richtig verstehe.

Soweit ich festgestellt habe, werden die Wechselrichter zeitweilig auch in den Standby gesetzt. Sind da in wenigen Sekunden aber anscheinend wieder in Betrieb.

Ich fände es daher trotzdem toll, wenn man mehrere WR priorisieren könnte. Genial, wenn das sogar zeitgesteuert ginge ;-)

Oder weiß jemand, ob man da API-mäßig was machen könnte, also zum Beispiel WR vom DPL ausnehmen (geht ja in der GUI)? Dann könnte ich skriptgesteuert etwas von extern realisieren.

HA-IDIPS, 2025-02-28:
Welche API-Kommandos wären denn für DPL relevant? Ich habe dazu leider nichts gefunden.

AndreasBoehm, 2025-02-28:
Aber pass auf das du die DPL Einstellungen nicht zu oft änderst, sonst schreibst du den flash des ESPs kaputt. Wie oft "okay" ist kann ich dir auswendig aber nicht sagen.

Deine Anfrage wäre hier gut angebracht: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1409

Manos1966, 2025-02-28:
> Ich habe **gelesen**, alles unter 10% Leistung bezogen auf die Wechselrichter Leistung hat sehr schlechten Wirkungsgrad. Bei mir könnte der HM800 über Nacht komplett alles abdecken und der HM1500 könnte "schlafen". 

Deswegen gibt es den Spruch "Glaube nichts von dem, was du hörst, und nur die Hälfte von dem, was du siehst"
Falls ich mich gut erinnere, die HM haben Maximaler Wirkungsgrad: 96,7 % CEC-Wirkungsgrad: 96,5 %

AUSSER wenn du eine 24V Batterie angeschlossen hast, diese Werte sollten deine Wechselrichter erreichen.

Du kannst natürlich das "Minimale Leistungslimit" deines HM-1500 auf 800W stellen, dann wird er erst aktiviert, wenn dein HM-800 auf volle Leistung laueft.

Oben mein HM-800 bei 8,7% (81W), unten der HM-1500 bei 35,5% (500W).
Wirkungsgrad 95% fuer beide. Ich sehe kein "schlechte  Wirkungsleistung" Problem.

[image]

AndreasBoehm, 2025-03-01:
> Du kannst natürlich das "Minimale Leistungslimit" deines HM-1500 auf 800W stellen, dann wird er erst aktiviert, wenn dein HM-800 auf volle Leistung laueft.

Ich bin mir gerade nicht 100% sicher aber ich denke nicht das es so funktioniert.

HA-IDIPS, 2025-03-03:
@Manos1966 OK, das überzeugt mich. Auch der Tipp mit minimalem Leistungslimit scheint zu funktionieren. Danke!

Aber kennt jemand die Web-API Befehle für DPL?

Manos1966, 2025-03-03:
> der Tipp mit minimalem Leistungslimit scheint zu funktionieren. Danke!

Sehr gut, 
ich behaupte trotzdem weiterhin, 
- ob du das "Minimale Leistungslimit" deines HM-1500 auf 800W stellst (= der HM-800 läuft hauptsachlich alleine nachts),
- oder das "Minimale Leistungslimit" deines HM-1500 auf (zum Beispiel) 400W stellst (dann teilen beide Wechselrichter die Last)

es solte keinen Unterschied an den  Maximalen Wirkungsgrad von circa 95% machen.

schlimmchen, 2025-03-06:
> Aber kennt jemand die Web-API Befehle für DPL?

Lass dir von der KI deiner Wahl helfen. Das ist "erweitertes Wissen" und das musst du dir bitte selbst aneignen. Kurz: Du lässt dir die JSON der aktuellen Einstellungen geben, modifizierst die passend, und schickst sie dann wieder ab.

> Auch der Tipp mit minimalem Leistungslimit scheint zu funktionieren. Danke!

Kann ich mir nicht vorstellen. Bei Leisungen bis zu 800W ist der HM800 aktiv und alles ist gut. Zwischen 800W und 1600W Haushaltsverbrauch aber aber ebenfalls **nur** der HM800 aktiv. Code sagt: Meine Einschätzung dürfte stimmen.

closed 2025-04-04 by github-actions[bot]