hoylabs/OpenDTU-OnBattery #1692: VE.direct: generiert manchmal Phantomgeräte

alexz707, 2025-03-01:
### What happened?

Ich habe einen SmartSolar MPPT 150|35 an der openDTU via VE.direct angehängt, Kabel hat RX&TX und wird über einen Levelshifter betrieben.

Prinzipell läuft das Setup sehr gut, jedoch habe ich heute schon das 2. Mal festgestellt, dass die openDTU meint es hängt ein zweiter SmartSolar MPPT 150|35 am System.
Die Seriennummern unterscheiden sich nur bei den letzten beiden Stellen:

HQ2148JY7KC <- den gibts wirklich
HQ2148JY792

Nach einem Neustart der openDTU ist das 2. Gerät wieder verschwunden. Im Home Assistant muss ich ihn dann natürlich selber löschen.

Gestern Nachmittag um ca 15 Uhr erschien der Geister SmartSolar und blieb scheibar bis heute früh, da hat sich die OpenDTU aus irgend einem Grund neu gestartet. (Passierte bis jetzt eigentlich auch so gut wie nie.)

Der ESP bekommt Strom über einen XW-1590-5-18W QC Female USB der auf 3 parallelen Batterien hängt.
Sollte also kein Stromproblem sein. Ich habe auch die BMS's gecheckt, die hatten alle kein Problem.

[image]

### To Reproduce Bug

konnte ich bis jetzt noch nicht herausfinden.

### Expected Behavior

Es sollte nur ein MPPT angezeigt werden.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

hhoebel, 2025-03-03:
Das klingt mir eher wie ein Übersprechen zwischen den UARTs - hast du die beiden nicht genutzten UARTs irgendwo angeschlossen  oder nur den einen Victron? Ich nutze 3 Victrons und habe sowas auch mit weniger angeschlossenen Victrons noch nie beobachtet.
Im Zweifelsfall einfach die unbenutzten UARTs in der Config deaktiveren (-1 statt denen einen GPIO-Pin zuzuordnen).

SW-Niko, 2025-03-03:
Hmmm... das Ganze ist etwas seltsam.

Das Victron-Protokoll hat eine einfache Checksumme. Ein Fehler wird erkannt. Mehrere Fehler können sich gegenseitig aufheben.
Wenn jetzt irgendwo in der Übertragung eine Störung oder Übersprechen passiert, dann müsste der Fehler verteilt über den ganzen Datensatz auftreten und nicht immer die Seriennummer mit der gleichen Änderung treffen.

Seit einiger Zeit werten auch SolarCharger über MQTT unterstützt.

Ich habe 2 Solar Charger und einem Smart Shunt im Einsatz. Ein Phantomgerät ist mir bis jetzt noch nicht aufgefallen.

alexz707, 2025-03-03:
> Das klingt mir eher wie ein Übersprechen zwischen den UARTs - hast du die beiden nicht genutzten UARTs irgendwo angeschlossen oder nur den einen Victron? Ich nutze 3 Victrons und habe sowas auch mit [...]

Hi! Danke für deinen Input, leider sind die UARTs in der Config schon auf -1, das Nebensprechen sollte es also nicht sein.

> Hmmm... das Ganze ist etwas seltsam.
>
> Das Victron-Protokoll hat eine einfache Checksumme. Ein Fehler wird erkannt. Mehrere Fehler können sich gegenseitig aufheben. Wenn jetzt irgendwo in der [...]

Das mit der Checksum dachte ich mir auch schon, deshalb hat mich das schon verwundert.
Beim 1. Mal könnte es Zufall sein, beim 2. Mal würd ich das eher bezweifeln.

Ich beobachte das einfach weiter, sonst funktioniert ja alles ohne Probleme.

SW-Niko, 2025-03-04:
Hallo @alexz707 
ich verfolge gerade eine Spur, die vielleicht zu deinen Beobachtungen führt.
Kann es sein das dein MPPT 150/35 die Daten in 2 Paketen/Frames liefert?
Prüfe das doch bitte mal. Wenn du das Verbose Logging für die Solar Charger aktivierst dann siehst du in welchen Blöcken die Daten reinkommen.
Wenn die Daten sich bei jedem Block wiederholen dann ist es nur ein Frame und wenn sich die Daten jeden zweiten Block wiederholen dann sind es 2 Frames.

alexz707, 2025-03-04:
Hallo @SW-Niko,
also soweit ich das verstanden habe müsste es 1 Frame sein oder?

```Shell
[VE.Direct MPPT 4/5] serial input (316 Bytes):
17:56:55.427 > [VE.Direct MPPT 4/5] 3a 37 44 42 45 44 30 30 35 32 30 44 32 37 0a 3a
17:56:55.437 > [VE.Direct MPPT 4/5] 37 32 37 32 30 30 30 46 46 46 46 46 46 46 46 30
17:56:55.444 > [VE.Direct MPPT 4/5] 42 0a 3a 37 45 43 45 44 30 30 46 46 46 46 37 37
17:56:55.454 > [VE.Direct MPPT 4/5] 0a 3a 37 46 36 45 44 30 30 45 41 31 35 36 43 0a
17:56:55.464 > [VE.Direct MPPT 4/5] 3a 37 46 37 45 44 30 30 34 34 31 36 31 30 0a 3a
17:56:55.474 > [VE.Direct MPPT 4/5] 41 44 35 45 44 30 30 38 38 31 34 45 44 0a 3a 41
17:56:55.485 > [VE.Direct MPPT 4/5] 42 42 45 44 30 30 36 45 31 43 31 39 0a 3a 41 42
17:56:55.491 > [VE.Direct MPPT 4/5] 43 45 44 30 30 41 31 30 30 30 30 30 30 30 31 0a
17:56:55.504 > [VE.Direct MPPT 4/5] 0d 0a 50 49 44 09 30 78 41 30 35 38 0d 0a 46 57
17:56:55.510 > [VE.Direct MPPT 4/5] 09 31 36 34 0d 0a 53 45 52 23 09 48 51 32 31 34
17:56:55.522 > [VE.Direct MPPT 4/5] 38 4a 59 37 4b 43 0d 0a 56 09 35 32 35 36 30 0d
17:56:55.529 > [VE.Direct MPPT 4/5] 0a 49 09 30 0d 0a 56 50 56 09 37 33 31 35 30 0d
17:56:55.538 > [VE.Direct MPPT 4/5] 0a 50 50 56 09 32 0d 0a 43 53 09 33 0d 0a 4d 50
17:56:55.544 > [VE.Direct MPPT 4/5] 50 54 09 32 0d 0a 4f 52 09 30 78 30 30 30 30 30
17:56:55.549 > [VE.Direct MPPT 4/5] 30 30 30 0d 0a 45 52 52 09 30 0d 0a 4c 4f 41 44
17:56:55.560 > [VE.Direct MPPT 4/5] 09 4f 46 46 0d 0a 48 31 39 09 31 35 34 33 38 31
17:56:55.568 > [VE.Direct MPPT 4/5] 0d 0a 48 32 30 09 38 30 37 0d 0a 48 32 31 09 31
17:56:55.574 > [VE.Direct MPPT 4/5] 35 32 38 0d 0a 48 32 32 09 38 34 39 0d 0a 48 32
17:56:55.582 > [VE.Direct MPPT 4/5] 33 09 31 34 35 39 0d 0a 48 53 44 53 09 32 31 38
17:56:55.586 > [VE.Direct MPPT 4/5] 0d 0a 43 68 65 63 6b 73 75 6d 09 9d
17:56:55.591 > [VE.Direct MPPT 4/5] Text Data 'PID' = '0XA058'
17:56:55.599 > [VE.Direct MPPT 4/5] Text Data 'FW' = '164'
17:56:55.606 > [VE.Direct MPPT 4/5] Text Data 'SER' = 'HQ2148JY7KC'
17:56:55.614 > [VE.Direct MPPT 4/5] Text Data 'V' = '52560'
17:56:55.622 > [VE.Direct MPPT 4/5] Text Data 'I' = '0'
17:56:55.625 > [VE.Direct MPPT 4/5] Text Data 'VPV' = '73150'
17:56:55.633 > [VE.Direct MPPT 4/5] Text Data 'PPV' = '2'
17:56:55.641 > [VE.Direct MPPT 4/5] Text Data 'CS' = '3'
17:56:55.647 > [VE.Direct MPPT 4/5] Text Data 'MPPT' = '2'
17:56:55.655 > [VE.Direct MPPT 4/5] Text Data 'OR' = '0X00000000'
17:56:55.662 > [VE.Direct MPPT 4/5] Text Data 'ERR' = '0'
17:56:55.667 > [VE.Direct MPPT 4/5] Text Data 'LOAD' = 'OFF'
17:56:55.676 > [VE.Direct MPPT 4/5] Text Data 'H19' = '154381'
17:56:55.680 > [VE.Direct MPPT 4/5] Text Data 'H20' = '807'
17:56:55.689 > [VE.Direct MPPT 4/5] Text Data 'H21' = '1528'
17:56:55.695 > [VE.Direct MPPT 4/5] Text Data 'H22' = '849'
17:56:55.704 > [VE.Direct MPPT 4/5] Text Data 'H23' = '1459'
17:56:55.708 > [VE.Direct MPPT 4/5] Text Data 'HSDS' = '218'
17:56:55.714 > [VE.Direct MPPT 4/5] Sending Hex Command: :727200007, Free FIFO-Buffer: 118
17:56:55.721 > [VE.Direct MPPT 4/5] Hex Data: Network total DC power value indicates non-networked controller
17:56:55.729 > Last missing
17:56:55.736 > Request retransmit: 3
17:56:55.799 > Success
17:56:55.808 > [DPL] waiting for sufficiently recent power meter reading
17:56:56.032 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0xEDBC (Panel Power), value: 0x000000D3, flags: 0x00
17:56:56.040 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0xEDD5 (Charger Voltage), value: 0x00001487, flags: 0x00
17:56:56.047 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0xEDBB (Panel Voltage), value: 0x00001C5F, flags: 0x00
17:56:56.235 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0x0200 (Device Mode), value: 0x00000001, flags: 0x00
17:56:56.244 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0x0202 (Remote Control Used), value: 0x00000002, flags: 0x00
17:56:56.252 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0x0201 (Device State), value: 0x00000003, flags: 0x00
17:56:56.262 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0xEDD7 (Charger Current), value: 0x00000000, flags: 0x00
17:56:56.418 > [VE.Direct MPPT 4/5] serial input (316 Bytes):
17:56:56.423 > [VE.Direct MPPT 4/5] 3a 37 32 37 32 30 30 30 46 46 46 46 46 46 46 46
17:56:56.432 > [VE.Direct MPPT 4/5] 30 42 0a 3a 41 42 43 45 44 30 30 44 33 30 30 30
17:56:56.439 > [VE.Direct MPPT 4/5] 30 30 30 43 46 0a 3a 41 44 35 45 44 30 30 38 37
17:56:56.446 > [VE.Direct MPPT 4/5] 31 34 45 45 0a 3a 41 42 42 45 44 30 30 35 46 31
17:56:56.451 > [VE.Direct MPPT 4/5] 43 32 38 0a 3a 41 30 30 30 32 30 30 30 31 34 38
17:56:56.459 > [VE.Direct MPPT 4/5] 0a 3a 41 30 32 30 32 30 30 30 32 30 30 30 30 30
17:56:56.465 > [VE.Direct MPPT 4/5] 30 34 35 0a 3a 41 30 31 30 32 30 30 30 33 34 35
17:56:56.471 > [VE.Direct MPPT 4/5] 0a 3a 41 44 37 45 44 30 30 30 30 30 30 38 37 0a
17:56:56.477 > [VE.Direct MPPT 4/5] 0d 0a 50 49 44 09 30 78 41 30 35 38 0d 0a 46 57
17:56:56.487 > [VE.Direct MPPT 4/5] 09 31 36 34 0d 0a 53 45 52 23 09 48 51 32 31 34
17:56:56.496 > [VE.Direct MPPT 4/5] 38 4a 59 37 4b 43 0d 0a 56 09 35 32 35 35 30 0d
17:56:56.507 > [VE.Direct MPPT 4/5] 0a 49 09 30 0d 0a 56 50 56 09 37 31 33 35 30 0d
17:56:56.514 > [VE.Direct MPPT 4/5] 0a 50 50 56 09 33 0d 0a 43 53 09 33 0d 0a 4d 50
[... 412 lines cut ...]
17:57:05.579 > [VE.Direct MPPT 4/5] Text Data 'SER' = 'HQ2148JY7KC'
17:57:05.589 > [VE.Direct MPPT 4/5] Text Data 'V' = '52540'
[... cut ...]
17:57:06.053 > [VE.Direct MPPT 4/5] Unhandled Hex Async Response, addr: 0xEDBB (Panel Voltage), value: 0x00001C8C, flags: 0x00
17:57:06.411 > [VE.Direct MPPT 4/5] serial input (237 Bytes):
17:57:06.421 > [VE.Direct MPPT 4/5] 3a 41 42 43 45 44 30 30 42 43 30 30 30 30 30 30
17:57:06.430 > [VE.Direct MPPT 4/5] 45 36 0a 3a 41 44 35 45 44 30 30 38 37 31 34 45
17:57:06.439 > [VE.Direct MPPT 4/5] 45 0a 3a 41 42 42 45 44 30 30 38 43 31 43 46 42
17:57:06.451 > [VE.Direct MPPT 4/5] 0a 0d 0a 50 49 44 09 30 78 41 30 35 38 0d 0a 46
```

SW-Niko, 2025-03-04:
>also soweit ich das verstanden habe müsste es 1 Frame sein oder?

Ja, richtig. 👍

Hmmmm... Kannst du dir das nochmals anschauen wenn viel Leistung über den MPPT kommt? Ich habe da deine Aussage im Kopf.

>Gestern Nachmittag um ca 15 Uhr erschien der Geister SmartSolar

Wir wissen nur das einige MPPT mit 2 Frames senden, aber nicht warum. Vielleicht wenn die Anzahl der Bytes eine definierte Grenze überschreiten? Und das könnte sein wenn der Strom (I), die Panel Spannung und weitere Daten mehr Bytes benötigen.

Sorry wenn ich dir so viel Arbeit mache, aber ich fische noch ziemlich im trüben. 🤔

alexz707, 2025-03-05:
> Ja, richtig. 👍
>
> Hmmmm... Kannst du dir das nochmals anschauen wenn viel Leistung über den MPPT kommt? Ich habe da deine Aussage im Kopf.
>
>
> Wir wissen nur das einige MPPT mit 2 Frames senden, aber [...]

Alles klar, werde ich hoffentlich am Wochenende beobachten können sofern das Wetter passt.
Ganz im Gegenteil, ich bin froh wenn jemand Ideen hat und mir helfen kann.
Sofern es ein Bug wäre, würde es ja anderen auch helfen!
Also vielen vielen Dank für deine Hilfe :-)

schlimmchen, 2025-03-06:
> Sofern es ein Bug wäre, würde es ja anderen auch helfen!

Gerne lass ich mich überzeugen, aber das kann ich mir nicht so recht vorstellen.

Plausibel ist für mich, dass es bei Übertragungsfehlern, die die sehr schwache Prüfsumme nicht entdeckt, ein zweiter Laderegler im Live View und im HASS auftaucht.

Übertragungsfehler sind aber in der Hardware abzustellen bzw. so gering zu halten, dass die Prüfsumme helfen kann, Fehler zu entdecken.

Welchen Level-Shifter hast du im Einsatz? Ein ADuM1201, also ein digitaler Isolator, den wir empfehlen und der auf dem CAN/Iso Shield des OpenDTU Fusion verbaut ist, ist kein Level-Shifter, after deutlich robuster als einer. Welche Kabel hast du verwendet und wie lang sind die?

Andrix82, 2025-03-20:
Ich habe das auch schon ein paar mal (~ alle paar Tage) gehabt:

[image]

Dabei ist die Kommunikation des MPPTs dann garnicht mehr möglich oder nur sehr eingeschränkt.

Der Gesamtwert der MPPTs ist dann plötzlich verdoppelt durch das "Schattengerät"... schlecht für externe Auswertungen

Ich habe dies aber definitiv erst seit einer der letzten ich glaube vorletzte Firmware wenn ich mich recht erinnere...

Davor gabs monatelang kein Problem daher würde ich prinzipiell physikalische Leitungsprobleme eher ausschließen.

Es hilft dann nur ein openDTU Reset.

Unten noch ein Console Auszug mit erweitertem Logging für den Solarladeregler:

Während ich das schreibe dürfte die openDTU mittlerweile selbst in einen SW-Reset gelaufen sein während diesem Zustand und läuft nun wieder unauffällig.

Wenn ich beim nächsten Auftreten noch was hilfreiches bereitstellen soll/kann bitte informieren

[VE.Direct MPPT 43/44] serial input (416 Bytes):
10:09:10.764 > [VE.Direct MPPT 43/44] 3a 41 30 32 30 32 30 30 30 32 30 30 30 30 30 30
10:09:10.968 > [VE.Direct MPPT 43/44] 34 35 0a 3a 41 30 30 30 32 30 30 30 31 34 38 0a
10:09:11.088 > [VE.Direct MPPT 43/44] 0d 0a 45 52 52 09 30 0d 0a 4c 4f 41 44 09 4f 46
10:09:11.383 > [VE.Direct MPPT 43/44] 46 0d 0a 52 65 6c 61 79 09 4f 46 46 0d 0a 48 31
10:09:11.497 > [VE.Direct MPPT 43/44] 39 09 31 38 33 35 30 38 0d 0a 48 32 30 09 34 33
10:09:11.684 > [VE.Direct MPPT 43/44] 37 0d 0a 48 32 31 09 33 33 39 30 0d 0a 48 32 32
10:09:11.807 > [VE.Direct MPPT 43/44] 09 31 35 31 34 0d 0a 48 32 33 09 33 39 38 32 0d
10:09:11.993 > [VE.Direct MPPT 43/44] 0a ec 2a d4 55 68 8a a2 ae 48 68 0a ab 6b ad cd
10:09:12.172 > [VE.Direct MPPT 43/44] d5 b5 25 30 3a 41 32 30 30 31 30 30 41 44 44 41
10:09:12.404 > [VE.Direct MPPT 43/44] 43 33 30 30 45 30 0a 3a 41 44 37 45 44 30 30 36
10:09:12.529 > [VE.Direct MPPT 43/44] 34 30 32 32 31 0a 3a 41 38 46 45 44 30 30 36 34
10:09:12.709 > [VE.Direct MPPT 43/44] 30 32 36 39 0a 3a 41 38 45 45 44 30 30 46 34 30
10:09:12.917 > [VE.Direct MPPT 43/44] 43 44 30 0a 3a 41 44 42 45 44 30 30 34 32 31 33
10:09:13.079 > [VE.Direct MPPT 43/44] 32 45 0a 3a 41 42 43 45 44 30 30 39 32 32 39 30
10:09:13.328 > [VE.Direct MPPT 43/44] 35 30 30 45 32 0a 3a 41 42 42 45 44 30 30 31 43
10:09:13.632 > [VE.Direct MPPT 43/44] 34 42 33 43 0a 3a 41 42 34 45 44 30 30 32 36 32
10:09:13.840 > [VE.Direct MPPT 43/44] 38 30 35 30 30 35 37 0a 3a 41 42 35 45 54 30 30
10:09:14.041 > [VE.Direct MPPT 43/44] 39 37 34 41 43 38 0a 3a 41 42 44 45 44 30 30 42
10:09:14.155 > [VE.Direct MPPT 43/44] 30 30 30 46 31 0a 0d 0a 50 49 44 09 30 78 41 31
10:09:14.352 > [VE.Direct MPPT 43/44] 31 34 0d 0a 46 57 45 09 33 31 36 46 46 0d 0a 53
10:09:14.552 > [VE.Direct MPPT 43/44] 45 52 23 09 48 51 32 33 32 30 57 43 59 37 51 0d
10:09:14.759 > [VE.Direct MPPT 43/44] 0a 56 09 35 34 32 30 30 0d 0a 49 09 36 31 32 30
10:09:14.964 > [VE.Direct MPPT 43/44] 30 0d 0a 56 50 56 09 31 39 32 31 35 30 0d 0a 50
10:09:15.056 > [VE.Direct MPPT 43/44] 50 56 09 33 73 38 33 0d 0a 4d 50 50 54 09 32 0d
10:09:15.164 > [VE.Direct MPPT 43/44] 0a 43 53 09 33 0d 0a 4f 52 09 30 78 30 30 30 30
10:09:15.509 > [VE.Direct MPPT 43/44] 30 30 30 30 0d 0a 43 68 65 63 6b 73 75 6d 09 94
10:09:15.680 > [VE.Direct MPPT 43/44] checksum 0x94 != 0x00, invalid frame
10:09:15.888 > Websocket: [/livedata][98] disconnect
10:09:16.020 > Websocket: [/huaweilivedata][94] disconnect
10:09:16.068 > [VE.Direct MPPT 43/44] Sending Hex Command: :727200007, Free FIFO-Buffer: 118
10:09:16.330 > [VE.Direct MPPT 43/44] Hex Data: Network total DC power value indicates non-networked controller
10:09:16.498 > [VE.Direct MPPT 43/44] Sending Hex Command: :7F7ED006A, Free FIFO-Buffer: 118
10:09:16.711 > [VE.Direct MPPT 43/44] Hex Data: MPPT Absorption Voltage (0xEDF7): 56.60V
10:09:16.915 > [VE.Direct MPPT 43/44] serial input (135 Bytes):
10:09:16.964 > [VE.Direct MPPT 43/44] 3a 37 32 37 32 30 30 30 46 46 46 46 46 46 46 46
10:09:17.166 > [VE.Direct MPPT 43/44] 30 42 0a 3a 37 46 37 45 44 30 30 31 43 31 36 33
10:09:17.421 > [VE.Direct MPPT 43/44] 38 0a 0d 0a 45 52 52 09 30 0d 0a 4c 4f 41 44 09
10:09:17.633 > [VE.Direct MPPT 43/44] 4f 46 46 0d 0a 52 65 6c 61 79 09 4f 46 46 0d 0a
10:09:17.836 > [VE.Direct MPPT 43/44] 48 31 39 09 31 38 33 35 30 38 0d 0a 48 32 30 09
10:09:18.037 > [VE.Direct MPPT 43/44] 34 33 37 0d 0a 48 32 31 09 33 33 39 30 0d 0a 48
10:09:18.382 > [VE.Direct MPPT 43/44] 32 32 09 31 35 31 34 0d 0a 48 32 33 09 33 39 38
10:09:18.549 > [VE.Direct MPPT 43/44] 32 0d 0a 48 53 44 53 09 33 31 34 0d 0a 43 68 65
10:09:18.750 > [VE.Direct MPPT 43/44] 63 6b 73 75 6d 09 30
10:09:18.959 > [VE.Direct MPPT 43/44] Text Data 'ERR' = '0'
10:09:19.167 > [VE.Direct MPPT 43/44] Text Data 'LOAD' = 'OFF'
10:09:19.470 > [VE.Direct MPPT 43/44] Text Data 'RELAY' = 'OFF'
10:09:19.675 > [VE.Direct MPPT 43/44] Text Data 'H19' = '183508'
10:09:19.879 > [VE.Direct MPPT 43/44] Text Data 'H20' = '437'
10:09:20.091 > [VE.Direct MPPT 43/44] Text Data 'H21' = '3390'
[... 45 lines cut ...]
10:09:29.509 > Queue size - NRF: 0 CMT: 3
10:09:29.707 > [VE.Direct MPPT 43/44] failed to disassemble the hex message: :A8DED002B1591

10:09:29.915 > [VE.Direct MPPT 43/44] failed to disassemble the hex message: :A8DED002B1591
ERR	0

10:09:30.017 > [VE.Direct MPPT 43/44] failed to disassemble the hex message: :A8DED002B1591
ERR	0
LOAD	OFF

10:09:40.418 > [VE.Direct MPPT 43/44] Unhandled Hex Async Response, addr: 0xEDBD (???), value: 0x000000B1, flags: 0x00
10:09:40.591 > [PowerMeterHttpJson] New total: 1738.42
[... cut ...]
10:09:42.201 > [VE.Direct MPPT 43/44] 44 42 45 44 30 30 35 34 31 33 31 43 0a 3a 41 42
10:09:42.487 > [VE.Direct MPPT 43/44] 43 45 44 30 30 46 45 32 44 30 35 30 30 37 32 0a
10:09:42.594 > [VE.Direct MPPT 43/44] 3a 41 42 42 45 44 30 30 36 39 34 41 46 30 0a 3a
10:09:42.824 > [VE.Direct MPPT 43/44] 41 42 34 45 44 30 30 42 44 32 46 30 35 30 30 42
10:09:43.019 > [VE.Direct MPPT 43/44] 39 0a 3a 41 42 35 45 44 30 30 37 44 34 41 45 32
10:09:43.158 > [VE.Direct MPPT 43/44] 0a 3a 41 42 44 45 44 30 30 42 31 30 30 46 30 0a

SW-Niko, 2025-03-20:
Hallo @Andrix82,
vielen Dank für die Meldung. Wie schaut deine Verbindung zu den MPPTs aus? Länge der Leitung? Galvanisch getrennt?
Einzelne Adern? Zusammen mit anderen Leitungen in einem Kabelbaum?

Ich habe mittlerweile eine Idee woran das liegen könnte und warum das bei einigen Systemen seit kurzem ab und zu auftaucht.

@AndreasBoehm: Mit der Umstellung auf die  "Solar Charger" Klasse wurde die Seriennummer als Unterscheidungsmerkmal eingeführt. Und wenn einmal eine neue Seriennummer dazukommt dann bleibt das Gerät (Phantomgerät)  bis zu nächsten Neustart.
Bei Systemen mit vielen Übertragungsfehlern auf der VE-Direkt Schnittstelle werden dann ab und zu Phantomgeräte erzeugt.
Ich denke das war mit älteren Versionen nicht anders aber da sind die Phantomgeräte entweder schnell wieder verschwunden oder wurden gar nicht angezeigt. Darum ist das bis jetzt nicht negativ aufgefallen.

Hmmm. ... ich schau mir das am WE mal genauer an.

schlimmchen, 2025-03-20:
> Ich habe dies Problem aber definitiv erst seit einer der letzten ich glaube vorletzte Firmware wenn ich mich recht erinnere...

> Davor gabs monatelang kein derartiges Problem daher würde ich prinzipielle physikalische Leitungsprobleme eher ausschließen.

Und ich behaupte das Gegenteil. Ich sehe nicht, dass die Software, geschweige denn eine Änderung an der Software, solche punktuellen Veränderungen an Daten hervorruft.

SW-Niko, 2025-03-21:
Hallo @Andrix82,

>Verbindung ist simpler 2 adriger nicht verdrillter "Klingeldraht".
Einzelne direkte Verbindung. Konfigurierte Pins ESP32S3 43+44
Zusätzlich Masseverbindung von openDTU zu BATT Minus 

Genau hier liegt vermutlich dein Problem.
- Nicht galvanisch getrennt
- Masseleitung nicht mit Rx, Tx verdrillt
- Länge ?
- Parallel zu anderen Leitungen mit hohen Strömen?

Ich würde das anders machen. Durch deine störanfällige Verbindung fängst du dir anscheinend viele Übertragungsfehler ein.
Das Phantomgerät ist vermutlich nur die Spitze vom Eisberg.
Mein Tipp: Verdrille die Masseleitung mit Tx und Rx. Halte Abstand von anderen Leitungen mit hohen Strömen und denke denke über eine galvanische Trennung nach.

>[VE.Direct MPPT 43/44] checksum 0x94 != 0x00, invalid frame
Die SW-Reaktion auf einzelne Invalid Frames ist evt. noch nicht wirklich robust...

Diesen Übertragungsfehler hat die SW erkannt und deshalb wurden die Daten auch verworfen. **Die SW ist an dieser Stelle sehr robust.**

Lass es mich mal so erklären....
Victron liefert eine einfache Checksumme um Fehler im Datenpaket zu erkennen.
Wenn in dem Datenpaket 2 oder mehr Fehler sind und die sich gegenseitig neutralisieren dann können wir das nicht erkennen.
Und wenn dann einer dieser (mehrfachen) Übertragungsfehler in dem Datenpaket mit ca 200 byte auch noch genau die Seriennummer treffen dann wird ein Phantomgerät erzeugt.

Du siehst es sind sehr viele "wenn" nötig. Und das sagt mir ... bei dir treten sehr viele Übertagungsfehler auf.
Die sind bis jetzt nur nicht aufgefallen, weil nach 1 Sekunde neue Daten ankommen und die fehlerhafte Daten wieder überschreiben.

ranma, 2025-03-23:
> Diesen Übertragungsfehler hat die SW erkannt und deshalb wurden die Daten auch verworfen. Die SW ist an dieser Stelle sehr robust.

Naja, wegen der einfachen Checksumme halt nicht so super robust. Wenn man Glück hat und es nur einen Einzelbitfehler gab wird der Erkannt, ansonsten kann man schnell pech haben. Das kann auch dazu führen das die MQTT-Verbindung instabil wird wenn der korrupte Name nicht-ascii Zeichen enthält und kein valides utf-8 ist, siehe https://github.com/hoylabs/OpenDTU-OnBattery/issues/1755

SW-Niko, 2025-03-23:
>Naja, wegen der einfachen Checksumme halt nicht so super robust. 

Richtig, aber das ist ja von Victron vorgegeben und ich denke wir können Victron nicht überreden hier was besseres zu liefern.

>Das kann auch dazu führen das die MQTT-Verbindung instabil wird wenn der korrupte Name nicht-ascii Zeichen enthält und kein valides utf-8 ist

Ja, das würde die Sicherheit nochmal erhöhen. Auf meinem System läuft das im Prinzip seit ein paar Wochen.
Ich habe das aber nicht forciert, weil es auch Bitfehler gibt die das Zeichen nicht aus dem ASCII  Bereich schieben.

Ich würde das Problem lieber an der Wurzel packen und ähnlich wie die "Radio Statistics" eine "VE.Direct Statistics" zur Verfügung stellen. Damit könnte man dann störanfällige Verbindungen leicht erkennen.

Hmmm....  Vielleicht auch mehrere Sachen:

- VE.Direct Statistics"
- ASCII Bereichs Überprüfung
- Phantomgeräte nach Ablauf eine Zeit x wieder automatisch entfernen

ranma, 2025-03-24:
> * Phantomgeräte nach Ablauf eine Zeit x wieder automatisch entfernen

Eine möglichkeit wäre auch den Gerätenamen pro Eingang auf den erwarteten Namen in der Konfiguration festzunageln und dann halt Daten von einem unerwartetem Gerät nicht zu verwenden.

schlimmchen, 2025-03-25:
Danke für diese Idee, ranma, die ist valide, aber ich mag sie nicht. Mehr Einstellungen => mehr schlechter :wink:

* Die Checksumme ist schwach. Damit müssen wir leben.
* Dein PR filtert nun ganze Frames, wenn diese wegen der schwachen Checksumme überleben würden, aber dennoch offensichtlich korrupt sind, weil sie unsinnige Zeichen enthalten. Das ist gut.

Allerdings ist es ja so, dass die Checksumme, auch wenn sie schwach ist, völlig ausreichend ist einen einzelnen Bitfehler zu erkennen in einem Frame. Und dass es mehr als einen Bitfehler gibt pro Frame gibt deutet meiner Meinung nach darauf hin, dass etwas mit der Hardware nicht stimmt.

Wieder andererseits ist mir aber auch schon aufgefallen, dass ich unter `solar/victron` schon Seriennummern gesehen habe im MQTT Explorer, die da nicht hingehören. Ob das aber mein Produktivsystem war oder nur das Test-System, weiß ich nicht sicher. Wenn das in meinem Setup auch passiert, trotz kurzer Leitungen und digitalen Isolatoren, dann würde ich mich überzeugen lassen, noch mehr zu tun. Aber eigentlich wäre ich der Meinung, dass der "Schutz" vor Übertragungsfehler nun ausreichen müsste, denn auch die Hardware muss passen. Das dürfen wir erwarten.

ranma, 2025-03-25:
> Wieder andererseits ist mir aber auch schon aufgefallen, dass ich unter `solar/victron` schon Seriennummern gesehen habe im MQTT Explorer, die da nicht hingehören. Ob das aber mein Produktivsystem war oder nur das Test-System, weiß ich nicht sicher.

Ich sehe die Bitfehler auch vom Ursprünglichen MPPT (Victron SmartSolar MPPT 100|20 48V), trotz Isolator und relativ kurzem Kabel, aber dort um Größenordnungen seltener.

Andrix82, 2025-03-25:
> Ich glaube die Phantom-Geräte führen auch zu einer Überschätzung der Solarproduktion und die DPL funktioniert dann nicht mehr zuverlässig.

Die Leistung zum Zeitpunkt der Entstehung des Phantomgeräts bleibt einfach als permanenter Fehler-Offest auf der MPPT Gesamtleistung bis zum nächsten Reset. Dem DPL im Kontext der Nulleinspeisungsregelung war das gleich, aber in Verbindung mit FullSolar Passthrough wirds sicher Probleme geben!

SW-Niko, 2025-03-25:
>Ich sehe die Bitfehler auch vom Ursprünglichen MPPT (Victron SmartSolar MPPT 100|20 48V), trotz Isolator und relativ kurzem Kabel, aber dort um Größenordnungen seltener.

Ich kann das bei meinem System sehr genau spezifizieren.

- 3 VE.Direkt Verbindungen (Nicht geschirmte Leitung)
- 2 Verbindungen zu 2 MPPTs sind ca 80cm lang und eine Verbindung zum Smart Shunt ca. 140cm
- Alle 3 Verbindungen sind galvanisch getrennt
- Möglichst größer Abstand zu allen Leitungen mit hohen Strömen
- Bei den beiden MPPTs habe ich im Schnitt einen Übertragungsfehler pro Tag.
- Bei dem Smart Shunt ist mir nach ca 14 Tagen Überwachung kein einziger Übertragungsfehler  aufgefallen

Ein Phantom Gerät hatte ich bis jetzt noch nicht.

Ich hätte eine Art "VE.Direct Übertragungsfehler Statistik" die man bei den VE.Direct Geräten in der UI anzeigen könnte.
Damit könnte jeder der eine VE.Direkt Schnittstelle hat, überprüfen wie gut sie funktioniert.

>Ich glaube die Phantom-Geräte führen auch zu einer Überschätzung der Solarproduktion und die DPL funktioniert dann nicht mehr zuverlässig.

Das sollte aber nach 30 Sekunden wieder vorbei sein weil der Time stamp für die Daten aus dem Rahmen fällt. 🤔  

>Die Leistung zum Zeitpunkt der Entstehung des Phantomgeräts bleibt einfach als permanenter Fehler-Offest auf der MPPT Gesamtleistung bis zum nächsten Reset. Dem DPL im Kontext der Nulleinspeisungsregelung war das gleich, aber in Verbindung mit FullSolar Passthrough wirds sicher Probleme geben!

Ok, Wenn das permanent bleibt dann haben wir noch eine weitere Baustelle.

AndreasBoehm, 2025-03-25:
> Das sollte aber nach 30 Sekunden wieder vorbei sein weil der Time stamp für die Daten aus dem Rahmen fällt. 🤔

Die Daten werden sogar nach 10 Sekunden schon nicht mehr verwendet wenn ich es im Code richtig gelesen habe.

AndreasBoehm, 2025-03-25:
> die Phantomgeräte werden nie aufgeräumt auch deren Stats bleiben für immer aktiv und werden genutzt

Ich kümmere mich heute darum das wir keine alten Stats nutzen.

ranma, 2025-03-25:
Was in diese Richtung sollte helfen: https://github.com/ranma/OpenDTU-OnBattery/commit/1463d6f033313cd1ad8a2b72dfd4eb7b4fc84649

ranma, 2025-03-25:
> Ich hätte eine Art "VE.Direct Übertragungsfehler Statistik" die man bei den VE.Direct Geräten in der UI anzeigen könnte.

Das wäre super, ja :)

AndreasBoehm, 2025-03-25:
@ranma dann stell bitte einen PR, müssen ja nicht zwei Leute an gleichen Thema arbeiten.

AndreasBoehm, 2025-03-25:
Sorry, das klang etwas schroff. Willst du einen PR stellen oder soll ich deinen Code in meine branch übernehmen?

SW-Niko, 2025-03-25:
Hey, super. Ihr seit ja fix.
Dann steuere ich noch eine "VE.Direct Übertragungsfehler Statistik" bei.
Aber .... so schnell bin ich nicht.

>EDIT: Ich hab falsch geschaut, die Phantomgeräte werden nie aufgeräumt auch deren Stats bleiben für immer aktiv und werden genutzt. Das ist ein Problem...

Ich wollte schon testen was passiert wenn ich von einem Gerät im laufenden Betrieb die VE.Direkt Verbindung trenne. 
Das kann ich mir vermutlich jetzt sparen. 👍

ranma, 2025-03-26:
@AndreasBoehm @schlimmchen 

Nochmal ganz andere Idee: 

Das Problem ist ja, das die map-Keys die Seriennummern sind, die aus der Datenübertragung kommen und fehlerbehaftet sein können. Warum ist das eigentlich der map-Key hier? Man könnte ja auch die Anschluss-ID nehmen, z.B. wie in den Logs "MPPT 1/43" oder "MPPT 44/-1", also "MPPT rxGPIO/txGPIO", was die Geräte ja auch eindeutig identifizert. Dann hat man das Problem mit dauerhaften Phantomgeräten in der Map gar nicht.

AndreasBoehm, 2025-03-26:
@ranma Die Idee finde ich super! Der Anschlusss kann sich auch nur ändern wenn man anschließend einen Neustart macht, dann wird eh alles aufgeräumt.

AndreasBoehm, 2025-03-26:
Ich hab den parameter schon mal vorsorglich von `serial` zu `key` umbenannt: https://github.com/hoylabs/OpenDTU-OnBattery/pull/1777/files#diff-d5574532619617945cb0b89ab00d3ed1c34cffefe74344f1d83513e6e00fa799R27

AndreasBoehm, 2025-03-26:
Nachtrag: Ich hab einfach direkt die LogId als key genutzt, war nur eine Zeile die ich ändern musste.
https://github.com/hoylabs/OpenDTU-OnBattery/pull/1777/files#diff-e85fa072517afe44a36f0508ca68c601cbb632808cfecca7a526bc82dca061a3R73

schlimmchen, 2025-03-26:
> Warum ist das eigentlich der map-Key hier? Man könnte ja auch die Anschluss-ID nehmen

Das ist ein hervorragende Idee, @ranma, gerne mehr davon :heart: 😉

ranma, 2025-03-26:
> Das ist ein hervorragende Idee, [@ranma](https://github.com/ranma), gerne mehr davon

Was eine Nacht drüber schlafen manchmal so bringt :)

schlimmchen, 2025-03-29:
Für mich ist das erledigt mit #1777.

Wenn so ein Phantomgerät mal auftritt, wird es mit hoher Wahrscheinlichkeit nach nur einer Sekunde wieder weggebügelt. Mit einem robusten Hardware-Setup *sollte* (tut es scheinbar in der Praxis dennoch) das Problem gar nicht erst auftreten. Eine zusätzliche Plausibilitätsprüfung der Daten ist äußerst aufwendig (unentdeckte Fehler können auch in numerischen Werten versteckt sein, die sich nur schwer auf Fehler prüfen lassen), weshalb ich es ablehne, in die Richtung Code und Rechenzeit zu investieren.

Habe ich etwas übersehen? Dann gerne wieder aufmachen und erläutern.

closed 2025-03-29 by schlimmchen