hoylabs/OpenDTU-OnBattery #1703: Solarladeregler Gesamtertrag Heute - falsche Werte

DerHavey1994, 2025-03-06:
### What happened?

Ich habe 3 Victron mit VE-Direct angeschlossen und beim zweiten Victron wird der Tagesertrag und die Tages Höchstleistung nicht zurückgesetzt.

[image]

### To Reproduce Bug

/

### Expected Behavior

/

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

SW-Niko, 2025-03-06:
Hallo @DerHavey1994,
Waren die Solarregler vor kurzem mal ohne Versorgungsspannung?
Die Regler haben keine RTC (Real Time Clock) und benötigen etwas Zeit um Anhand der Sonnenaufgänge und der Sonnenuntergänge die Mitternacht zu ermitteln.
Das kann man auch einstellen, ab welcher Panelspannung die Nacht bzw der Tag beginnt.

SW-Niko, 2025-03-06:
>tatsächlich vor zwei Wochen ca.

So lange sollte es nicht dauern. 🤔 

Schau einfach mal in die Victron App. OpenDTU-OnBattery und die Victron App sollten bezüglich dieser beiden Werte immer im Gleichschritt sein. Wenn das so ist dann kannst du die Schwellenwerte in der App noch anpassen. Das ist zum Beispiel nötig wenn Nachts zu viel künstliches Licht vorhanden ist. Der MPPT tut sich dann schwer mit dem Erkennen von Sonnen Auf- und Untergang.

DerHavey1994, 2025-03-06:
Seit 5 Tagen ist in der Victron App der Wert so hoch.

Habe jetzt die Vorgängerversion 2025.01.23 geflasht. Jetzt sinds bei dem Laderegler nur noch 50 Wh heute. Als wenn OpenDTU über den TX Port den Laderegler auf 0 setzt?! Quasi wie der Neustart des Wechselrichters zu einer geplanten Uhrzeit.

schlimmchen, 2025-03-06:
> Seit 5 Tagen ist in der Victron App der Wert so hoch.

Dann ist das kein OpenDTU-OnBattery Problem, dann kommt dieser Wert nunmal vom Laderegler.

> Als wenn OpenDTU über den TX Port den Laderegler auf 0 setzt?! Quasi wie der Neustart des Wechselrichters zu einer geplanten Uhrzeit.

Gibt's nicht.

SW-Niko, 2025-03-06:
>Seit 5 Tagen ist in der Victron App der Wert so hoch.

Es ist wie schlimmchen schon sagt. Wenn das in der App schon falsch ist dann können wir auch nichts machen.

Schau dir mal die Konfiguration von deinem MPPT an. Mehr fällt mir momentan auch nicht ein.

[image]

DerHavey1994, 2025-03-08:
Gestern Morgen um 05:00 alles für 5 Minuten stromlos gemacht. Scheint jetzt alles zu funktionieren

Der 150/35 hatte um 5:00 auch noch 50V PV Spannung angezeigt, nachdem PV getrennt war.

Das muss ein Softwarebug gewesen sein.

SW-Niko, 2025-03-08:
Gut das du dein Problem gelöst hast. 👍 

Aber ich muss jetzt nochmal nachfragen nur zur Sicherheit. Du hast nur den MPPT 150/35 stromlos gemacht?
Hintergrund ist das es noch einen anderen Fehlerfall mit "Phantomgeräten" gibt. Siehe: #1692 
Und so ein Fehler könnte auch zu falschen Anzeigen in OpenDTU-OnBattery führen.

Ansonsten vergiss nicht das Issue zu schließen wenn nichts mehr offen ist.

DerHavey1994, 2025-03-08:
Ne hab die ganze Anlage.

- PV getrennt bei allen 3
- Pylontech ausgeschaltet

Wird geschlossen 👍🏻

closed 2025-03-08 by DerHavey1994