hoylabs/OpenDTU-OnBattery #1708: ESP32-S3: Reading temperature leads to watchdog timeout

schlimmchen, 2025-03-07:
### What happened?

More and more often I see backtraces that indicate problems with reading the temperature. The backtraces look very, very similar across the times I have seen them: Both cores are somehow trying to read the temperature and the system locks up. The watchdog then reboots the system.

### To Reproduce Bug

This is an intermittend issue, as it seems. https://github.com/espressif/esp-idf/issues/8088 seems to suggest that this happens while WiFi is not connected?!

### Expected Behavior

Reading the temperature shall not cause intermittend reboots.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

[2025-03-07_aviv_1.txt](https://github.com/user-attachments/files/19128064/2025-03-07_aviv_1.txt)
[2025-03-07_aviv_2.txt](https://github.com/user-attachments/files/19128065/2025-03-07_aviv_2.txt)

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-03-07:
As I have seen this several times now and given that unexpected reboots are a real PITA and ruin the user experience, I am going to disable temperature readings on ESP32-S3, similar to what Thomas did to disable them for ESP32-S2. The temperature reading is interesting, but not worth taunting the watchdog.

It seems this is resolved in espressif-idf 5.2.2, so once that lands in OpenDTU(-OnBattery) through the respective Arduino framework update, we may re-enable temperature readings.

@tbnobody FYI

closed 2025-03-07 by schlimmchen

ranma, 2025-03-09:
Oooh, I wonder if that is the reason for the unexpected reboots on mine (Also an S3)... Strangely this week has been unexpectedly stable, but also it hasn't seen RSSI drop issues recently (maybe the Fritzbox got an update and the mesh is behaving nicer now?).

Sorbat, 2025-03-16:
bei mir tritt das Problem mit den Reboots damit nicht mehr auf  

[image]

ranma, 2025-03-23:
FWIW ich habe jetzt auch mal ein update auf https://github.com/hoylabs/OpenDTU-OnBattery/commit/114c67bec1f956867d94ec7109955128611bc08c gemacht, aber mit der neueren Version habe ich jetzt das problem das die MQTT daten nur unregelmäßig kommen obwohl das WiFi gute RSSI hat. Das korreliert mit größerem Speicherverbrauch und sieht so aus als ob dann nur noch unregelmäßig daten ins MQTT kommen wenn der Speicher niedrig ist...

Im syslog sehe ich dann regelmäßige tcp-disconnects:
```
2025-03-23T13:53:16.534125+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connecting to MQTT...
2025-03-23T13:53:16.840010+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connected to MQTT.
2025-03-23T13:53:17.454126+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnected from MQTT.
2025-03-23T13:53:17.454126+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnect reason:TCP_DISCONNECTED
2025-03-23T13:53:19.098123+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connecting to MQTT...
2025-03-23T13:53:19.552049+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connected to MQTT.
2025-03-23T13:53:19.599050+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnected from MQTT.
2025-03-23T13:53:19.599050+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnect reason:TCP_DISCONNECTED
```

commit cde47996 (2025-03-07): Fix: disable temperature readings for ESP32-S3

reading the temperature is flawed and leads to unexpected reboots, which
we observed more and more often recently. since the user experience is
trash if the module resets unexpectedly from time to time, it seems to
be more important to prevent reboots where ever possible rather than
keeping the temperature readings.

closes #1708.