hoylabs/OpenDTU-OnBattery #1713: Inverter-Gesamtleistung wird falsch berechnet, DLP regelt deswegen falsch

Github-Nutzer, 2025-03-08:
### What happened?

Auslangslage: 3 Hoymiles-Wechselrichter, alle 3 geregelt via DLP. Stromquelle "Solarmodule" bei 2 davon, der 3. ist als "Smart Buffer" definiert da dieser an einem B2500 - Speicher hängt.
Fällt nun ein Wechselrichter weg (Sonne aus, Akku leer) so wird weiterhin die zuletzt gemeldete Ausgangsleitung in die Inverter-Gesamtleistung eingerechnet und in weiterer Folge dann auch durch den DLP berücksichtigt. Mir ist das heute beim Kalibrieren des Akkus aufgefallen weil es da um rd. 150 Watt ging welche zuviel ins Netz eingespeist wurden.
Jetzt am Abend der umgekehrte Weg, die beiden "Solarmodul-WR" werden mit dem letzten Wert in die Gesamtleistung mit eingerechnet obwohl diese schon über 3 Stunden nicht mehr produzieren / erreichbar sind.

[image]

### To Reproduce Bug

In **meiner Konfiguration** am Einfachsten den Speicher ausschalten, WR produziert nicht mehr, letzte Leistung wird aber angerechnet.

### Expected Behavior

Sobald ein Wechselrichter nicht mehr erreichbar ist (definierbare Zeitspanne?) Ausgangsleistung auf 0 stellen.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
21:20:37.987 > [DPL] inverter 111111111111 is behind power meter producing 13 W
21:20:38.034 > [DPL] inverter 222222222222 is behind power meter producing 8 W
21:20:38.080 > [DPL] inverter 333333333333 is behind power meter producing 255 W
21:20:38.129 > [DPL] skipping 111111111111 as it is not reachable
21:20:38.175 > [DPL] skipping 222222222222 as it is not reachable
21:20:38.226 > [DPL] requesting 2000 W from 1 smart-buffer-powered inverter currently producing 255 W (diff 1745 W, hysteresis 25 W)
21:20:38.286 > [DPL inverter 333333333333]:
21:20:38.340 >     expected AC power per MPPT 99 W
21:20:38.481 >     MPPT-a AC power 128 W
21:20:38.584 >     MPPT-b AC power 128 W
21:20:38.634 > [DPL] will cover 250 W using 1 smart-buffer-powered inverter
21:20:38.696 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 1750 W AC
21:20:38.749 > [DPL inverter 111111111111]:
21:20:38.877 >     solar-powered, producing 13 W
21:20:38.934 >     lower/current/upper limit: 200/200/1600 W, output capability: 1600 W
21:20:38.981 >     sending commands enabled, offline, disqualified (unreachable)
21:20:39.031 >     max reduction production/standby: 0/0 W, max increase: 0 W
21:20:39.185 >     target limit/output/state: -1 W (unchanged)/13 W/unchanged, 0 update timeouts
21:20:39.229 >     MPPTs AC power: a: 4 W b: 3 W c: 3 W d: 4 W
21:20:39.283 > [DPL inverter 222222222222]:
21:20:39.341 >     solar-powered, producing 8 W
21:20:39.390 >     lower/current/upper limit: 200/200/1600 W, output capability: 1600 W
21:20:39.435 >     sending commands enabled, offline, disqualified (unreachable)
21:20:39.488 >     max reduction production/standby: 0/0 W, max increase: 0 W
21:20:39.547 >     target limit/output/state: -1 W (unchanged)/8 W/unchanged, 0 update timeouts
21:20:39.593 >     MPPTs AC power: a: 1 W b: 3 W c: 0 W d: 5 W
21:20:39.639 > [DPL inverter 333333333333]:
21:20:39.714 >     smart-buffer-powered, producing 255 W
21:20:39.785 >     lower/current/upper limit: 100/249/250 W, output capability: 800 W
21:20:39.869 >     sending commands enabled, reachable, eligible
21:20:39.925 >     max reduction production/standby: 155/255 W, max increase: 65531 W
21:20:39.971 >     target limit/output/state: 250 W (update)/250 W/production, 0 update timeouts
21:20:40.016 >     MPPTs AC power: a: 128 W b: 128 W
21:20:40.063 > [DPL inverter 333333333333]: sending limit of 31.2 % (250 W respectively), max output is 800 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-03-08:
> Jetzt am Abend der umgekehrte Weg, die beiden "Solarmodul-WR" werden mit dem letzten Wert in die Gesamtleistung mit eingerechnet obwohl diese schon über 3 Stunden nicht mehr produzieren / erreichbar sind.

Ach, das auch noch.... Ja, da müsstest du recht haben.

Ich bin dran und das wird durch den Ansatz, den ich gerade versuche umzusetzen, ebenfalls erschlagen. Ich nutze das Konzept der "Eligibility" eine Schicht weiter hoch und es fliegen dann alle WR raus aus allen relevanten Berechnungen, nicht nur bei manchen.

Github-Nutzer, 2025-03-08:
Teste gerne :-)

schlimmchen, 2025-03-08:
> Teste gerne :-)

Sehr gerne, morgen dann. Läuft jetzt mal auf meiner Anlage und ich beobachte das mal. Dass Dein Fehler behoben ist, da bin ich mir recht sicher, aber bei den potentiellen Seiteneffekten des angepassten Codes bin ich mir noch nicht sicher...

Github-Nutzer, 2025-03-09:
Ich werf auch einen Blick auf eventuelle Seiteneffekte - versprochen :-)

Harisjems, 2025-03-10:
Ich habe es eben auch installiert, danke @schlimmchen für die fixe Reaktion, unabhängig davon, wie es ausgeht.

Kurze Verständnisfrage dazu! Muss ich den Mindestlimit von 15 % bei HM1500 weiterhin beachten? Oder hat es damit nichts zu tun? Und ist es einfach hardwaregebunden, dass der WR nicht mag, untertourig zu fahren!

schlimmchen, 2025-03-10:
> Muss ich den Mindestlimit von 15 % bei HM1500 weiterhin beachten?

Unsere gemeinsam erarbeitete und erprobte Faustregel ist 13W/Eingang.

> Und ist es einfach hardwaregebunden, dass der WR nicht mag, untertourig zu fahren!

Ja.

spcqike, 2025-03-10:
> 13W/Eingang.

Pro Eingang oder pro MPPT? Ich dachte pro MPPT

schlimmchen, 2025-03-10:
> hab aber nur inverter wo jeder Eingang ein MPPT ist

Dito. Ich weiß es nicht 🤷‍♂

Harisjems, 2025-03-10:
Ich bin mutig, ich lasse HM1500 und HMS1600 4t auf 100 W Mindestleistung stehen, sehe ich dann morgen im Laufe des Tages :-)? War das gut oder nicht :-)?

Github-Nutzer, 2025-03-10:
Soweit ich das beobachten hab können - heute Büro und keine Sonne :-( - hat der Testbuild zumindest mein Problem beseitigt. Ausgefallene WR wurden nicht mehr in die aktuelle Gesamtleistung eingerechnet. Unabhängig davon verfolge ich die engl. Konversation "drüben" wo ja doch einige Punkte diskutiert werden.

Harisjems, 2025-03-11:
Bis jetzt sieht es gut aus bei HM1500 und HMS1600, bei 100 W als Mindestwert.  „Als Dauertest bestanden“ wäre bei wenig Sonne heute übertrieben.

spcqike, 2025-03-11:
@schlimmchen 
Wir haben ja 2 kritische leistungswerte, oder? 
1. die 13W pro Eingang oder MPPT. Wenn ich mich recht erinnere können sich die Hoymiles „aufhängen“ wenn sie darunter betrieben werden und reagieren nicht mehr auf Hoch Regel Befehle, oder? 
2. das Limit, bis zu dem der wechselrichter nicht sauber regelt und daher bisher als verschattet interpretiert wird. So dass er von der DTU keine neuen, höheren Limits bekommen hat. 

1. haben wir, sollte hoymiles nichts an der Firmware geändert haben, noch immer. 

2. ist mit der Anpassung aus dem PR behoben? 

Habe ich das richtig aufgenommen?

AndreasBoehm, 2025-03-11:
> das Limit, bis zu dem der wechselrichter nicht sauber regelt und daher bisher als verschattet interpretiert wird. So dass er von der DTU keine neuen, höheren Limits bekommen hat.

Das Problem besteht weiterhin. Auch mit den Änderungen von schlimmchen.

schlimmchen, 2025-03-11:
Das sehe ich auch so.

Außerdem musste ich mal drüber nachdenken und sehe ein: Im PR habe ich wirklich nur das Problem gelöst, dass unerreichbare Inverter gleichartig behandelt werden in der Gänze der DPL Schleife. Das erlaubt es, die verbleibenden Inverter tatsächlich und wenn es geht auch "korrekt" zu nutzen, um die target consumption (dennoch) zu erreichen.

Es ist falsch, dass ich einen inkrementellen Ansatz implementiert hätte, der anders wäre als zuvor. Unter gleichen Bedinungen (abgesehen von non-eligible WR) kommt immer noch die gleiche Anforderung an die WR heraus. Der Code sieht ganz anders aus und spiegelt eine etwas andere Denkweise wieder, aber die rechnerisch ist es das gleiche.

closed 2025-03-18 by schlimmchen