hoylabs/OpenDTU-OnBattery #1716: DTU get's stuck in bad wifi connection although a better endpoint is available

markusdd, 2025-03-09:
### What happened?

Hi. I've been hunting weird data delivery dropouts with my onBattery setup and I finally found out why:

In a setup where a wifi spans a large area and multiple access points are avilable, the DTU sometimes decides to connect to the Fritzbox main router, which is by far the furthest away from it (frankly it's impressive that it can connect at all to be honest). In the pictures you see the layout of the network. the one it should (and normally will) connect to is the AX repeater. It's the closest aand with that one everything is rock stable. alternatively the other 2 would still work, they are also much closer than the main router.

I assume this happens when one of the repeaters restarts, there as a short power cut or the Fritzbox runs an update and the Mesh is down briefly. Maaybe the DTU also sometimes just decides to switch into this horribly bad connection.
The only way to fix this then is to manually reboot the DTU via the web interface - if I can still reach it. (which is annoying because this setup is like 300km away from me) Otherwise I have to call and tell somebody to power cycle it.

So I believe we'd need some logic to make sure the DTU switches to a better wifi connection once available and does not get stuck with a much more horrible connection to a wwrong endpoint.

[image]
[image]

### To Reproduce Bug

mmh. difficult without a similar setup. but not really needed aalso.

### Expected Behavior

once a better wifi endpoint is avaailable - use it - do not get stuck with a horrible connection

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

pretty much latest - but this exists since basically forever, been having this issue for a year noww

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-10:
https://github.com/hoylabs/OpenDTU-OnBattery/issues/1495 sounds very similar to your issue.

Manos1966, 2025-03-10:
I am afraid I experience the same pain and misery, too
Thanks @markusdd for identifying "probable cause" = Restart of repeater.
I have installed a SHELLY to be able to cut off the power supply of the OpenDTU-OnBattery and restart it in case it loses WiFi connection. After restart, it always connects to the right repeater. 

[image]

spcqike, 2025-03-10:
isn't that a case for [WiFi-Roaming](https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7560/1614_Was-ist-WLAN-Roaming-und-wie-funktioniert-es/) or [Mesh Steering](https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7560/3515_Was-ist-WLAN-Mesh-Steering-und-wie-funktioniert-es/)?

in my case (UniFi) it's connected to the best AP

[image]

and it looks at if it roamed (or was roamed?) once, two weeks ago

[image]

Unfortunately i don't see, if my APs "moved" the DTU or if the DTU changed the AP itself.

markusdd, 2025-03-10:
As far as I know steering WiFi devices requires k/v standard which the ESPs do not have.

I think the fix would not even be that difficult: just rescan from time to time and select the best endpoint

spcqike, 2025-03-10:
> Ältere WLAN-Geräte, die die WLAN-Standards 802.11k und 802.11v nicht unterstützen, kann der Mesh Master nur durch eine kurze Trennung der WLAN-Verbindung in ein anderes Funknetz steuern. Zu einem anderen WLAN-Zugangspunkt kann der Mesh Master diese Geräte nicht steuern.

sounds linke the "Mesh Master" could force a reconnect. somehow.

spcqike, 2025-03-10:
Yes. A client driven roaming would be way better. 
But this steering could be an alternative and a quick workaround, instead of using a Shelly to reboot the DTU. As a reboot also resets its internal states (charge/ discharge cycle state and things)

swingstate, 2025-03-10:
I think what Markus described is what I asked the upstream project to "fix" a year a ago [here](https://github.com/tbnobody/OpenDTU/issues/1354).

I wonder if this is something Andreas/Schlimmchen would be open to look into too.

schlimmchen, 2025-03-10:
No, Markus is asking to keep scanning the Wi-Fi landscape and to re-connect to a different AP if it broadcasts a matching SSID but whose RF quality is significantly better.

No, I am not motivated to implement that. This will open a can of worms that I am not prepared to deal with. First of all: What is "significantly better" such that it warrants re-connecting? Second: How many setups will be break trying to implement this mechanism?

Your link, @swingstate, leads to something else, which we also track in #1495. That is something that I can get behind. It should also solve Markus' problem. @markusdd Can you accept setting a fixed BSSID to connect with in order to solve your issue?

markusdd, 2025-03-10:
I think as long as the logic is such that it does this when available at all and uses another one otherwise is ok.

I am aware that this isn't a very elegant thing to do at all but fact of the matter is the current behavior is even more broken, and skimming through issues I am far from the only one being affected by that.

It should be optional in any case, but I think even the re-scan is not that difficult. You get a couple of endpoints for a given SSID and you take the one with the best rssi indication. Certainly this is better than connecting to an insane -90dB endpoint and then never consider again to look for something better although you have transmission retries for like 50% of your packets.

I do agree though that this is probably an upstream issue to deal with.

It's just much more impactful for on Battery because it interferes with getting energy meter data reliably and hence regulation can be very off, which means potentially wasting real money. That is what I have observed. I was wondering how in winter the battery could be at 70% until I realized the inverter was stuck shut at 0 limit because the DTU wasn't able to read power data.

So this isn't just a statistics collection issue, it fundamentally impacts operation.

schlimmchen, 2025-03-10:
I agree with you. However, I also trust that you understand what I meant by "opening this can of worms". On a high level, this feels trivial. Implementing this is not. And I do see a real danger of breaking stuff for many people while fixing it merely for a few.

> I think as long as the logic is such that it does this when available at all and uses another one otherwise is ok.

That's just asking for re-scanning in a different way :wink: Assume we connected to the forced BSSID, which then goes down for a minute. Now what? Accoding to your "uses another one" request we need to scan and connect to the worse AP as the forced BSSID is gone. Then we are connected to it and... we stay connected unless we implement re-scanning.

markusdd, 2025-03-10:
> I agree with you. However, I also trust that you understand what I meant by "opening this can of worms". On a high level, this feels trivial. Implementing this is not. And I do see a real danger of breaking stuff for many people while fixing it merely for a few.
> 
> > I think as long as the logic is such that it does this when available at all and uses another one otherwise is ok.
> 
> That's just asking for re-scanning in a different way 😉 Assume we connected to the forced BSSID, which then goes down for a minute. Now what? Accoding to your "uses another one" request we need to scan and connect to the worse AP as the forced BSSID is gone. Then we are connected to it and... we stay connected unless we implement re-scanning.

Yes, this is what it amounts to. Wifi re-search should be strictly optional, as to not breaking existing setups.

What I am saying is: there are setups now which are broken, because of the state we have now. So basically we need to 'unbreak' these, with an option this imho works best.
The fixed BSSID is a hotfix, but not really sustainable either, because of said can of worms. 'what if?'.
Another approach could be to be able to give a ranked list, a la these 3 you can use, always prefer the one with highest prio.

But at the end a real fix can only be achieved by re-scanning.

The only other method I could think of is an optional parameter which says: "do not connect to APs with RSSI below xyzdB".
This would reliably work in my case as the wrong AP is so far away that I'm even wondering how it manages to connect.

markusdd, 2025-03-10:
> At least the S3 should be capable of doing roaming

uh oh!

nice find!

ranma, 2025-03-16:
> Idk but wouldn’t it be possible to activate it in software (if it isn’t already) and let the APs handle the roaming request?

Looks like there is example code in the esp-idf repository:
- https://github.com/espressif/esp-idf/blob/v5.2.5/examples/wifi/roaming/README.md
- https://github.com/espressif/esp-idf/blob/v5.2.5/examples/wifi/roaming/main/roaming_example.c

Jason2866, 2025-03-16:
I answered to the feature request. Don't expect this will be enabled soon. It was activated for a while and WiFi functionality was broken with a lot of routers APs. Probably wrong implementation of the steering protocol in this devices.
In project Tasmota (i am responsible there for the customized Arduino framework) we scan after a specific time for a stronger AP and change too. This feature can be switched on/off from the user. Without this we see the same issue you encounter.

Jason2866, 2025-03-16:
You can enable the needed settings when changing to Arduino core 3.1.3 using Platformio fork [pioarduino](https://github.com/pioarduino/platform-espressif32) with using  Hybrid Compile feature.
```
custom_sdkconfig =
    CONFIG_ESP_WIFI_11KV_SUPPORT=y
    CONFIG_ESP_WIFI_SCAN_CACHE=y
    CONFIG_ESP_WIFI_MBO_SUPPORT=y
    CONFIG_ESP_WIFI_11R_SUPPORT=y
```
to the Platformio config.

markusdd, 2025-03-16:
> I answered to the feature request. Don't expect this will be enabled soon. It was activated for a while and WiFi functionality was broken with a lot of routers APs. Probably wrong implementation of [...]

That's exactly the way I would've done it as well. 

I meanwhile also have some more insight: I was close to the install this weekend and in the Fritzbox router I noticed Mesh steering was enabled. I disabled it. Since then I have not seen the device moving to the horribly slow connection. But this of course is not ideal as this is a network level setting and now the (usually useful) steering feature is completely disabled.

So I think the tasmota approach with an autonomous background re-scan and switching to a stronger AP when available is much better than relying on the standards steering feature, which apparently is bricked for either the ESPs or some router models.

Jason2866, 2025-03-16:
@schlimmchen It is not a can of worms. It works perfectly in Tasmota. Before we added this feature to Tasmota. Tasmota was not useable with Mesh networks.

markusdd, 2025-03-16:
yes I have not observed any such problems with Tasmota devices in the network, they all work perfectly fine.

but to be fair, I think @schlimmchen has a point when he says this is actually an upstream issue that should be solved in OpenDTU, this is not onBattery specific.

Jason2866, 2025-03-16:
@markusdd Didn't do any research about project dependencies. Just came here since the opened issue in espressif github was linked to this repo

markusdd, 2025-04-03:
Ok, so after some more observation I can confirm that the issue is gone after deactivating client steering in the network.
So the issue most likely is that the AP tries to bump the device to a better connection occasionally, leading to a brief disconnect when not implemented properly/fully on the ESP side, and then it ends up with this horrible connection to the Mesh master which it never comes back out of.

This of course is a not really desirable situation as this implies deactivating a rather useful Wifi feature for all devices just to make sure that one single device does not drop into a worse connection.

So the most pragmatic approach maybe really is the choice of having a fixed BSSID as an option or the way that Jason suggested, like it is implemented in Tasmota.

markusdd, 2025-04-25:
ok, I am not sure what has changed, but now for me the problem is fully back, even with steering turned off.

every second day or so the DTU is now hopping back onto the most crappy connection it can possibly find and gets stuck there.

As this device is remote, all I can do right now is actually completely reboot the FritzBox to temprarily fix this until the device decides to once again trap itself into a 1MBit -78dBm connection that does not actually work.

We need some type of fix for this. Either a limit 'how bad' a connection is allowed to be, BSSID selection, or better the Tasmota approach of re-scanning periodically and choosing a better endpoint.
At this point I'd rate this problem as 'severe' as it completely disables the power regulation as obviously the DTU cannot read the power meter anymore in that connection state, so besides being annoying, this actually costs real money.

schlimmchen, 2025-04-25:
> I am however concerned that this is a task related to the upper fork and not an OpenDTU-OnBattery issue.

That is true. However, Thomas is quite happily adopting most of our changes (that touch the upstream code), so I feel okay working up a solution and feeling positive, that it will be adopted.

So, I started working on this a while back. Then I realized, that the whole network handling code in OpenDTU is in a rather bad shape. Not sure Thomas would agree, but it seems to me that this part of OpenDTU has not received the same love and diligence than the rest of it. For that reason I decided that I would attempt to rewrite most of it. That is, of course, a daunting task. There are definitely users with problems right now, but messing this up means creating problems for a lot more users...

Using AI assistance and a couple of hours I understood most of the code and the reason for respective functionality (I guess), and I wrote a test plan. However, after 26 test cases and a feeling that still a lot of stuff is missing... I gave up.

All this to say: Fixed BSSID is the best I can offer mid term. Everything else need meticulous planning and review and working together with Thomas and Andreas, as I am not taking on a full refactor of the NetworkSettings class otherwise. Just too risky. Also the re-scanning part, which is a good idea in my eyes, is just not integrated as straight-forward as one might think.

I'll make it a priority to implement the support to connect to a fixed BSSID. That should be doable.

spcqike, 2025-04-25:
> My WiFi Network is a mess... it is spread among several buildings.

Habe you ever considered using proper network gear? Like a UniFi network? 

I run 3 access points, each runs 4 SSIDs. I have no problems with my devices whatsoever. 

I mean, at some point a AVM Fritz network just isn’t enough anymore (:

Manos1966, 2025-05-10:
I take the opportunity to share something we just identified:

OpenDTU-OnBattery has been working without issues in our "messy" WiFi Network (5 FritzRepeaters) for almost a year.
During the past two months problems startet, OpenDTU-OnBattery was getting "lost" and only a restart could bring it back to life again.

As @markusdd mentionned, the DTU was not connecting to the right Repeater or, losing connection after some time (hours/days).

We decided to get smarter last week 🐵 and gave a FritzRepeater 6000 a different WiFi SSID, then had the DTU (and the Shelly we have to restart the DTU remotely) connect to that SSID in order to prevent it from switching among Repeaters.
For a few days it worked! 🍾 

Today, it didn't 😥 Neither DTU, nor (first time) Shelly were reacheable!
This time we **restarted the Fritz!Repeater** 6000 ... and both Shelly and OpenDTU-OnBattery were again **connected**!

I remember there was a Firmware update from Fritz about that time, when our problems started.
Maybe it is a change of the Fritz Firmware that causes this misery, just saying....

Manos1966, 2025-05-19:
UPDATE: It is more than one week gone after the last restart (not of OpenDTU-OnBattery but, of the FRITZ!Repeater 6000 😆 )

No problems ever since so, the solution "gave a FritzRepeater 6000 a different WiFi SSID, then had the DTU (and the Shelly we have to restart the DTU remotely) connect to that SSID in order to prevent it from switching among Repeaters." works 👍 

I was not aware that it is possible to run a Fritz!Repeater with different SSID name than the Master it is connected via WLAN to.

AndreasBoehm, 2026-10-01:
@markusdd @Manos1966 I implemented the feature to switch to a better AP if available by doing a rescan every X minutes.
Please give it a go and let me know if it works and fixes your issues: https://github.com/hoylabs/OpenDTU-OnBattery/pull/2739#issuecomment-5938623704

Manos1966, 2026-10-02:
@AndreasBoehm Sorry Andreas, ich kann es nicht testen, weil die OpenDTUs haben deine Firmware Version die die MQTT Daten verschiedene Batterien zum gleichen HomeAssistant senden 😢

AndreasBoehm, 2026-10-13:
> 30\. [@AndreasBoehm](https://github.com/AndreasBoehm) Sorry Andreas, ich kann es nicht testen, weil die OpenDTUs haben deine Firmware Version die die MQTT Daten verschiedene Batterien zum gleichen HomeAssistant senden 😢

Du kannst es jetzt testen, das MQTT Batterie Daten Thema habe ich gemerged und das ist jetzt auch Bestandteile der Wifi Test Version 👍

Manos1966, 2026-10-20:
Hallo Andreas,
ich habe am Wochenende die ersten zwei DTUs upgedated, sie funktionieren seitdem einwandfrei (WiFI & MQTT). 👍

closed 2026-11-06 by AndreasBoehm