hoylabs/OpenDTU-OnBattery #1717: DPL Skalierung bei nur 2 von 4 genutzen Eingängen regelt die Leistung zu hoch!

CWILDMOUNTAIN, 2025-03-10:
### What happened?

Moin!
ich habe ein ähnliches Problem wie in Issue #1707, allerdings wird dies nicht durch Aktivieren der Verschattungsfunktion gelöst, sondern verschlimmert.
Wenn ich diese aktiviere, regelt der Wechselrichter das Limit sky-high und ohne Bezug zur aktuell benötigten Leistung. In den Log wurden zwischendurch auch Werte angezeigt, die eine Limit jenseits des Maximums (>3000W) angezeigt haben ... 
Mein Setup: HMS-2000-4T (firmware 1.0.16, noch keine Power Distribution Logic), openDTU (2025.02.10), 2 Zendure Hub 2000 (1 pro 2 Eingänge).

Daraus ergibt sich folgendes Problem: Wenn ein Hub (= zwei Eingänge) keine Energie mehr liefern kann (Batterie leer, keine Sonne) regelt die DTU hoch bis auf das maximale Limit (2000W) und der der SolarHub 2000 speist Vollgas treudoof 1000W über die zwei aktiven Eingänge ein und leer dabei die Batterie, die Energie "verpufft" jedoch ,denn wie gesagt, die tatsächliche Einspeiseleistung wird dann in keiner Weise mehr mit der Stromzählermessung gematched.

Liegt es daran, dass die Firmware des WR noch zu "dumm" ist und keine Power Distribution Logic kann?
Ich scheue mich ein bisschen vor einem Upgrade auf die aktuelle Version, da es ja Probleme mit dem Setzen der Limits geben soll ... 

Any Ideas?

### To Reproduce Bug

Verschattetete Eingänge ausgleichen aktiviert, 2 von 4 Eingängen ohne Leistung

### Expected Behavior

wie bei vorigen openDTU FW Versionen- hier wurde die Leistung der verbleibenden Eingänge verdoppelt, sodass am Ende wieder einen Nulleinspeisung herauskam

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-10:
Schau mal hier: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1665#issuecomment-2675196117

CWILDMOUNTAIN, 2025-03-10:
Vielen Dank für die schnelle Antwort!! 
wie kann ich die dort verlinkte Testversion installieren?

schlimmchen, 2025-03-11:
> wie kann ich die dort verlinkte Testversion installieren?

Passende ZIP-Datei [herunterladen](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/13462326933), entpacken/öffnen und die darin enthaltene `.bin` (ohne `factory` im Namen) über die Web UI als Firmware Update einspielen.

AndreasBoehm, 2025-03-19:
@Thaui57 Probier mal die aktuellste `development` version: https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/13941685611

Wenn das nicht hilft benötigen wir bitte ein Konsolen Log des DPL mit aktiviertem verbose logging.

Thaui57, 2025-03-20:
@AndreasBoehm Danke für die Beta. Bis dato scheint DPL zu funktionieren. Vielleicht ist es noch zu früh ein endgültiges okay zu geben. Läuft ja erst einen Tag.

AndreasBoehm, 2025-03-20:
@Andre0be Nutzt du einen Smart-Buffer/Akku oder hast du direkt Solarmodule an deinem WR?

AndreasBoehm, 2025-03-20:
@Andre0be Dann schau dir bitte [diesen Issue](https://github.com/hoylabs/OpenDTU-OnBattery/issues/1675) an.

Andre0be, 2025-03-20:
@AndreasBoehm vielen Dank. Hab mir gerade das neuste daily build installiert und werde es nun beobachten...

Andre0be, 2025-03-21:
@AndreasBoehm trotz day build und fix, korrigiert er weiterhin nicht nach oben. 

Ich habe DPL erstmal abgeschaltet. Soll ich einen neuen Fall eröffnen?

[image]

AndreasBoehm, 2025-03-21:
> @Andre0be Dann schau dir bitte [diesen Issue](https://github.com/hoylabs/OpenDTU-OnBattery/issues/1675) an.

Hier ist nochmal der Issue der dein Problem behandelt inklusive Workaround.

CWILDMOUNTAIN, 2025-03-21:
> [@CWILDMOUNTAIN](https://github.com/CWILDMOUNTAIN) Ich wollte nur Fragen wie es mit der Testversion bei dir läuft?

Moin @AndreasBoehm! 

DPL tut in meinem Setup seither wie sie soll ohne Probleme: regelt weder zu hoch noch zu niedrig. 
very happy so far!

AndreasBoehm, 2025-03-21:
Super, das freut mich! Danke für deine Rückmeldung, dann mache ich hier zu.

closed 2025-03-21 by AndreasBoehm