hoylabs/OpenDTU-OnBattery #1719: [Bug] Cannot un-govern solar-powered inverters

schlimmchen, 2025-03-11:
### What happened?

Fiddling with the setup at my parents' house, I deselected the solar-powered inverter as governed. Afterwards the DPL was unresponsive.

A new limit was sent to the solar-powered inverter (the same it already had) in each iteration of trying to get rid of the inverter from the list of governed inverters. That caused the `reloadConfig()` method to never conclude reloading the config.

### To Reproduce Bug

Setup the DPL to govern an inverter that is marked as solar-powered. Edit the config and unselect the inverter, i.e., un-govern ist. Save that config. The DPL will become unresponsive.

### Expected Behavior

Any inverter that is un-governed shall have it's `standby()` method called and monitored until it reached the respective target state.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-11:
Is this the same as https://github.com/hoylabs/OpenDTU-OnBattery/issues/1641?

schlimmchen, 2025-03-11:
Wow, I am really impressed by your memory! You absolutely nailed it! That one is exactly the same issue.

Shame on me for creating a duplicate :grin:

PR #1720 (2025-03-18): Fix: cannot un-govern solar-powered inverters
the standby() method of battery- and smart-buffer-powered inverters only sets the inverter to actual standby. reaching that state can be checked for, so the update() method returns false (read: not busy) immediately if the inverter is already in standby.

on the other hand, solar-powered inverters update their limit to the configured minimum limit. testing for "is that limit already set" is not implemented and shall not be implemented. instead, we record a timestamp when starting the update, send the limit, and wait for it to complete before the up [... cut]
closed 2025-03-18 by schlimmchen