hoylabs/OpenDTU-OnBattery #1726: [Request] Temperaturabhängiger Ladegrenzwert

indie89, 2025-03-15:
### Is your feature request related to a problem? Please describe.

Ich würde gerne die Ausgangstemperatur meines Huawei 4850 unter 63 Grad halten und dennoch den Ladestrom so hoch wie möglich ansetzen, insbesondere an Tagen mit wechselhaftem Wetter. Die Temperatur steigt insbesondere bei Ladeströmen ab 30 A nach 30-60 min auf den o.g. Wert. 

### Describe the solution you'd like

Wenn im Abschnitt "Ladegerät" 2 weitere Einstellungen möglich wären, wäre ich glücklich: "Grenztemperatur" und "Abgesenktes Leistungslimit". Würde konkret bedeuten, dass man das "globale Ladelimit" auf den physikalisch höchsten Wert setzen kann (zB 2000w) und dann bei Erreichen / Überschreiten der "Grenztemperatur" (Ausgangstemperatur des Huawei) das "Abgesenktes Leistunglimit" gültig ist. Das DPL würde also abhängig von der Temperatur (kleiner oder größer/gleich der Grenztemperatur) das entsprechende Limit nutzen.

Natürlich könnte man es noch feingranularer machen, aber insbesondere für kurzfristige Spitzen vs langfristiges Laden ist o.g. Lösung ein guter Aufwand / Nutzen Kompromiss, denke ich.

### Describe alternatives you've considered

Natürlich kann man sich das manuell auch in HA o.ä. basteln, ich glaube aber, dass das eine breite Zielgruppe hat und es hier mit recht überschaubarem Aufwand implementiert werden könnte.

### Additional context

Vielen Dank fürs Prüfen!!

gitisgreat2023, 2025-03-18:
@indie89 Zufällig beschäftige ich mich gerade wieder mit dem Thema Huawei R4850G2 und T_out. Gestern den Powersupply für meine 3x 140mm Kühler aufgerüstet. Jetzt bei genug PV Überschuss laufen die mit 6.0W, einer pustet auf dem kleinen internen Kühler und einer sitzt hinten beim Ausgang und zieht.  Heute wird's sonnig und kann ich mal schauen welche Temperaturen er damit erreicht. Ich werde die Tabelle heute mal füllen.

T_in ist 18°C
| P_out [W]  | T_out [°C] |
| ------------- | ------------- |
| 750  | 26  |
|   |   |

indie89, 2025-03-18:
Danke für die Daten. Meine Erfahrung ist auch, dass sich unter längerer last auf einer Stufe die Temperatur nochmal etwas erhöht. Dein genaues Setup habe ich natürlich nicht, aber ähnliche Mechanismen etras mehr zu kühlen. Dennoch würde ich mich freuen, wenn ich eine "abgesenkte Leistung" bei einer einstellbaren "Grenztemperatur" definieren könnte. Das kann dann ja jeder auf seine Gegebenheiten bzw Ansprüche einstellen.

gitisgreat2023, 2025-03-19:
Ja, das Feature ist so oder so sehr sinnvoll, darum geht es mir nicht direkt.
Das Ziel ist ja maximal viel zu laden in der Batterie unter Einhaltung einer Temperaturgrenze. Zu 63°C: intern sollte der R4850G2 laut Datenblatt eine Drosselung ab 75°C automatisch machen... aber ich selber habe das noch nie beobachtet und habe im Netz auch keine gefunden. Irgendwie merkwürdig, es könnte sein dass man irgendwelchen Register setzen muss in der Software damit es explizit aktiv ist?! In dem Netzteil sind caps verbaut die 105°C als Rating haben.

Wenn man dieses Ziel: maximal laden mit <=X°C, ist es natürlich gut zu wissen was man typisch erreichen kann mit einer einfache Kühlung, damit erreicht man höhere Ladeleistung unter Einbehaltung von dieser Bedingung. In meinem Fall Kühlrippen und 2x140mm Lüfter (die pro Stuck 2W AC ziehen).

Seit ich aktive Kühlung habe, ist die Zeitkonstante zu Equilibrium auch deutlich kürzer geworden als ganz am Anfang wo ich gar keine Kühlung hatte. Einee rein passive Kühlung hat eine lange Zeitkonstante, da wird das ganze Gerät nach und nach immer wärmer, inkl. die Luft die um das Gerät hängt.

gitisgreat2023, 2025-06-18:
@indie89 [Siehe den bug report hier](https://github.com/hoylabs/OpenDTU-OnBattery/issues/2061). Alle Untersuchungen hier waren mit der Einstellung "auto" für den fan mode. Mit "full" sind die Temperaturen viel niedriger...
Kannst Du bitte mal [2025.06.11](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2025.06.11) installieren und schauen was Deine Temperaturen sind bei 3kW bei jeweils fan mode "auto" und "full"? Sind bei Charger einstellbar:

[image]