hoylabs/OpenDTU-OnBattery #1727: [Request] Netzdienliches Laden des Speichers

indie89, 2025-03-15:
### Is your feature request related to a problem? Please describe.

Die aktuelle Implementierung des DPL unterstützt maximal den Nutzer und seine Bedürfnisse. Vielen Dank dafür!! 

Es ermöglicht allerdings nicht, die (relativ stupide) Möglichkeit den Speicher etwas netzdienlicher zu laden, indem man eine Uhrzeit setzen kann, vor der das Ladegerät nicht anfängt zu laden.

Thema "netzdienliches Laden" erläutere ich hier nicht ausführlich, o.g. Idee wäre auch nur der Anfang zu dem Thema, da würde theoretisch langfristig noch mehr gehen.

### Describe the solution you'd like

Einfügen eines Feldes unter "Ladegerät", in dem man die Uhrzeit eintragen kann, vor der nicht geladen wird. So hat man etwas im Griff, dass über Mittag der Akku geladen wird und nicht direkt morgens, wenn der erste Überschuss da ist.

Das Netz und alle Stromkunden danken :-)!

gitisgreat2023, 2025-03-15:
Zu überlegen ist es direkt so aufzusetzen dass netzdienliches laden in der Nacht auch möglich wird.
Netzdienliches Verhalten geht ja in beide Richtungen. Dafür braucht man nur noch ein Ladegrenze (V oder SoC target).

Laden: von Uhrzeit x bis Y bis Z Ladeschlussspannung laden

Für tagsüber wäre die Ladeschlussspannung auf den üblichen zu setzen (vielleicht Option die zu nutzen).

indie89, 2025-03-15:
> Zu überlegen ist es direkt so aufzusetzen dass netzdienliches laden in der Nacht auch möglich wird. Netzdienliches Verhalten geht ja in beide Richtungen. Dafür braucht man nur noch ein Ladegrenze (V [...]

Du meinst "netzdienliches Entladen nachts" richtig? 
Das halte ich für erstmal nicht so wichtig, da das Entladen ja gemäß des Hausverbrauch in Richtung quasi-Nulleinspeisung passieren soll und damit nie wirklich "netzundienlich" ist. Und ein reines entladen in der Nacht, nur um das netz zu stabilisieren halte ich für nicht sinnvoll, da hier ein potentieller Nachteil für uns als Nutzer entsteht und es keine Kompensation für die zusätzlichen Batteriezyklen gibt.
Würde das Thema daher erstmal super simpel halten und diese eine Uhrzeit, aber der geladen wird, einführen (wenn leer, dann unbegrenzt). Alles weitere können wir anschließend optimieren... Ist tatsächlich erst der Auftakt zum Thema Netzdienlichkeit.

oldman4U, 2025-03-15:
Wäre es nicht sinnvoller statt einer Uhrzeit den EPEX Spotpreis zu nehmen? Den Überschuss nach Mitternacht bis 6 (so wie die Jahre davor) sehe ich zB heuer nicht mehr. Die netzdienlichen Zeiten scheinen sich zu verändern und werden je nach Jahreszeit auch unterschiedlich sein.

indie89, 2025-03-15:
Gute Idee, aber sicherlich nicht unwesentlich komplizierter zu implementieren 😅. Wie gesagt: beim Thema Netzdienlichkeit eröffnet sich ein ganz eigenes Universum. Mein Vorschlag zum Start war und bleibt die Uhrzeit :-). Ist nicht perfekt, aber immerhin ein Beginn.

gitisgreat2023, 2025-03-15:
> Du meinst "netzdienliches Entladen nachts" richtig? Das halte ich für erstmal nicht so wichtig, da das Entladen ja gemäß des Hausverbrauch in Richtung quasi-Nulleinspeisung passieren soll und damit [...]

Nein, laden.
Nachfrage geringer, schiebt die Nachfrage tagsüber bei wenig PV in die Nacht.

schlimmchen, 2025-03-16:
Die Kommentare habe ich noch nicht gelesen, aber ich gebe schon einmal zu bedenken, dass das Thema zwar spannend ist und wir es definitiv auch betrachten sollen, aber dass die Logik dazu recht komplex werden kann und wohl auch Daten aus dem Internet braucht, weshalb ich in OpenDTU-OnBattery eigentlich nur eine Schnittstelle sehe um netzdienliches Laden zu unterstützen, aber Entscheidungen darüber müssen woanders getroffen werden.

indie89, 2025-03-16:
Absolut! Daher bitte (erstmal) auf meine ursprüngliche Idee konzentrieren: Startuhrzeit zum Laden (bzw Zeitfenster). Dann ist keine sonstige Schnittstelle notwendig :-)

Harisjems, 2025-03-16:
Ich bin jetzt nicht so tief in der Materie wie manche andere hier, aber anstatt WR bei Überschuss zu drosseln, kann man auch, wie z. B. bei Tasmota Google Charts Script, einen temporären Verbraucher zuschalten; bei mir ist es z. B. ein Liebherr Kühlschrank im Gartenhaus, um Bier im Sommer gut zu kühlen.

if ((power2<-300) and (swespflg==0))
{
=>websend [192.168.1.12] power ON
print power ON %power2%W
swespflg=1
}
if ((power2>100) and (swespflg==1))
{
=>websend [192.168.1.12] power OFF
print power OFF %power2%W
swespflg=0
}

[image]

spcqike, 2025-03-16:
@schlimmchen 
Ich hatte auch vor 2 Jahren oder wann bereits die Idee eingebracht, ob man nicht die Spot Markt Preise irgendwie integrieren will. Damals noch Unter dem Aspekt, dass ich mit der Solar only Anlage zur Mittagszeit nicht einspeisen will und niemanden „zur Last“ fallen möchte. 

Stand 2025 ist dies ja sogar Pflicht für neue und größere Anlagen. Bei 0 bzw. Negativ Preisen eben nicht mehr einzuspeisen. 

Vielleicht macht es doch früher oder später Sinn, sowas direkt in OpenDTU zu integrieren. 

Einmal am Tag (und zum Neustart) die maximal 48 Preise (24h heute und ab 14 Uhr die 24h morgen) abzuholen und basierend dadraus eben Steuerungen und Entscheidungen zu ermöglichen. 

Wie eben die hier gewünschte Verschiebung des Ladens oder auch die netzdienliche Einspeisung für Solar only Anlagen. 

Auch wenn letzteres bereits heute mit dem DPL Modi (Mode 2) und einer externen Steuerung geht, wie du korrekt anmerkst (:

gitisgreat2023, 2025-03-16:
> Absolut! Daher bitte (erstmal) auf meine ursprüngliche Idee konzentrieren: Startuhrzeit zum Laden (bzw Zeitfenster). Dann ist keine sonstige Schnittstelle notwendig :-)

@schlimmchen 
Sehe ich genauso. Mit 3 Parameter wäre sowohl laden nachts (§14a Modul 3, oder Octopus Energy, oder typische Zeit für Tibber) von zB zwischen 0 und 5 Uhr (Start und Stop einstellbar) und Zielspannung oder Ziel SoC als auch Tagsüber netzdienlich laden abgedeckt (11 bis 16 Uhr zB). Damit wäre das rüdimentäre netzdienliches Verhalten vorhaden, ohne zusätzliche Schnittestelle, mit 3 Parameter extra.

wusel42, 2025-04-10:
Hmm, @indie89 schrieb:

> Es ermöglicht allerdings nicht, die (relativ stupide) Möglichkeit den Speicher etwas netzdienlicher zu laden, indem man eine Uhrzeit setzen kann, vor der das Ladegerät nicht anfängt zu laden.

Inwiefern ist es netzdienlich, einen Akku durch Solarladeregler erst zeitverzögert zu laden? Und bei (Ergänzungs-) Ladung des Akkus aus dem Netz wäre doch der aktuelle Preis relevant, nicht eine statisch Uhrzeit, und eine Ladung in Niedrig- bis Negativkostenzeiträumen gerade netzdienlich?

spcqike, 2025-04-11:
> Inwiefern ist es netzdienlich, einen Akku durch Solarladeregler erst zeitverzögert zu laden?

Die PV Leistung reicht am Morgen noch nicht, den gesamt deutschen Energiebedarf zu decken. Daher fahren frühs Kohle und Gas entsprechend stark mit. 
Wenn nun jeder seinen Speicher nach diesem Morgenpeak bzw. erst dann, wenn sowieso genug Überschuss da ist, lädt, ist das eben netzdienlich. 

> Und bei (Ergänzungs-) Ladung des Akkus aus dem Netz wäre doch der aktuelle Preis relevant, nicht eine statische Uhrzeit, und eine Ladung in Niedrig- bis Negativkostenzeiträumen gerade netzdienlich?

Ja, schon. Das wäre noch besser. Aber eine lastverschiebung, egal welcher Art, außerhalb der beiden peaks (morgens und abends) ist bereits gut. 

Zum Strompreis:
Dieser ist eben leider auch nicht das Maß der Dinge. Dafür ist das Netz in Deutschland zu groß und zu schlecht ausgebaut. 
Trotz negativpreis kann dein lokales Netz eine unterdeckung haben und Energie benötigen. Nicht ohne Grund gibt es die redispatch Maßnahmen, die erneuerbare abriegeln und anderorts konventionelle Kraftwerke hochfahren. (Da haben wir aber, soweit ich weiß, keine Einsicht rein. Also In die Belastung des lokalen Netzes)

indie89, 2025-04-11:
Genau was @spcqike schreibt. 

Wenn dich, @wusel42 , das näher interessiert, schau mal hier auf die Homepage, die alle relevanten Daten zu Erzeugung und Preisen hat und vom Fraunhofer betrieben wird: https://www.energy-charts.info/charts/price_spot_market/chart.htm?legendItems=2w5w6 . Auf YouTube ist zB der Akkudoktor eine gute Quelle für Infos zu dem Thema, in der Forschung, aber sehr anwendungsnah, zB Prof. Lion Hirth.

gitisgreat2023, 2025-04-11:
Ich habe es vor einer Woche in Node-RED implementiert, um entsprechend OpenDTU-onBattery zu steuern. Über die GUI kann ich es einfach ein- und ausschalten sowie die Uhrzeiten und den Ladestrom einstellen.

[image]

Ein Feature, das ich ebenfalls direkt implementiert habe und über das wir uns hier Gedanken machen sollten (was ich vorher nicht bedacht hatte), ist der PV-Überschuss beim Laden aus dem Netz ja/nein.

Zum Beispiel wollte ich an einem wolkigen Nachmittag mit 25A (etwa 1250W Ladeleistung) die Batterie ein bisschen laden, um es mal zu testen. Das lief super, allerdings kam die Sonne ab und zu durch, und entsprechend wurde ins Netz eingespeist, statt in die Batterie. Also habe entsprechend erlaubt, den Ladestrom temporär nach oben beizustellen wenn Überschuss vorhanden ist:

''''
// Enable full power if price is low
if (grid_charge_active && currentHour >= grid_charge_hr_start && currentHour < grid_charge_hr_stop) {
    msg.payload = 2;  // Let openDTU enable
    if (charge_actual_dc_current > grid_charge_amps && grid_value < 0){
        current_adjustment = charge_actual_dc_current - grid_charge_amps;
    }
    if (grid_value < 0){
        current_adjustment = current_adjustment + -1*grid_value / battery_dc; // have to fix this, alternates. store it and give it an update.
    }
    msg.current = grid_charge_amps+current_adjustment;
    msg.voltage = grid_charge_volts;
    msg.text = "Grid charge " + grid_charge_hr_start + " to " + grid_charge_hr_stop + " hrs with + " + (msg.current * battery_dc).toFixed(0) + "W";
    return msg;
}
''''

wusel42, 2025-04-11:
Moin,

> Die PV Leistung reicht am Morgen noch nicht, den gesamt deutschen Energiebedarf zu decken. Daher fahren frühs Kohle und Gas entsprechend stark mit. Wenn nun jeder seinen Speicher nach diesem Morgenpeak bzw. erst dann, wenn sowieso genug Überschuss da ist, lädt, ist das eben netzdienlich.

Hängt vielleicht vom Setup ab: ich habe hier Panele an Victron MPPTs, jene laden die Batterie, an der auch die HMS hängen – simples Setup. Die Victrons kann ich nicht regeln, die werden durch die Batterieparameter geregelt.

Der DPL müßte $hier also zwischen Sonnenaufgang und definiertem "netzdienlichem Ladestart" sekündich exakt die momentan vorhandene Solarleistung aller Victrons ins Netz speisen, damit die Batterie nicht weiter entladen wird. Mit einem separat schaltbaren Batterieladeregler ist das natürlich wirklich einfach umzusetzen. Dennoch würde ich das lieber vorhersageabhängig geschaltet sehen, denn die Batterie soll schon tagsüber voll werden, das wäre meine oberste Priorität.

FTR: Aufgrund des variablen Tarifs (Tibber) lasse ich in den Stunden mit überdurchschnittlich hohen Preisen bis zu 600W aus der Batterie einspeisen und reduziere damit die Last aufs Grid (Grundlast $hier ist höher) ... und meinen Geldbeutel ;-)