hoylabs/OpenDTU-OnBattery #1732: Überstrom /Überspannung OpenDTU on Battery Hmt-2000-4t

Blade138, 2025-03-17:
### What happened?

Moin
vor weg, ich bin ziemlicher neuling was Github angeht und openDTU bin ich auch gerade erst angefangen.

Ich habe nun eine openDTU on battery und mein Hoymiles HMT-2000-4T wird damit mit gesteuert. Das funktioniert auch soweit. 

### To Reproduce Bug

Ich bekomme nur ständig in der Ereignisseanzeige der openDTU die Meldungen:

46	FB Überspannung
47	FB Überstrom

Der Wechselrichter funktioniert einwandfrei und macht seine Arbeit. Ich habe mittlerweile auch die Spannungen, Strom und Leistung der 4 String in Grafana zur Überwachung (alle 2 sek). Hier ist aber nichts zu sehen. Maximale Spannung gestern war 50,1V und Strom max 10A. Der Hoymiles kann ja mehr ab pro Eingang.

### Expected Behavior

Zur Installation. Der WR wird separat betrieben. Also eigene Zuleitung vom Zähler 5x4qmm, Mit FI und Sicherung. Betrieben wird der Wechselrichter mit zwei Zendure HUB 2000. Habe alle 4 Eingänge belegt, pro MPPT ein Hub.
Der Wechselrichter ist ja ein 3 Phasen WR, gibt's da vielleicht Probleme mit der openDTU?

Vielleicht kann mir ja jemand helfen und vielen Dank im voraus.

Gruß
Blade 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

0.1.29-OnBattery-5

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

hhoebel, 2025-03-17:
Ich schätze das ist eher ein Hoymiles-Problem. Ich habe 3 Stk 56V-LiFePo4-Akkus direkt an einem HM2250 und bekomme auch jeden Morgen 1x FB Überspannung (aber kein Überstrom). Das passiert interessanterweise aber genau 1x pro Tag und immer ungefähr dann, wenn zum ersten Mal am Tag der Leistungsbedarf im Haus steigt - der Zeitpunkt deckt sich ungefähr mit dem Start der Kaffeemaschine. Eventuell sorgt die für einen Spannungspuls im Netz. Ich glaube weniger das die Überspannung auf der Akku-Seite gemessen wird - da habe ich morgens naturgemäß die geringste Spannung und die 3 Akkus können problemlos über 300A liefern - da gibt es sicher keinen Spannungspuls über 60V.

Blade138, 2025-03-19:
Das kann ich nun bestätigen. Es liegt an einem großen bzw blöd gewählten Verbraucher in der Nähe. Habe den WR einzeln abgesichert aber dort such ne Steckdose gesetzt. Dort steckt ein shelly plug mit ein paar Lüftern dran. Und immer wenn der Schalter tritt die Meldung auf. Shelly und Lüfter auf einer anderen Steckdose, keine Probleme mehr.

closed 2025-03-19 by Blade138