hoylabs/OpenDTU-OnBattery #1735: [Bug] Powermeter: SMA Home Manager: DTU unable to read/receive readings on-time

AndreasBoehm, 2025-03-18:
### What happened?

DTU is not able to receive and use SMA Home Manager powermeter values within the 1 second interval that the SMA HA is sending them.

> Zumindest erscheinen die Broadcast-Werte vom SMA EnergyMeter nun innerhalb von 1 Sekunde - allerdings erst nach einem Neustart. (Vielleicht hätte ich den Neustart früher probieren sollen. Möglicherweise haben die vielen Konfigurations-Versuche zu dem Lag geführt... Nee, 15 Minuten nach Neustart sind es schon wieder 3 Sekunden Verzögerung, komisch.)

### To Reproduce Bug

According to @db5bnm its working fine after a restart of the DTU but is getting slower and slower after ~15 minutes of runtime.

### Expected Behavior

Powermeter values are read and worked with as fast as they arrive.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

- https://github.com/hoylabs/OpenDTU-OnBattery/pull/1606#issuecomment-2708216729
- https://github.com/hoylabs/OpenDTU-OnBattery/pull/1606#issuecomment-2708495872

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Snoopy-HSS, 2025-03-21:
Schau ich mir mal an, ist mir so noch nicht aufgefallen.
Läuft seit Monaten stabil

AndreasBoehm, 2025-03-21:
Vielen Dank @Snoopy-HSS !

db5bnm, 2025-03-21:
Hallo, danke für das Angebot. Noch hatte ich nicht alle Informationen zusammen, deshalb habe ich mich noch nicht gemeldet.
Zuerst ein paar Infos zu meiner Systemlandschaft:
- OpenDTU-OnBattery (ESP32-S3) mit Ethernet-Port am USB-Anschluss (aus dem Shop)
- SMA EnergyMeter 2.0 (gleiche Broadcast-Nachrichten, wie SMA HomeManager 2.0).
- WR: Hoymiles HMS-1800-4T
- (smart) Akku: Sunlit BK215

Mir fiel auf, dass der SMA-Daemon, der alle Leistungswerte in eine InfluxDB schreibt, diese Werte 3 bis 6 Sekunden früher loggt - was etwas überraschend ist. (Alle relevanten Geräte hängen per Netzwerkkabel am gleichen Netgear-Switch.)
Also habe ich ein weiteres Auslesetool gestartet. Dieses zeigt die Werte "zeitgleich" (auf Werte unter 100ms möchte ich nicht eingehen) an wie der SMA-Daemon.
Da das Phänomen unbekannt scheint, hätte ich erst einmal gefolgert, dass dies möglicherweise am USB-Ethernet-Interface (das vielleicht nur wenige nutzen) liegt und ein Buffer immer voller läuft.
Allerdings werden WR-Limits, die ich ans OpenDTU-OnBattery über die API sende, sofort ausgeführt. Dies erkenne ich daran, dass das SMA EnergyMeter schon bei der nächsten Messung die entsprechende Leistungsdifferenz anzeigt. Der WR erhält die Nachricht also scheinbar sofort und kann diese (erstaunlicherweise) auch sofort umsetzen.

Aktuell bin ich überfragt, warum es zu dem Phänomen bei der Leistungsanzeige (auch im OpenDTU-Websocket) kommt.
Dass der DPL damit gegen einen "alten" Wert regelt, konnte ich nicht mit einer Verringerung des DTU-Abfrageintervalls erreichen, sondern mit einer Vergrößerung. Wenn nur alle 10 Sekunden ein Wert vom WR abgefragt wird, funktioniert das Ausregeln gut, da der "verzögerte" Wert und der aktuelle Wert dann ziemlich gleich sind.
Damit ist das Problem für mich hinreichend lösbar.

OffTopic, sorry:
Problematisch bleibt, dass das SMA EnergyMeter (quasi) Momentanwerte misst - also auch Einschaltspitzen von 1,8 kW, die schon in der nächsten Sekunde zurück bei 200 W sind. Dies führt zu unnötigen und kontraproduktiven Regelversuchen des DPL und begünstigt das Schwingen.
Volkszähler, Shellys, etc. liefern möglicherweise leseintervall-bedingt immer nur gemittelte Werte, so dass hier das Problem wahrscheinlich nicht auftritt.

Mein externes Tool, nimmt ganz simpel den Durchschnitt der letzten 5, 10 oder 15 Messwerte und steuert damit den WR. Das funktioniert gut.
(Nebenbei löse ich mit dem Tools auch das Problem, dass der WR scheinbar gelegentlich den Kanal wechselt und dann von OpenDTU-OnBattery nicht gesehen wird: Liefert der Akku Leistung an den WR, muss dieser ja "an" und damit erreichbar sein. Empfängt OpenDTU-OnBattery aber nichts, werden die Nachbarkanäle getestet. Bisher musste ich nur zwischen 865,00 und 865,25 MHz wechseln, um den Kontakt zum WR wiederherzustellen.)

Snoopy-HSS, 2025-03-22:
Also auch nach Stunden liefert der SMA noch alle Sekunde die Werte.

[image]

Was mir allerdings aufgefallen ist, wenn der Request Transmit auf >10 geht, schafft die openDTU anscheinend nicht mehr die Werte innerhalb 1 Sec. zu verarbeiten und zeigt erst nach ca. 3 Sekunden der Wert an.
Dies ist aber sporadisch. 

Wo die Retransmits herkommen, konnte ich noch nicht prüfen. Allerdings scheint er während der Retrys, nichts anderes zu machen...

Die Idee mit dem Average Wert hatte ich auch schon, allerdings wird dadurch die gesamte Regeleung auch etwas träger, wenn er von  200 auf 5000 Watt wechselt, dauert das dann eben bis zu 5 Sekunden bis er komplett nachgeregelt hat...

db5bnm, 2025-03-22:
Da habe ich mich vielleicht missverständlich ausgedrückt.
Natürlich meldet OpenDTU-OnBattery (quasi) jede Sekunde einen neuen Messwert vom SMA-Meter - auch nach zig Tagen.
Jedoch geraten diese Messwerte (bei mir) immer mehr in Rückstand zum Sendezeitpunkt der Broadcast-Message vom SMA-Meter - was ich mir nicht erklären kann und auch bei keinem anderen Empfänger der Broadcast-Nachrichten sehe.

Auszug eines Logs z.B. vom EnergyMonitor (Tool aus 2016 bzw. 2021 aus Photovoltaikforum.com):
14:21:44:823 | 378,5
14:21:45:813 | 421,7
14:21:46:823 | 372,7
14:21:47:823 | 358,9
14:21:48:823 | 405,6
14:21:49:823 | 372,8
14:21:50:823 | 362,8
14:21:51:812 | 408,3
14:21:52:823 | 479,5

Gleicher "Messwerte-Zeitraum" vom OpenDTU-OnBattery:
14:21:50.037 > [PowerMeterUdpSmaHomeManager] Leistung = 378.5 (timestamp 3187949892)
14:21:50.081 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 102.3 (timestamp 3187949892)
14:21:50.129 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 242.7 (timestamp 3187949892)
14:21:50.176 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.5 (timestamp 3187949892)
14:21:51.037 > [PowerMeterUdpSmaHomeManager] Leistung = 421.7 (timestamp 3187950891)
14:21:51.095 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.8 (timestamp 3187950891)
14:21:51.143 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 285.9 (timestamp 3187950891)
14:21:51.189 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.9 (timestamp 3187950891)
14:21:52.037 > [PowerMeterUdpSmaHomeManager] Leistung = 372.7 (timestamp 3187951891)
14:21:52.090 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.4 (timestamp 3187951891)
14:21:52.137 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 237.6 (timestamp 3187951891)
14:21:52.184 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.6 (timestamp 3187951891)
14:21:53.037 > [PowerMeterUdpSmaHomeManager] Leistung = 358.9 (timestamp 3187952891)
14:21:53.089 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 102.2 (timestamp 3187952891)
14:21:53.136 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 222.6 (timestamp 3187952891)
14:21:53.182 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 34.1 (timestamp 3187952891)
14:21:54.037 > [PowerMeterUdpSmaHomeManager] Leistung = 405.6 (timestamp 3187953891)
14:21:54.089 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.7 (timestamp 3187953891)
14:21:54.137 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 270.9 (timestamp 3187953891)
14:21:54.184 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 32.9 (timestamp 3187953891)
14:21:55.036 > [PowerMeterUdpSmaHomeManager] Leistung = 372.8 (timestamp 3187954891)
14:21:55.089 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.6 (timestamp 3187954891)
14:21:55.135 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 237.7 (timestamp 3187954891)
14:21:55.182 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.5 (timestamp 3187954891)
14:21:56.038 > [PowerMeterUdpSmaHomeManager] Leistung = 362.8 (timestamp 3187955890)
14:21:56.085 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.6 (timestamp 3187955890)
14:21:56.133 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 227.3 (timestamp 3187955890)
14:21:56.180 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.9 (timestamp 3187955890)
14:21:57.037 > [PowerMeterUdpSmaHomeManager] Leistung = 408.3 (timestamp 3187956890)
14:21:57.086 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 102.1 (timestamp 3187956890)
14:21:57.133 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 272.8 (timestamp 3187956890)
14:21:57.179 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.3 (timestamp 3187956890)
14:21:58.037 > [PowerMeterUdpSmaHomeManager] Leistung = 479.5 (timestamp 3187957889)
14:21:58.092 > [PowerMeterUdpSmaHomeManager] Leistung L1 = 101.9 (timestamp 3187957889)
14:21:58.140 > [PowerMeterUdpSmaHomeManager] Leistung L2 = 343.9 (timestamp 3187957889)
14:21:58.185 > [PowerMeterUdpSmaHomeManager] Leistung L3 = 33.7 (timestamp 3187957889)

Beide Geräte sind NTP-synchronisiert und zeigen auch die gleiche Zeit an. OpenDTU-OnBattery "erhält" die Messwerte scheinbar ca. 5 Sekunden später, was ich mir einfach nicht erklären kann.

---

Aufgrund dieser Verzögeung ist eine sekunden-scharfe Steuerung für mich nicht realisierbar und wegen Verbrauchern mit Einschaltspitzen auch nicht wünschenswert. Ich nutze deshalb ein kleines Messwerte-Array. Die Länge des Arrays bestimmt, wie viele Messwerte zusammengefasst werden. Da die Messungen jede Sekunde eintreffen, entspricht eine Array-Länge von 5 also 5 Sekunden bzw. 5 Messwerten. Damit pendelt sich das System recht schnell auf die gewünschte Grundlast ein.
...trotzdem wäre es natürlich toll, wenn der DPL auch irgendwann eine frei definierbare Anzahl von Messwerten (1 bis 10 würde wohl reichen) glätten könnte.