hoylabs/OpenDTU-OnBattery #1746: [Request] Verlängerung der Abfrageintervalle für nicht erreichbare Wechselrichter

KEausW, 2025-03-20:
### Is your feature request related to a problem? Please describe.

Hallo, hier mal eine Anfrage.
Wenn von zwei Wechselrichtern einer nicht erreichbar ist,
z.B. weil dieser gerade nichts einspeist wenn eine Batterie geladen wird,
versucht die DTU trotzdem diesen Wechselrichter ständig zu erreichen.  ->

'''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
11:29:34.330 > All missing
11:29:34.428 > Nothing received, resend whole request
11:29:34.919 > [PowerMeterHttpJson] New total: -98.00
11:29:35.121 > All missing
11:29:35.169 > Nothing received, resend whole request
11:29:35.383 > Fetch inverter: 116xxxxxxx
11:29:35.435 > Queue size - NRF: 0 CMT: 3
11:29:35.912 > All missing
11:29:35.969 > Nothing received, resend whole request
11:29:36.022 > [PowerMeterHttpJson] New total: -98.00
11:29:36.702 > All missing
11:29:36.751 > Nothing received, resend whole request
11:29:36.921 > [PowerMeterHttpJson] New total: -98.00
11:29:37.454 > All missing
11:29:37.503 > Nothing received, resend count exeeded
11:29:37.947 > [PowerMeterHttpJson] New total: -100.00
11:29:38.385 > Fetch inverter: 114xxxxxxx
11:29:38.436 > Request SystemConfigPara
11:29:38.486 > Queue size - NRF: 0 CMT: 6
11:29:38.922 > [PowerMeterHttpJson] New total: -97.00
11:29:39.494 > Success
11:29:39.924 > [PowerMeterHttpJson] New total: -100.00
11:29:40.038 > Success
11:29:40.163 > All missing
11:29:40.264 > Nothing received, resend whole request
11:29:40.368 > All missing
11:29:40.418 > Nothing received, resend whole request
11:29:40.484 > All missing
11:29:40.537 > Nothing received, resend whole request
11:29:40.586 > All missing
11:29:40.636 > Nothing received, resend whole request
11:29:40.685 > All missing
11:29:40.735 > Nothing received, resend count exeeded
11:29:40.865 > All missing
11:29:40.920 > Nothing received, resend whole request
11:29:40.972 > [PowerMeterHttpJson] New total: -100.00
11:29:41.403 > All missing
11:29:41.452 > Nothing received, resend whole request
11:29:41.502 > Fetch inverter: 116xxxxxxxx
11:29:41.553 > Queue size - NRF: 0 CMT: 4
11:29:41.942 > All missing
11:29:42.110 > Nothing received, resend whole request
11:29:42.211 > [PowerMeterHttpJson] New total: -97.00
11:29:42.483 > All missing
11:29:42.535 > Nothing received, resend whole request
11:29:42.929 > [PowerMeterHttpJson] New total: -88.00
11:29:43.029 > All missing
11:29:43.084 > Nothing received, resend count exeeded
11:29:43.815 > All missing
11:29:43.870 > Nothing received, resend whole request
11:29:43.926 > [PowerMeterHttpJson] New total: -86.00

*******************************************************************************
was dann natürlich mit einem All missing endet. Das ganze macht aber die 
Regelung mit dem DPL und auch insgesamt die Abfrage des anderen Wechselrichter
sehr sehr zerrig. Dann zählt der Counter bei mir mitunter bis zu 30 Sekunden hoch.
Wie man sehen kann läuft auch die CMT Queue voll. Wenn beide WR zu erreichen
sind steht da immer nur CMT:0 oder 1
Wäre es nicht sinnvoll das für einen Wechselrichter der nach einer gewissen Anzahl
von Abfragen nicht antwortet, die Abfrage-Intervalle verlängert werden ?

Grüße Karsten

### Describe the solution you'd like

Verlängern der Abfrageintervalle von nicht erreichbaren Wechselrichtern.

tappas197112, 2025-04-09:
Bin ebenfalls von dem Problem mit den (bis zu) 30 Sekunden betroffen und habe festgestellt, dass es davon abhängt, ob noch ein Befehl ausstehend und in der Queue "hängengeblieben" ist. Ist dies nicht der Fall, wird die Zeit auch bei einem nicht erreichbaren Wechselrichter eingehalten bzw. die Abfrage-Zeit ist nur geringfügig höher.

[image]

Feuerossi, 2025-04-28:
Hallo
So ein Problem habe ich auch. An der Dtu onbattery sind 2 WR angeschlossen. Einmal 2 Panels a 340w mit Hoymiles 600, sowie ein Panel 320w mit Hoymiles 300. Der 600er lädt hauptsächlich den Marstek B2500, der 300er übernimmt die Grundlast. Steuern tut die Dtu nur den 600er. Dadurch geht der 600er oftmals in Standby. Logischerweise braucht es dann ewig, bei erhöhter Last, bis er wieder aufwacht und dazuliefert. Meistens ist dann schon alles zu spät, der Kaffeeautomat ist fertig, der Toaster und ähnliches Gerät ebenfalls.
Ich hatte auch schon einen Beitrag hier über gelegentliche Spannungsspitzen gelesen, da war ein ähnliches Problem.
Vielleicht könnte man eine Funktion einbauen, die kurzfristig einen Verbraucher simuliert, damit der WR nicht in Standby geht.
Ansonsten tolle App, weiter so 👌

WalRp, 2025-05-16:
Hallo. Habe auch das Problem mit zwei WR. WR1 an Batterie Noah 2000, WR2 an einem Solarpanel also Hybridbetrieb. Wenn WR2 abschaltet, hängt das gesamte System im DPL Betrieb. WR1 geht dann in standby. Manuell WR2 aus DPL zu deaktivieren reicht dann nicht aus. DTU muss neugestartet werden.

Ich dachte über die Lösung den WR2 aus DPL zu deaktivieren solange er nicht erreichbar ist, also zeitgesteuert oder wenn WR2 offen ist aber wenig Energie liefert z.B. ab einem Schwellenwert von 20 Watt, dann ergibt es keinen Sinn ihn zu steuern. Also dann einfach 100% setzen und warten bis er den max output überschreitet und dann wieder anfangen zu steuern.

 Dachte schon darüber mit Pi über REST API den switch WR2 in DLP zu deaktivieren wenn es eine API für DTU gäbe (gibt es die API?).
Oder wäre es möglich über MQTT die Settings in DLP zu ändern?

Momentan deaktiviere ich den WR2 (an Solar) in DLP manuell jeden Abend rechtzeitig 🫤

AndreasBoehm, 2025-05-16:
@WalRp Das passt nicht zu diesem Issue. Bitte lies dir https://github.com/hoylabs/OpenDTU-OnBattery/issues/1829 durch, dort haben wir eine Lösung erarbeitet.

AndreasBoehm, 2025-05-16:
@Feuerossi Auch dein Problem passt nicht zu diesem Issue.