hoylabs/OpenDTU-OnBattery #1750: Dynamic Power Limiter hat heute zweimal versagt

ioannis2, 2025-03-22:
### What happened?

 Der Dynamic Power Limiter hat heute zweimal versagt, als zwei Hoymiles (Hoymiles HMS-2000-4T) im Einsatz waren. 
 Dabei wurde einer der Hoymiles nicht mehr korrekt geregelt, obwohl die Solarmodule noch mehr Leistung hätten erbringen können. 
 Bei einem Hoymiles wurde die Prozentzahl nicht mehr verändert, während beim zweiten Hoymiles die Leistung auf 100 % gesetzt wurde. 
 Zudem war die Netzaufnahme über einen Zeitraum von mehr als 5 Minuten höher als 1000 W. 
 Nach einem Neustart der DTU wurde die Regelung bei beiden Hoymiles wieder korrekt durchgeführt

### To Reproduce Bug

nicht immer

### Expected Behavior

Regelung in abhängigkeit beide Hoymiles zu regeln

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-23:
Das klingt nach https://github.com/hoylabs/OpenDTU-OnBattery/issues/1675

Da ist auch ein worakround beschrieben bis ich das Problem gelöst habe.

ioannis2, 2025-03-23:
Hallo Andreas, zur Information: Ich habe das Limit auf 17,5% eingestellt, 
und das scheint zu funktionieren. Die Situation ist wie folgt: Bei höherem Verbrauch wird die Prozentzahl erhöht, 
damit die Wechselrichter mehr Leistung zulassen (alles in Ordnung). 
Allerdings wird bei geringerem Verbrauch mit dieser Einstellung das Einspeiselimit nicht mehr korrekt angepasst, 
und es bleibt auf dem Limit des Wechselrichters. Das bedeutet, wenn der Hausverbrauch niedriger ist als das Wechselrichter-Limit, 
wird der überschüssige Strom eingespeist. Dadurch erreichen wir momentan nicht die Null-Einspeisung. 
Trotzdem muss ich sagen, der Workaround ist wirklich super, tolle Arbeit!

ioannis2, 2025-03-23:
Leider hängt der Wechselrichter gerade wieder und bleibt auf dem Limit, obwohl wir mehr Strom aus dem Netz beziehen. Leider hat auch dieser Ansatz nichts gebracht.

ioannis2, 2025-03-23:
noch eine Info
Leider hängt der Wechselrichter gerade wieder und bleibt auf dem Limit, 
obwohl wir mehr Strom aus dem Netz beziehen. Allerdings wird nach dem zweiten oder dritten Mal der Aktualisierung 
wieder auf die richtige Einstellung umgeschaltet.

AndreasBoehm, 2025-03-23:
Kannst du dein Log des DPL mit verbose logging bereitstellen in dem die Probleme zu sehen sind?

ioannis2, 2025-03-23:
gerne mache ich

AndreasBoehm, 2025-03-23:
> Das bedeutet, wenn der Hausverbrauch niedriger ist als das Wechselrichter-Limit,
wird der überschüssige Strom eingespeist.

Das ist leider der nachteil des höheren Limits.
In PR https://github.com/hoylabs/OpenDTU-OnBattery/pull/1756 gibts eine Test-Version die das Problem ohne workaround lösen sollte. Bitte gerne testen und bescheid geben ob das hilft.

AndreasBoehm, 2025-03-23:
Alles weiter zu diesem Thema bitte in Issue https://github.com/hoylabs/OpenDTU-OnBattery/issues/1675 diskutieren, auch deine Logs bitte dort posten.

closed 2025-03-23 by AndreasBoehm