hoylabs/OpenDTU-OnBattery #1752: [Request] negative values for solarcharger

ves1820, 2025-03-23:
### Is your feature request related to a problem? Please describe.

i use the solarcharger setting via MQTT to get charging values from my victron multiplus II
at the moment negative values are discarded because of implausibility ...
but it is usefull for me to see if the multiplus is charging or discharging 

### Describe the solution you'd like

set negative values to color red and do'nt discard it

AndreasBoehm, 2025-03-23:
What is your usecase? To control the inverter based on the charging power of your multiplus?

ves1820, 2025-03-23:
- For me just to monitor on what value the battery is charging / discharging like on a Dashboard
- With DPL it may helpful to increase the Limit faster
- It may helpfull to know that the actual power on powermeter comes from a battery assisted system
that supports mqtt

AndreasBoehm, 2025-03-23:
> * For me just to monitor on what value the battery is charging / discharging like on a Dashboard

Sorry but thats not what the solarcharger feature is meant for. If the DTU should know about battery charging/discharging you should use the battery mqtt feature.

> * With DPL it may helpful to increase the Limit faster

What do you mean?

> * It may helpfull to know that the actual power on powermeter comes from a battery assisted system
>   that supports mqtt

Thats not what the solarcharger feature is meant for. What you need here is some piece of software that controls two different types of inverters. Why not integrate the DTU in Victron VenusOS for that?

ves1820, 2025-03-23:
the battery mqtt feature is in use - but there is only SoC, Voltage and charge Limit - no power flow ...
so i was happy to found the solarcharger setting

for DPL: if the DTU does not know there is a external battery and outpot power is at 0 - the battery is discharging while the sun is shining

AndreasBoehm, 2025-03-23:
Ah, thats true! Sorry. But i don't see why power flow is relevant for the DTU to be honest.

> for DPL: if the DTU does not know there is a external battery and outpot power is at 0 - the battery is discharging while the sun is shining

I am having a hard time to understand what you mean. Can you describe your whole PV setup please? I don't understand how everything is connected and why the DTU should know that the victron is charging the battery. Is your hoymiles WR connected to that same battery or is it solar powered? How is the solar power reaching the victron? Another WR that is different from the one that you are trying to connect to your victron?

ves1820, 2025-03-23:
i have 2 AC coupled systems the Panels connected to Hoymiles without Battery are in the Garden / the Victron with battery stays indoor.
if we switch on for example the coffe machine - the battery gets discharged while DPL thinks everything is OK and does not increase the limit

AndreasBoehm, 2025-03-23:
Thank you. Got it.

Shouldn't the DTU or rather the inverters connected to it always run at 100% to make sure the Battery gets charged? You can't really use the DPL if you have a AC coupled battery.

What would you expect the DTU to do in the case that you described?

ves1820, 2025-03-23:
we do not need much power in garden - sometimes the battery is charged after some hours and DPL reduces the power.
the victron is set to DPL Level of 20W and the DTU has DPL 0W - it works like this pretty good :-)

so if the battery charge is negative i would suggest to increase the limit (if reduced) of the inverter and the battery is happy again :-)

AndreasBoehm, 2025-03-23:
What you are trying to achieve is not possible with OpenDTU.

if you configure a solarcharger the DPL will limit the inverters based on the solar power depending on the settings, a value of zero or below value will shutdown the inverters

What you are requesting is a whole new feature and not a simple adjustment of the solarcharger (which should not be used for this scenario anyway)

How does the victron know how much it can charge? When its based on the powermeter and the DTU is also working based on the powermeter they will fight against each other and victron won't be able to only charge surplus power.

What you need is a tool that can control the inverters (via the DTU) and your victron based on solar production + powermeter readings. I would suggest to take a look at the venusOS or potentially the integrated GX in your victron.

closed 2025-03-23 by AndreasBoehm