hoylabs/OpenDTU-OnBattery #1755: MQTT "disconnected due to malformed packet"

ranma, 2025-03-23:
### What happened?

MQTT connection is unstable since updating to https://github.com/hoylabs/OpenDTU-OnBattery/commit/114c67bec1f956867d94ec7109955128611bc08c.

### To Reproduce Bug

Use 
https://github.com/hoylabs/OpenDTU-OnBattery/commit/114c67bec1f956867d94ec7109955128611bc08c with MQTT enabled and multiple data sources (inverter, battery, two MPPT).

It looks like the actual issue is that the serial connection to one of the MPPTs is not 100% stable and gets the occasional bit error not caught by the simplistic checksum.

Upon attempting to publish a topic like `homeassistant/sensor/dtu_victron_BM-602/mppt_firmware_version_fw/config`, mosquitto drops the TCP connection and any remaining data from that connection is lost.

It seems to me like either the homeassistant publishing of the MPPTs is new, or now has some cache that it didn't have before and it keeps the corrupted string in cache, causing the TCP connections to drop once a bad entry makes it into the cache.

The Victron interface handling should be more robust and escape or ignore invalid strings.

### Expected Behavior

MQTT connection does not get "disconnected due to malformed packet".

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

114c67bec1f956867d94ec7109955128611bc08c

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
Syslog:

2025-03-23T13:53:16.534125+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connecting to MQTT...
2025-03-23T13:53:16.840010+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connected to MQTT.
2025-03-23T13:53:17.454126+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnected from MQTT.
2025-03-23T13:53:17.454126+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnect reason:TCP_DISCONNECTED
2025-03-23T13:53:19.098123+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connecting to MQTT...
2025-03-23T13:53:19.552049+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Connected to MQTT.
2025-03-23T13:53:19.599050+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnected from MQTT.
2025-03-23T13:53:19.599050+00:00 OpenDTU-3480B8 OpenDTU[b9c8dcaf] Disconnect reason:TCP_DISCONNECTED

Mosquitto log:

1742744705: New connection from 192.168.2.69:54882 on port 1883.
1742744705: New client connected from 192.168.2.69:54882 as OpenDTU-3440824 (p2, c1, k15).
1742744705: Client OpenDTU-3440824 disconnected due to malformed packet.
1742744707: New connection from 192.168.2.69:54883 on port 1883.
1742744707: New client connected from 192.168.2.69:54883 as OpenDTU-3440824 (p2, c1, k15).
1742744707: Client OpenDTU-3440824 disconnected due to malformed packet.
1742744709: New connection from 192.168.2.69:54885 on port 1883.
1742744709: New client connected from 192.168.2.69:54885 as OpenDTU-3440824 (p2, c1, k15).
1742744709: Client OpenDTU-3440824 disconnected due to malformed packet.
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-23:
More details about your MPTT serial number issue can be found here: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1692

The MQTT issue is of course still valid, thank you.

ranma, 2025-03-23:
> More details about your MPTT serial number issue can be found here: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1692

Thanks, I only searched for MQTT, so I didn't see the "phantom devices" bug.

AndreasBoehm, 2025-03-25:
Can we resolve this issue now that https://github.com/hoylabs/OpenDTU-OnBattery/pull/1757 has been merged?

ranma, 2025-03-25:
> Can we resolve this issue now that [#1757](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1757) has been merged?

So its not fully solved, but should reduce the frequency. It ran into MQTT disconnects again today, this time because '\t' became part of the serial number, which is one of the allowed characters (and I don't want to make the parsing more complex by introducing a state machine).

Thus this probably needs some additional validation in the homeassistant topic publisher still.

AndreasBoehm, 2025-03-25:
IMHO this should be solved at the source of the problem, VE.Direct and not on the MQTT side. Because even if MQTT ignores data it can't handle the problem of phantom devices would still be present within the DTU, thats why i feel like we should close this issue.

ranma, 2025-03-26:
Should be resolved by https://github.com/hoylabs/OpenDTU-OnBattery/commit/6fbc99bd1bfe6463d80e0ec68076d8f79ed9d7cc

closed 2025-03-26 by ranma