hoylabs/OpenDTU-OnBattery #1759: [Bug] DPL desperately tries to use solar-powered inverter at dawn

schlimmchen, 2025-03-23:
### What happened?

User C0debreaker in discord noticed a very strange problem in their setup: At dawn, the battery-powered inverters would be **dis**abled, even though the battery has power left. This happened multiple times in the last days around dawn, right about when a solar-powered inverter switched to standby.

### To Reproduce Bug

Need at least two inverters, one solar-powered, at least one other battery-powered, battery discharge cycle active, dawn forces solar-powered inverter into standby.

### Expected Behavior

The battery-powered inverters shall cover the household consumption, regardless of whether or not solar-powered inverters are in standby or not.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
18:36:57.567 > [DPL] up 86 s, next inverter restart at 19500 s (set to 0)
18:36:57.626 > [DPL] targeting 10 W, base load is 100 W, power meter reads -164.0 W (valid)
18:36:57.725 > [DPL] inverter **Inverter 1** is behind power meter producing 611 W
18:36:57.799 > [DPL] inverter **Inverter 2** is behind power meter producing 0 W
18:36:57.870 > [DPL] inverter **Inverter 3** is behind power meter producing 1 W
18:36:57.946 > [DPL] requesting 438 W from 1 solar-powered inverter currently producing 1 W (diff 437 W, hysteresis 10 W)
18:36:58.035 > [DPL] will cover 1 W using 1 solar-powered inverter
18:36:58.088 > [DPL] skipping **Inverter 2** as it is not reachable
18:36:58.151 > [DPL] requesting 437 W from 1 smart-buffer-powered inverter currently producing 611 W (diff -174 W, hysteresis 10 W)
18:36:58.222 > [DPL inverter **Inverter 1**]:
18:36:58.279 >     expected AC power per MPPT 294 W
18:36:58.343 >     MPPT-a AC power 306 W
18:36:58.989 >     MPPT-b AC power 306 W
18:36:59.075 > [DPL] will cover 426 W using 1 smart-buffer-powered inverter
18:36:59.133 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 11 W AC
18:36:59.189 > [DPL inverter **Inverter 1**]:
18:36:59.289 >     smart-buffer-powered, producing 611 W
18:36:59.347 >     lower/current/upper limit: 26/600/600 W, output capability: 600 W
18:36:59.415 >     sending commands enabled, reachable, eligible
18:36:59.469 >     max reduction production/standby: 585/611 W, max increase: 65525 W
18:36:59.573 >     target limit/output/state: 426 W (update)/426 W/production, 0 update timeouts
18:36:59.725 >     MPPTs AC power: a: 306 W b: 306 W
18:36:59.842 > [DPL inverter **Inverter 2**]:
18:36:59.964 >     smart-buffer-powered, standing by at 0 W
18:37:00.091 >     lower/current/upper limit: 26/0/600 W, output capability: 0 W
18:37:00.202 >     sending commands enabled, offline, disqualified (unreachable)
18:37:00.310 >     max reduction production/standby: 0/0 W, max increase: 0 W
18:37:00.485 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
18:37:00.593 >     MPPTs AC power: a: 0 W b: 0 W
18:37:00.670 > [DPL inverter **Inverter 3**]:
18:37:00.757 >     solar-powered, producing 1 W
18:37:00.823 >     lower/current/upper limit: 26/25/600 W, output capability: 600 W
18:37:00.882 >     sending commands enabled, reachable, eligible
18:37:00.929 >     max reduction production/standby: 0/0 W, max increase: 0 W
18:37:00.995 >     target limit/output/state: -1 W (unchanged)/1 W/unchanged, 0 update timeouts
18:37:02.140 >     MPPTs AC power: a: 1 W b: 1 W
18:37:02.197 > [DPL inverter **Inverter 1**]: sending limit of 71.0 % (426 W respectively), max output is 600 W
18:37:02.284 > [DPL] waiting for a start/stop/restart/limit command to complete
18:37:08.268 > [DPL inverter **Inverter 1**]: actual limit is 71.0 % (426 W respectively), effective 9933 ms after update started, requested were 71.0 %
18:37:08.323 > [DPL] waiting for sufficiently recent inverter data
18:37:12.301 > [DPL] waiting for sufficiently recent inverter data
18:37:15.795 > [DPL] waiting for sufficiently recent power meter reading
18:37:21.117 > [DPL] up 115 s, next inverter restart at 19500 s (set to 0)
18:37:21.312 > [DPL] targeting 10 W, base load is 100 W, power meter reads 0.0 W (valid)
18:37:21.432 > [DPL] inverter **Inverter 1** is behind power meter producing 437 W
18:37:21.494 > [DPL] inverter **Inverter 2** is behind power meter producing 0 W
18:37:21.577 > [DPL] inverter **Inverter 3** is behind power meter producing 0 W
18:37:21.639 > [DPL] requesting 427 W from 1 solar-powered inverter currently producing 0 W (diff 427 W, hysteresis 10 W)
18:37:21.695 > [DPL] will cover 427 W using 1 solar-powered inverter
18:37:21.780 > [DPL] skipping **Inverter 2** as it is not reachable
18:37:21.861 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 437 W (diff -437 W, hysteresis 10 W)
18:37:21.925 > [DPL] will cover 0 W using 1 smart-buffer-powered inverter
18:37:21.989 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
18:37:22.077 > [DPL inverter **Inverter 1**]:
18:37:22.155 >     smart-buffer-powered, producing 437 W
18:37:22.225 >     lower/current/upper limit: 26/426/600 W, output capability: 600 W
18:37:22.285 >     sending commands enabled, reachable, eligible
18:37:22.340 >     max reduction production/standby: 411/437 W, max increase: 163 W
18:37:22.420 >     target limit/output/state: -1 W (unchanged)/0 W/standby, 0 update timeouts
18:37:22.521 >     MPPTs AC power: a: 219 W b: 219 W
18:37:22.587 > [DPL inverter **Inverter 2**]:
18:37:22.717 >     smart-buffer-powered, standing by at 0 W
18:37:22.758 >     lower/current/upper limit: 26/0/600 W, output capability: 0 W
18:37:22.811 >     sending commands enabled, offline, disqualified (unreachable)
18:37:22.897 >     max reduction production/standby: 0/0 W, max increase: 0 W
18:37:22.979 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
18:37:23.029 >     MPPTs AC power: a: 0 W b: 0 W
18:37:23.093 > [DPL inverter **Inverter 3**]:
18:37:23.172 >     solar-powered, standing by at 0 W
18:37:23.241 >     lower/current/upper limit: 26/25/600 W, output capability: 600 W
18:37:23.301 >     sending commands enabled, reachable, eligible
18:37:23.360 >     max reduction production/standby: 0/0 W, max increase: 600 W
18:37:23.442 >     target limit/output/state: 427 W (update)/427 W/production, 0 update timeouts
18:37:24.664 >     MPPTs AC power: a: 0 W b: 0 W
18:37:24.838 > [DPL inverter **Inverter 1**]: Stopping inverter...
18:37:24.888 > [DPL inverter **Inverter 3**]: sending limit of 71.2 % (427 W respectively), max output is 600 W
18:37:25.063 > [DPL] waiting for a start/stop/restart/limit command to complete
18:37:30.104 > [DPL] waiting for a start/stop/restart/limit command to complete
18:37:30.986 > [DPL inverter **Inverter 3**]: actual limit is 71.1 % (427 W respectively), effective 15768 ms after update started, requested were 71.2 %
18:37:31.051 > [DPL inverter **Inverter 3**]: Starting inverter...
18:37:33.312 > [DPL] waiting for a start/stop/restart/limit command to complete
18:37:41.768 > [DPL inverter **Inverter 3**]: timeout (1 in succession), state transition pending: yes, limit pending: no
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-03-23:
As we (in general) cannot know how much power solar-powered inverters in standby could produce if they are woken up, we assume they can produce as much as their rating. The rational is that we'll know in the next round whether or not the inverter can actually deliver that amount of power.

However, at dawn, this assumption is... wrong. It leads to the DPL thinking that the solar-powered inverter will be covering all or a significant portion of the household consumption, leading to reduced power requests from the battery-powered inverters.

Even worse: We desperately try to set a new limit at that inverter, leading to an intentional restart of the ESP32 🙊 

I suggest the following: solar-powered inverters are non-eligible at night. Assuming that this state of forced standby **and** reachability only occurs when it is already "night", this will resolve this issue.

schlimmchen, 2025-03-23:
The same problem may happen at dusk. Hoping that `SunPositon.isDayPeriod()` is only true once solar-powered inverters are able to produce at least one Watt, i.e., they can be producing (not in standby), the respective test should also resolve the issue at dusk.

However, I am not entirely confident that it will in all cases... What about particularly cloudy dusks?

@spcqike Have you never experienced this issue?!

schlimmchen, 2025-03-23:
Another indicator that this should work as expected: The solar-powered inverter at my parents' house is configured to not receive commands and to not be polled at night, so ist became "unreachable" while still producing:

[image]

AndreasBoehm, 2025-03-23:
Does this issue only occur when „request data at night“ and/or „send commands at night“ are enabled for solar inverters? If not, we should make sure that the DPL respects those settings.

schlimmchen, 2025-03-23:
> Does this issue only occur when „request data at night“ and/or „send commands at night“ are enabled for solar inverters?

Yes. If those switches are disabled, the inverter is unreachable as soon as it is "night" and hence the inverter is non-eligible and the DPL does nothing with it.

spcqike, 2025-03-23:
Sry ich kann die Frage grad nicht nachvollziehen, deswegen schreib ich einfach auf deutsch. 

Was genau soll oder kann ich bemerkt haben? 

Ich nutze den DPL seit Oktober oder November nicht. Da da eh nicht viel runter kam und ich keinen Akku habe. Das bisschen Überschuss sollte da eingespeist werden und irgendwie zur Energiewende beitragen (: ich wollt morgen die neue Version von @AndreasBoehm installieren und testen, wie sich das mit dem DPL und Limits <15% verhält. Aber auch das noch nicht dauerhaft aktivieren. Erst, wenn der Strompreis wieder regelmäßig gegen 0 geht und das Limit auch tatsächlich notwendig wird. 

Und ich habe erst seit Dezember den zweiten WR mitlaufen 

Ich erinnere mich aber, dass ich letztes Jahr „wechselrichter Starts“ bei >20W gesehen habe. Das konnte ich mir erst nicht erklären, es hing aber damit zusammen, dass die DTU noch dachte es sei Nacht. Und da hat sie den WR einfach noch nicht abgefragt. Normalerweise startet der bereits bei 3-5W

Ich hatte da mal mit der Dämmerung gespielt irgendwann. Hab sie aber seither auf der Standard  (90.8) 

Meinst du sowas? 

Wenn nicht, was genau meinst du? Dann kann ich gern versuchen etwas nachzustellen oder in den Aufzeichnungen zu suchen.

schlimmchen, 2025-03-23:
> Ich erinnere mich aber, dass ich letztes Jahr „wechselrichter Starts“ bei >20W gesehen habe.

Klingt als hättest du "Abfrage und Kommandos senden bei Nacht" **aus**geschaltet, und dann siehst du das Problem nicht.

Aber andererseits hatte ich auch einen Denkfehler, denn wenn du nur solarbetriebene Inverter nutzt, siehst du es so oder so nicht... Entschuldige, hab dich umsonst geweckt 😉

spcqike, 2025-03-23:
> Klingt als hättest du "Abfrage und Kommandos senden bei Nacht" ausgeschaltet

Korrekt. Habe ich auch. Um so erstaunter bin ich dass die DTU dennoch Mitternacht versucht den einen inverter zu erreichen :) Aber #1700 wurde ja gekonnt ignoriert ;D (oder was auch immer sie da Mitternacht und auch so gelegentlich nachts macht) 

Ne dann kann ich hier wohl nicht viel beitragen (: 

> Entschuldige, hab dich umsonst geweckt 

Ganz so weit ist es noch nicht (: Aber bald 👍

PR #1760 (2025-03-29): Fix: solar-powered inverters are non-eligible at nighttime
we currently rely on solar-powered inverters becoming unreachable during the night. they eventually do, but until they do, there is a (potentially) long stretch of time where they are unable to produce any power, i.e., they are in standby, but still reachable. in that case we don't know what the inverter would produce if it was woken up, so we assume it can produce its rated power. that is nonsense when it is dawn.

it seems that it is already "night" according to SunPosition when the inverter switches to standby on its own, we add tha [... cut]
closed 2025-03-29 by schlimmchen