hoylabs/OpenDTU-OnBattery #1761: [Bug] DPL: Undeflow in smart-buffer inverter maxIncrease() calculcation

schlimmchen, 2025-03-24:
### What happened?

Info from C0debreaker on discord.

### To Reproduce Bug

Not sure.

### Expected Behavior

Prevent erreneous maxIncrease() return values.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10 ?

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
10:44:13.801 > [DPL] up 56834 s, next inverter restart at 104594 s (set to 0)
10:44:13.859 > [DPL] targeting 10 W, base load is 100 W, power meter reads 759.0 W (valid)
10:44:13.915 > [DPL] inverter **Inverter 1** is behind power meter producing 612 W
10:44:14.123 > [DPL] inverter **Inverter 2** is behind power meter producing 327 W
10:44:14.178 > [DPL] inverter **Inverter 3** is behind power meter producing 138 W
10:44:14.236 > [DPL] requesting 1800 W from 1 solar-powered inverter currently producing 138 W (diff 1662 W, hysteresis 10 W)
10:44:14.290 > [DPL] will cover 138 W using 1 solar-powered inverter
10:44:14.534 > [DPL] requesting 1662 W from 2 smart-buffer-powered inverters currently producing 939 W (diff 723 W, hysteresis 10 W)
10:44:14.605 > [DPL inverter **Inverter 1**]:
10:44:14.668 >     expected AC power per MPPT 294 W
10:44:14.726 >     MPPT-a AC power 306 W
10:44:14.884 >     MPPT-b AC power 306 W
10:44:14.945 > [DPL] will cover 927 W using 2 smart-buffer-powered inverters
10:44:15.042 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 735 W AC
10:44:15.097 > [DPL inverter **Inverter 1**]:
10:44:15.351 >     smart-buffer-powered, producing 612 W
10:44:15.416 >     lower/current/upper limit: 26/600/600 W, output capability: 600 W
10:44:15.490 >     sending commands enabled, reachable, eligible
10:44:15.578 >     max reduction production/standby: 586/612 W, max increase: 65524 W
10:44:15.638 >     target limit/output/state: 600 W (update)/600 W/production, 0 update timeouts
10:44:15.692 >     MPPTs AC power: a: 306 W b: 306 W
10:44:15.965 > [DPL inverter **Inverter 2**]:
10:44:16.032 >     smart-buffer-powered, producing 327 W
10:44:16.088 >     lower/current/upper limit: 26/318/600 W, output capability: 600 W
10:44:16.148 >     sending commands enabled, reachable, eligible
10:44:16.216 >     max reduction production/standby: 301/327 W, max increase: 273 W
10:44:16.268 >     target limit/output/state: -1 W (unchanged)/327 W/unchanged, 7 update timeouts
10:44:16.331 >     MPPTs AC power: a: 164 W b: 164 W
10:44:16.379 > [DPL inverter **Inverter 3**]:
10:44:16.426 >     solar-powered, producing 138 W
10:44:16.476 >     lower/current/upper limit: 26/600/600 W, output capability: 600 W
10:44:16.780 >     sending commands enabled, reachable, eligible
10:44:16.834 >     max reduction production/standby: 112/112 W, max increase: 0 W
10:44:16.996 >     target limit/output/state: -1 W (unchanged)/138 W/unchanged, 9 update timeouts
10:44:17.104 >     MPPTs AC power: a: 103 W b: 36 W
10:44:17.222 > [DPL inverter **Inverter 1**]: sending limit of 100.0 % (600 W respectively), max output is 600 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-03-24:
@AndreasBoehm Can cou have a look? As the logs don't lie, there has to be some kind of issue in `getMaxIncrease()`.

AndreasBoehm, 2025-03-24:
Sure @schlimmchen, I will take care of this.

closed 2025-03-25 by schlimmchen