hoylabs/OpenDTU-OnBattery #1762: [Bug] DPL: Underflow in solar power calculation

schlimmchen, 2025-03-24:
### What happened?

See #1733.

### To Reproduce Bug

Not sure.

### Expected Behavior

Calculation of available solar power must be robust against underflows.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10 ?

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
[DPL] up 666549 s, next inverter restart at 730908 s (set to 0)
[DPL] battery interface enabled, SoC 33.0 % (ignored), age 2 s (valid)
[DPL] BMS: 39.07 V, MPPT: 39.09 V, inverter 114181870469: 39.10 
[DPL] battery voltage 39.07 V, load-corrected voltage 39.07 V @ 0 W, factor 0.00400 1/A
[DPL] battery discharge restricted, start 42.50 V or 80 %, stop 39.00 V or 20 %
[DPL] full solar-passthrough dormant, start 47.20 V or 100 %, stop 46.95 V
[DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
[DPL] total max AC power is 800 W, conduction losses are 5 %
[DPL] targeting -20 W, base load is 100 W, power meter reads 106.0 W (valid)
[DPL] inverter 114181870469 is behind power meter producing 0 W
[DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 126 W AC
[DPL] requesting 0 W from 1 battery-powered inverter currently producing 0 W (diff 0 W, hysteresis 30 W)
[DPL inverter 114181870469]:
    battery-powered, standing by at 0 W
    lower/current/upper limit: 50/119/700 W, output capability: 700 W
    sending commands enabled, reachable, eligible
    max reduction production/standby: 0/0 W, max increase: 700 W
    target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
    MPPTs AC power: a: 0 W b: 0 W
[DPL] up 666550 s, next inverter restart at 730908 s (set to 0)
[DPL] battery interface enabled, SoC 33.0 % (ignored), age 4 s (valid)
[DPL] BMS: 39.07 V, MPPT: 39.08 V, inverter 114181870469: 39.10 
[DPL] battery voltage 39.07 V, load-corrected voltage 39.07 V @ 0 W, factor 0.00400 1/A
[DPL] battery discharge restricted, start 42.50 V or 80 %, stop 39.00 V or 20 %
[DPL] full solar-passthrough dormant, start 47.20 V or 100 %, stop 46.95 V
[DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
[DPL] total max AC power is 800 W, conduction losses are 5 %
[DPL] targeting -20 W, base load is 100 W, power meter reads 111.0 W (valid)
[DPL] inverter 114181870469 is behind power meter producing 0 W
[DPL] battery allowance is 0/0 W DC/AC, solar power is 65535/59145 W DC/AC, requested are 131 W AC
[DPL] requesting 131 W from 1 battery-powered inverter currently producing 0 W (diff 131 W, hysteresis 30 W)
[DPL] will cover 131 W using 1 battery-powered inverter
[DPL inverter 114181870469]:
    battery-powered, standing by at 0 W
    lower/current/upper limit: 50/119/700 W, output capability: 700 W
    sending commands enabled, reachable, eligible
    max reduction production/standby: 0/0 W, max increase: 700 W
    target limit/output/state: 131 W (update)/131 W/production, 0 update timeouts
    MPPTs AC power: a: 0 W b: 0 W
[DPL inverter 114181870469]: sending limit of 18.7 % (131 W respectively), max output is 700 W
[DPL] waiting for a start/stop/restart/limit command to complete
[DPL inverter 114181870469]: actual limit is 18.7 % (131 W respectively), effective 2040 ms after update started, requested were 18.7 %
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-24:
@spiderhl53 Dein Problem passt nicht zu diesem Issue. Bitte leg einen neuen Issue oder eine Diskussion an, dann schauen wir uns das Problem gerne an.

ranma, 2025-03-25:
Ah, ich glaube den Bug hier sehe ich auch, muss nochmal ein rebase auf die aktuellste Version machen:
```
[DPL] battery allowance is 0/0 W DC/AC, solar power is 65535/62258 W DC/AC, requested are 183 W AC
                                                                                                   ^^^^^-1
[DPL] requesting 183 W from 1 battery-powered inverter currently producing 0 W (diff 183 W, hysteresis 20 W)
[DPL] will cover 183 W using 1 battery-powered inverter
```

closed 2025-03-25 by schlimmchen

schlimmchen, 2025-03-25:
FYI: Die Gesamtausgangsleistung, die ein VE.Direct Laderegler meldet, darf in der Tat negativ sein, das bedeutet nämlich, dass Strom aus der Batterie in den Lastausgang des Ladereglers fließt, was ein gültiger Zustand ist.

PR #1765 (2025-03-25): Fix: DPL: prevent underflow in solar charger output calculation
For reasons/explanations, see new comments in changeset.

Closes #1762.