hoylabs/OpenDTU-OnBattery #1771: [Request] DPL: Don't request power from non-solar inverters when battery requested immediate charging

AndreasBoehm, 2025-03-25:
### Is your feature request related to a problem? Please describe.

I discharged my battery to 5% by using a way too low stop voltage which made the battery return `true` for `getImmediateChargingRequest()`

When the mppts started producing power the DPL enabled the inverter for solar-passthrough and the battery wasn't charged.

### Describe the solution you'd like

Respect `getImmediateChargingRequest()` when calculating the available solar power for solar-passthrough.

AndreasBoehm, 2025-03-25:
Whats your opinion on that @schlimmchen?

schlimmchen, 2025-03-25:
> Whats your opinion on that [@schlimmchen](https://github.com/schlimmchen)?

Yup.

However, documenting this priority would be in order :wink: Would you, please? The AI can help you add a sentence or two that will make it clear that solar-passthrough is essentially off if an immediate charging request is recognized by the battery.

closed 2025-03-25 by schlimmchen

schlimmchen, 2025-03-25:
I know you better than to assume that this is some passive-aggressive talk, but it kind of reads like it 😘

I don't think it's documented anywhere that the AC charger does overwrite the DC power bus usage. And I am also not sure where to put it.

Regarding the immediate charging override, I would put it in a `!!!notice` box next to the description of [solar passthrough ON](https://opendtu-onbattery.net/firmware/solar_passthrough/#solar-passthrough-is-on). I guess the AC charger override thingy can go there as well?

AndreasBoehm, 2025-03-25:
Sorry if it sounded like that, i was not intended to be passive-aggressive.

I am really only looking for a good place to put it :)

Thanks for the hint, will put it there and include the AC charger override as well.

PR #1776 (2025-03-25): Feature: DPL: respect immediate charging request
When the battery is requesting immediate charging we don't take any power from the DC bus and let the battery charge.

Closes #1771