hoylabs/OpenDTU-OnBattery #1772: [Request] DPL: Power limit too sticky with Zendure CT Mode

bolzerrr, 2025-03-25:
### Is your feature request related to a problem? Please describe.

I want to use the Zendure CT Mode to control whats put into the house or the battery. I also have a panel outside Zendure directly attached to the inverter. Therefore i want to use the DPL to have a fixed max output of the combined production. Right now the limit is often sticky below 100% even tho the demand is higher and Zendure would have battery capacity to fullfill.

### Describe the solution you'd like

Right now i use HA to reset the limit to 100% as soon as the Grid power is > 50 for a minute and the limit is not 100 already. It would be nice if the DPL would have a option that would act similar or increase the limit to give Zendure the chance to also increase output till the demand is covered or 100% is reached.

AndreasBoehm, 2025-03-25:
Which version are you using?
Did you configure your inverter with the powersource `Smart-Buffer` in the DPL settings?

Did i get it right that you have one inverter with 2 inputs, 1 input connected to a solar panel and 1 input connected to the zendure battery?

bolzerrr, 2025-03-25:
Which version are you using?
Latest release: 2025.02.10

Did you configure your inverter with the powersource Smart-Buffer in the DPL settings?
Yes

Did i get it right that you have one inverter with 2 inputs, 1 input connected to a solar panel and 1 input connected to the zendure battery?
Yes

AndreasBoehm, 2025-03-25:
Thanks for the information.

Which inverter are you using and what firmware version is the inverter running?

Can you provide a log of the DPL with verbose logging enabled that shows when the limit is 'sticky'?

Without having a log i would assume that you experience the following:
- Inverter set to 400 W Limit to match the grid meter value
- Inverter is producing 200 W because the solar panel is not delivering anything and the 200 W are coming from the battery
Is that correct?

AndreasBoehm, 2025-03-25:
Looking forward to the log.

I am very confused that the DPL is not setting the limit to 100% when needed.

Do you have multiple inverters in your system? Asking because you mentioned that grid is highly negative when the battery is full. But if controlled by the DPL this would only happen when you have your battery integrated in the DTU, do you?

bolzerrr, 2025-03-25:
No just a single one. Battery is not setup in the DTU.

When battery is fully charged ZD does bypass all input, plus input from the direct connected panels easily covers the demand and results in high negative values.

AndreasBoehm, 2025-04-05:
Please try the test-version linked in the description of this pull-request and report back if this fixes your issue.
https://github.com/hoylabs/OpenDTU-OnBattery/pull/1756

bolzerrr, 2025-04-08:
Hi,
sorry was busy the last weeks. Today i updated using the PR build and let you know if things improve. 
Thank you!

AzraelsDisk, 2025-04-09:
Setup:
- HM-600 1 panel direct attached
- HM-1500 1 MPPT connected to Zendure HUB1200 (DPL-Mode)
- Data for Battery coming via MQTT
- Shelly EM3 on MQTT

I have just installed it and when I active the "Verschattetet/Ungenutzte Eingänge ausgleichen" DPL mode extends the power to max 1500W while there is no request for it 

When I deactive "Verschattetet/Ungenutzte Eingänge ausgleichen" then the delivers around 40 - 50% of the required power to home

I hope someone can help me

AndreasBoehm, 2025-04-09:
@AzraelsDisk which OpenDTU version are you using?

If you dont use all inputs of your inverter it can not produce the expected output power as the limit is equaly divided between all inputs/mppts.

Why do you only use one mppt for the connection to the zendure?

AzraelsDisk, 2025-04-10:
@AndreasBoehm 
> https://github.com/AzraelsDisk) which OpenDTU version are you using?

Sorry, I was not specific enough: 2025.02.10 out of your link https://github.com/hoylabs/OpenDTU-OnBattery/pull/1756
> If you dont use all inputs of your inverter it can not produce the expected output power as the limit is equaly divided between all inputs/mppts.

Copy, that's why I want to use the option for unused inputs, which leads to an total overflow of energy
> Why do you only use one mppt for the connection to the zendure?

The output of the Zendure System is much more stable if it only has to deal with only one mppt instead ob 2 mppt fighting against each other ;) and the HM-1500 is capable to handle the power on only one mppt

[image]

[image]

total escalation....

AndreasBoehm, 2025-04-10:
@AzraelsDisk hm?

AzraelsDisk, 2025-04-10:
SDK-Version
v4.4.7-dirty
Konfigurationsversion
0.1.29-OnBattery-5
Firmwareversion / git Hash
2025.02.10
Firmware Branch
master
PIO Umgebung
generic_esp32s3_usb
Firmware-Aktualisierung

Reset Grund CPU 0
Software reset CPU

AndreasBoehm, 2025-04-10:
> Firmwareversion / git Hash
> 2025.02.10

You did not install the test version :)

AzraelsDisk, 2025-04-10:
Strange… can you guide me directly? Deep link maybe?

AndreasBoehm, 2025-04-10:
Download this [file](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259/artifacts/2916954283), unzip it (maybe you forgot to unzip it?), upload as update at Settings -> Firmware upgrade

AzraelsDisk, 2025-04-10:
Just tried your link on my mobile - but I got an 404

AndreasBoehm, 2025-04-10:
Try this one: https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259

AzraelsDisk, 2025-04-10:
Sorry, noob question… there is no file to Download do I need to build it?

AndreasBoehm, 2025-04-10:
[image]

AzraelsDisk, 2025-04-10:
I was not logged in... on github 
now it looks like

[image]

AndreasBoehm, 2025-04-16:
@bolzerrr I assume that everything is working fine because you did not report anything after 1 week :) Is that correct?

@AzraelsDisk Everything working fine on your side?

AzraelsDisk, 2025-04-16:
Everything working very well! 

I just wanted to wait for the Hub to switch to bypass and then provide more information…. But due to perfect steering of DPL the power is used in the house during the day! 

Thanks a lot

bolzerrr, 2025-04-22:
Hey,
i was on vacation, but what i can tell already is that my homeassistant reset script did execute way less then before. Will observe the rare cases from now.

Thank you!

closed 2025-05-21 by github-actions[bot]