hoylabs/OpenDTU-OnBattery #1774: DPL with Battery-powered Inverter is still often ignoring start threshold

ranma, 2025-03-25:
### What happened?

Despite the start threshold not yet being reached, the DPL is increasing the inverter power limit above the solar production by a largeish amount, causing the battery to be drained to the stop threshold.

[image]

### To Reproduce Bug

Battery use start threshold set to `50%`, and stop threshold set to `20%`, power source set to `Battery`,  wiring Panels -> MPPT -> Battery -> Inverter.

### Expected Behavior

Inverter will not use more than solar production until start threshold is reached.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

273cf67

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
2025-03-25T11:56:44.292344+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] up 51471 s, it is day, next inverter restart at 105718 s (set to 4)
2025-03-25T11:56:44.293251+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery interface enabled, SoC 20.1 % (used), age 0 s (valid)
2025-03-25T11:56:44.294248+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] BMS: 48.86 V, MPPT: 48.43 V, inverter 114184614039: 48.80 
2025-03-25T11:56:44.295271+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery voltage 48.86 V, load-corrected voltage 49.17 V @ 308 W, factor 0.00100 1/A
2025-03-25T11:56:44.295271+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery discharge restricted, start 51.50 V or 50 %, stop 45.94 V or 20 %
2025-03-25T11:56:44.298297+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] full solar-passthrough dormant, start 53.20 V or 97 %, stop 52.00 V
2025-03-25T11:56:44.298297+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night enabled a>
2025-03-25T11:56:44.298297+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] total max AC power is 800 W, conduction losses are 0 %
2025-03-25T11:56:44.299253+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] targeting 10 W, base load is 200 W, power meter reads 277.2 W (valid)
2025-03-25T11:56:44.300241+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery allowance is 0/0 W DC/AC, solar power is 331/314 W DC/AC, requested are 575 W >
2025-03-25T11:56:44.301240+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] requesting 314 W from 1 battery-powered inverter currently producing 308 W (diff 6 W, >
2025-03-25T11:56:44.304627+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL inverter 114184614039]:
2025-03-25T11:56:44.304627+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    battery-powered, producing 308 W, output included in power meter reading
2025-03-25T11:56:44.304627+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    lower/current/upper limit: 26/300/800 W, output capability: 800 W
2025-03-25T11:56:44.304627+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    sending commands enabled, reachable, eligible
2025-03-25T11:56:44.305168+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    max reduction production/standby: 282/308 W, max increase: 500 W
2025-03-25T11:56:44.306158+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    target limit/output/state: -1 W (unchanged)/308 W/unchanged, 0 update timeouts
2025-03-25T11:56:44.307154+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    MPPTs AC power: a: 154 W b: 154 W
2025-03-25T11:56:45.322300+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] up 51472 s, it is day, next inverter restart at 105718 s (set to 4)
2025-03-25T11:56:45.323168+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery interface enabled, SoC 20.1 % (used), age 0 s (valid)
2025-03-25T11:56:45.324151+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] BMS: 48.86 V, MPPT: 48.43 V, inverter 114184614039: 48.80 
2025-03-25T11:56:45.324151+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery voltage 48.86 V, load-corrected voltage 49.17 V @ 308 W, factor 0.00100 1/A
2025-03-25T11:56:45.325201+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery discharge restricted, start 51.50 V or 50 %, stop 45.94 V or 20 %
2025-03-25T11:56:45.326180+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] full solar-passthrough dormant, start 53.20 V or 97 %, stop 52.00 V
2025-03-25T11:56:45.327168+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night enabled a>
2025-03-25T11:56:45.328181+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] total max AC power is 800 W, conduction losses are 0 %
2025-03-25T11:56:45.329193+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] targeting 10 W, base load is 200 W, power meter reads 277.2 W (valid)
2025-03-25T11:56:45.330200+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery allowance is 0/0 W DC/AC, solar power is 331/314 W DC/AC, requested are 575 W >
2025-03-25T11:56:45.331185+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] requesting 314 W from 1 battery-powered inverter currently producing 308 W (diff 6 W, >
2025-03-25T11:56:45.332170+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL inverter 114184614039]:
2025-03-25T11:56:45.333179+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    battery-powered, producing 308 W, output included in power meter reading
2025-03-25T11:56:45.334179+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    lower/current/upper limit: 26/300/800 W, output capability: 800 W
2025-03-25T11:56:45.335166+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    sending commands enabled, reachable, eligible
2025-03-25T11:56:45.343251+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    max reduction production/standby: 282/308 W, max increase: 500 W
2025-03-25T11:56:45.343251+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    target limit/output/state: -1 W (unchanged)/308 W/unchanged, 0 update timeouts
2025-03-25T11:56:45.343251+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    MPPTs AC power: a: 154 W b: 154 W
2025-03-25T11:56:45.855213+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] Fetch inverter: 114184614039
2025-03-25T11:56:45.857218+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] Queue size - NRF: 1 CMT: 0
2025-03-25T11:56:46.352358+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] up 51473 s, it is day, next inverter restart at 105718 s (set to 4)
2025-03-25T11:56:46.353283+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery interface enabled, SoC 20.1 % (used), age 1 s (valid)
2025-03-25T11:56:46.356383+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] BMS: 48.87 V, MPPT: 48.43 V, inverter 114184614039: 48.80 
[... 98 lines cut ...]
2025-03-25T11:57:02.096966+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery interface enabled, SoC 20.1 % (used), age 0 s (valid)
2025-03-25T11:57:02.099977+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] BMS: 48.67 V, MPPT: 48.18 V, inverter 114184614039: 48.50 
2025-03-25T11:57:02.099977+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery voltage 48.67 V, load-corrected voltage 49.16 V @ 488 W, factor 0.00100 1/A
2025-03-25T11:57:02.099977+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery discharge restricted, start 51.50 V or 50 %, stop 45.94 V or 20 %
2025-03-25T11:57:02.105050+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] full solar-passthrough dormant, start 53.20 V or 97 %, stop 52.00 V
2025-03-25T11:57:02.105050+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night enabled a>
2025-03-25T11:57:02.105050+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] total max AC power is 800 W, conduction losses are 0 %
2025-03-25T11:57:02.105050+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] targeting 10 W, base load is 200 W, power meter reads 1509.2 W (valid)
2025-03-25T11:57:02.109003+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] battery allowance is 0/0 W DC/AC, solar power is 516/490 W DC/AC, requested are 800 W >
2025-03-25T11:57:02.109003+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL] requesting 490 W from 1 battery-powered inverter currently producing 488 W (diff 2 W, >
2025-03-25T11:57:02.109003+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b] [DPL inverter 114184614039]:
2025-03-25T11:57:02.109003+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    battery-powered, producing 488 W, output included in power meter reading
2025-03-25T11:57:02.109003+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    lower/current/upper limit: 26/476/800 W, output capability: 800 W
2025-03-25T11:57:02.110960+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    sending commands enabled, reachable, eligible
2025-03-25T11:57:02.110960+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    max reduction production/standby: 462/488 W, max increase: 324 W
2025-03-25T11:57:02.113972+00:00 OpenDTU-3480B8 OpenDTU[1ecf882b]    target limit/output/state: -1 W (unchanged)/488 W/unchanged, 0 update timeouts
[... cut ...]    MPPTs AC power: a: 245 W b: 244 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

ranma, 2025-03-25:
This is when the limit got increased unexpectedly:
```
11:56:46 [DPL] up 51473 s, it is day, next inverter restart at 105718 s (set to 4)
11:56:46 [DPL] battery interface enabled, SoC 20.1 % (used), age 1 s (valid)
11:56:46 [DPL] BMS: 48.87 V, MPPT: 48.43 V, inverter 114184614039: 48.80 
11:56:46 [DPL] battery voltage 48.87 V, load-corrected voltage 49.18 V @ 308 W, factor 0.00100 1/A
11:56:46 [DPL] battery discharge restricted, start 51.50 V or 50 %, stop 45.94 V or 20 %
11:56:46 [DPL] full solar-passthrough dormant, start 53.20 V or 97 %, stop 52.00 V
11:56:46 [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night enabled and dormant
11:56:46 [DPL] total max AC power is 800 W, conduction losses are 0 %
11:56:46 [DPL] targeting 10 W, base load is 200 W, power meter reads 277.2 W (valid)
11:56:46 [DPL] battery allowance is 0/0 W DC/AC, solar power is 511/485 W DC/AC, requested are 575 W AC
11:56:46 [DPL] requesting 485 W from 1 battery-powered inverter currently producing 308 W (diff 177 W, hysteresis 20 W)
11:56:46 [DPL] will cover 477 W using 1 battery-powered inverter
11:56:46 [DPL inverter 114184614039]:
11:56:46    battery-powered, producing 308 W, output included in power meter reading
11:56:46    lower/current/upper limit: 26/300/800 W, output capability: 800 W
11:56:46    sending commands enabled, reachable, eligible
11:56:46    max reduction production/standby: 282/308 W, max increase: 500 W
11:56:46    target limit/output/state: 477 W (update)/477 W/production, 0 update timeouts
11:56:46    MPPTs AC power: a: 154 W b: 154 W
11:56:46 [DPL inverter 114184614039]: sending limit of 59.6 % (477 W respectively), max output is 800 W
11:56:46 [DPL] waiting for a start/stop/restart/limit command to complete
```

Looks like it is overestimating solar power: `solar power is 511/485 W`

Maybe the phantom MPPTs are being counted? (https://github.com/hoylabs/OpenDTU-OnBattery/issues/1692)

ranma, 2025-03-25:
Shouldn't the phantom-device value be ignored once the data is stale (older than a minute or so)?

AndreasBoehm, 2025-03-25:
> Maybe the phantom MPPTs are being counted?

Thats exactly what happened here, yes.

> Shouldn't the phantom-device value be ignored once the data is stale (older than a minute or so)?

it should be ignored, yeah, but it looks like its not: https://github.com/hoylabs/OpenDTU-OnBattery/issues/1692#issuecomment-2751931689

closed 2025-03-25 by AndreasBoehm