hoylabs/OpenDTU-OnBattery #1780: Random restarts

j1gg3, 2025-03-26:
### What happened?

I check the uptime of the running OpenDTU-OnBattery pretty often and I recognized sometimes it is running nearly 2 days without restart but in most cases I do see an uptime of 5 - 6 hours. 
Honestly I do not know what could cause that issue. 

[image]

### To Reproduce Bug

Just let OpenDTU-OnBattery run...

### Expected Behavior

No restarts or at least scheduled restarts at the same time. 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

spcqike, 2025-03-26:
You may need to provide Serial logs 

https://opendtu-onbattery.net/firmware/howto/serial_console/

schlimmchen, 2025-03-26:
> You may need to provide Serial logs

You ~~may~~ **need** to provide Serial logs

gitisgreat2023, 2025-03-26:
Can you post the radio statistics of your inverters?

j1gg3, 2025-03-27:
[image]

[image]

[image]

Today my OpenDTU-OnBattery is running for 1 day and 14hours which is a new record so far.

gitisgreat2023, 2025-03-27:
I've observed the same thing: short uptimes (hours, max days).
I now have much better uptimes, after tuning the radio parameters and trying different antenna orientations. Parameters to change: transmission power (too high is bad, too low as well), in case of HMT also frequency. For all: the antenna typically have 6 orientations (x,y,z and each axis two directions), can make easily 10-20 dBm difference.
Example: for HMT: I changed the frequency from 868.25  (20% failure rate, 20 dBm, ) to 866.5MHz 0% failure, -30dBm, both  had transmitting power -10 dBm. Within 20 sec the HMT synchronized to the new frequency.
The radio statistics is perfect to optimize: change a setting, hit reset and look whether it got better. Flip it through the 6 orientations, play with transmission power.

j1gg3, 2025-03-28:
Indeed, I just changed the radio parameter and it's running without restart.

schlimmchen, 2025-03-29:
Please re-open the issue if the problem persists and share logs.

commit 9b7df71d (2024-03-20): webapp: Fix typo

Fixes #1780
closed 2025-03-29 by schlimmchen