hoylabs/OpenDTU-OnBattery #1784: [Bug] DPL: Inverter not shutdown when stop threshold is reached because of hysteresis

AndreasBoehm, 2025-03-27:
### What happened?

See https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1783

### To Reproduce Bug

Set the hysteresis higher than the lowest possible output your battery inverter will produce. This only happens if you don't use all inputs of the inverter.

### Expected Behavior

Battery powered inverters are shutdown when stop threshold is reached no matter what.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

6fb8618

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] up 43482 s, it is night, next inverter restart at 56927 s (set to 0)
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] battery interface enabled, SoC 37.0 % (ignored), age 3 s (valid)
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] BMS: 39.20 V, MPPT: 39.17 V, inverter 114181870469: 39.20 
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] battery voltage 39.20 V, load-corrected voltage 39.32 V @ 29 W, factor 0.00400 1/A
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] battery discharge restricted, start 42.50 V or 80 %, stop 39.00 V or 20 %
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] full solar-passthrough dormant, start 47.20 V or 100 %, stop 46.95 V
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] start NOT reached, stop NOT reached, solar-passthrough enabled, use at night disabled and dormant
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] total max AC power is 800 W, conduction losses are 5 %
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] targeting 0 W, base load is 100 W, power meter reads 211.0 W (valid)
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 240 W AC
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL] requesting 0 W from 1 battery-powered inverter currently producing 29 W (diff -29 W, hysteresis 30 W)
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - - [DPL inverter 114181870469]:
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     battery-powered, producing 29 W, output included in power meter reading
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     lower/current/upper limit: 50/63/700 W, output capability: 700 W
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     sending commands enabled, reachable, eligible
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     max reduction production/standby: 0/29 W, max increase: 637 W
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     target limit/output/state: -1 W (unchanged)/29 W/unchanged, 0 update timeouts
192.168.0.80	Mar 26 20:17:19	1	user	info		- wt32-eth01-OpenDTU OpenDTU a0a38a57 - -     MPPTs AC power: a: 29 W b: 0 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-03-27:
Leaving this for you @schlimmchen as I am not confident that my suggested change would help here.

schlimmchen, 2025-03-29:
I did not look at your suggestion, but the best that my brain came up with since thinking about this was the same, but a little different:

The problem occurs at the extremes.

When approaching the upper limit, we risk not setting the highest limit value permissible. However, we can simply ignore the upper range where this issue occurs. An even higher demand will eventually set the highest possible limit. If we would want to tackle this, we would first need to know if we are at that upper edge at all, which is non-trivial.

Very trivial to determine is whether or not we are at the other extreme, which is what you did. The only difference I think we should do is make the test "powerRequested < hysteresis". Does that make sense to you, @AndreasBoehm?

AndreasBoehm, 2025-03-29:
Your solution would ignore the hysteresis during normal operation, of course only for the extreme of already operating at a very low limit or output power. my idea would only ignore it when we actually need to shutdown the inverter.

PR #1818 (2025-04-07): Fix: DPL: ignore hysteresis when inverter should be shutdown
This pull request includes a change to the `PowerLimiterClass::updateInverterLimits` method in the `src/PowerLimiter.cpp` file. The change modifies the logic to handle cases where 0 W are requested, allowing battery-powered inverters to go into standby and prevent the battery from being fully discharged.

* [`src/PowerLimiter.cpp`](diffhunk://#diff-c502fa72d03eefad0a66ed1345a8f70eded4c8f0303ee7b38bb2a32094dbcc91L605-R608): Added a condition to ignore the hysteresis when 0 W are requested, enabling battery-powered inverters to go into [... cut]