hoylabs/OpenDTU-OnBattery #1793: [Bug] VE.Smart Network Total Power not used in DPL and not shown in live view

AndreasBoehm, 2025-03-29:
### What happened?

> dabei ist mir gerade aufgefallen, dass OpenDTU onBattery bei der Solarladeregler Gesamtleistung nur die Summe der MPPT-Leistungen anzeigt und per MQTT reported, nicht aber die eventuell höhere Leistung, die das Victron VE.Smart Network als Gesamtleistung meldet - kein Beinbruch, aber unschön wenn man nicht MPPT an OpenDTU anschließen kann. Eventuell ist das dann auch beim Full-Solar-Passthrough ein Problem?

_Originally posted by @hhoebel in https://github.com/hoylabs/OpenDTU-OnBattery/issues/1658#issuecomment-2757342415_

Right now we ignore the output of devices that are part of a VE.Smart Network but not directly connected to the DTU.

This applies to the DPL and to the live view.

### To Reproduce Bug

Add multiple Victron MPPT to a VE.Smart Network but only connect `count-1` MPPTs to the DTU.

### Expected Behavior

Output of all MPPTs is shown in the live view and used by the DPL.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.20

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-04-04:
If this is true, this is a regression. The VE.Direct network power should be prioritized.

AndreasBoehm, 2025-04-04:
Yes, its sad bit its true, its a regression that came with the solarcharger refactoring.

schlimmchen, 2025-04-04:
Oh, I see: This prioritization is only in getPanelPowerWatts(), but getOutputPowerWatts() is used to determine the solar output power. Well, I guess we need... I'll put it in a PR...

PR #1817 (2025-04-05): Fix: prioritize VE.Direct network power
the DPL shall use the VE.Direct network input power to derive (guesstimate) the available solar output. we did this before already, but a regression was introduced with the solar charger refactoring.

we now also try to be smarted about guesstimating the output: we reduce the network total input power by the input power of all charge controllers we are listening to and account their battery output power instead. the remainder of the network input power, i.e., the input power of all networked charge controllers that we are not listening to, is corrected f [... cut]
closed 2025-04-05 by schlimmchen