hoylabs/OpenDTU-OnBattery #1812: Bei mir klemmt es auch hms-2000 steigert Leistung nicht

firedp, 2025-04-04:
### What happened?

Hallo,
auch bei mir hakt es leider. Habe einen hms-2000 gekauft, der Firmware 2.0.4 drauf hat.
Freudig angeschlossen, bisher nur 2 Module.
Aber trotz Mindestleistung von 400W und Maximalleistung 1800W kommt er bei Bedarf nicht von den 400W weg, erst mit Wechselrichter Neustart gibt er ab was die 2 Module können. Am heutigen Sonnentag und der eben gestarteten Spülmaschine konnte ich es zweifelsfrei sehen.
Gibt es eine Lösung oder ist eine in Arbeit?
Downgrade Firmware derzeit nicht möglich, ich besitze keine Hoymiles DTU

### To Reproduce Bug

X

### Expected Behavior

X

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025 2 10

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
X
```

### Anything else?

X

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-04:
Hallo,

Ist dein Problem das der WR trotzt 100% Limit nicht die volle Leistung der zwei Module liefert oder hast du Dynamik Power Limiter aktiviert und das Limit wird nicht hochgesetzt?

Für ersteres ist die Firmware 2.0.4 verantworlich, hier würde ich ein downgrade empfehlen.

Für zweiteres kannst du in Issue #1675 und PR #1756 (hier gibts auch ne Test-Version) nachlesen und verfolgen wie das Problem behoben wird.

firedp, 2025-04-04:
Ach ja, ich habe keinen Speicher, bin Mitte 50, Laie und muss mir die englischen Texte übersetzen lassen.
Einerseits und vielleicht liegt es daran, das am provisorischen Aufbau nur 2 Module hängen, fällt im Grünen Feld das aktuelle Limit auf 1600W / 80%, auch wenn ich manuell es auf 100% / 2000W setze.
Dann habe ich schon die Mindestleistung auf 402W und die Höchstleistung auf 1800W begrenzt. Selbst die Nulleinspeisung habe ich auf -400W zwecks Einspeisung zugelassen. Aber egal was ich mache, sobald der Wechselrichter ins Minimum 402W fällt und ich den 2000W Wasserkocher einschalte, bleibt der hms-2000 unten hängen und springt nur durch Neustart des WR hoch, bis der Verbrauch wegfällt und er nicht Na untere Limit geht. Auch die Anzeige 1600W / 80% ist wie angenagelt. Also am Ende eine Nulleinspeisung bekomme ich nicht realisiert, das lief mit dem hm-600 noch richtig gut.

Daher vielleicht für den Laien in Deutsch. Soll ich eine originale DTU kaufen und welche Firmware soll Hoymiles aufspielen? Oder warten und hoffen, ihr findet ne Lösung für die Firmware 2.0.4? 
Ich bin echt dankbar für Eure Erfindung, es macht Spaß und bietet selbst ohne Speicher echten Mehrwert.
Nur jetzt weiß ich gerade nicht weiter.
Bei mir geht es schon länger nicht mehr ums Sparen, der Zug ist weg 😂 Aber das zu beherrschen was Ihr erfunden habt ist ein toller Zeitvertreib.

AndreasBoehm, 2025-04-05:
Bitte aktiviere in den DPL Einstellungen 'ausführliche Protokollierung'
[image]

Wenn du dann feststellst das der Wechselrichter das Limit nicht so ändert wie du es erwartet würdest öffnest du die Konsole und wartest etwas bis sie sich füllt. Anschließend den Inhalt der Konsole hier posten, dann können wir uns das anschauen.
[image]

Dann kann ich auch beurteilen ob wirklich die Firmware 2.0.4 schuld ist.
Da du bereits 400 W als Mindestleistung konfiguriert hast gehe ich nicht davon aus das du von Issue #1675 betroffen bist.
Das du nur 2 von 4 Eingängen nutzt sollte eigentlich kein Problem darstellen, das sehen ich dann aber in der Konsole.

Ich würde Firmware 1.1.12 empfehlen, die bietet Power Distribution Logic aber weist sonst keine bekannten Probleme auf. Vorerst muss dir aber keine DTU kaufen, lass uns erstmal versuchen das Problem so zu lösen.

firedp, 2025-04-05:
Ich muss ne Pause machen. Wechselrichter zurückgeschickt. Entgegen der Annonce war er gebraucht, nicht neu. Erst fehlte die Kaufrechnung, dann kam die falsche/ andere, wohl gedrosselt mit Hoymiles DTU auf 80%, in openDTU oB sprang das Limit manuell auf 100% gesetzt zurück auf 80% und und. Gut das der Hoymiles Händler mir Auskünfte gegen den Verkäufer gegeben hat.
Neuen WR bestelle ich gleich und dann geht es weiter.

AndreasBoehm, 2025-04-07:
> in openDTU oB sprang das Limit manuell auf 100% gesetzt zurück auf 80% 

Hattest du den DPL deaktiviert als du das Limit manuell gesetzt hast?
Falls nicht, dann hat der DPL das Limit einfach überschrieben.

firedp, 2025-04-07:
Sicher bin ich nicht. Glaube DPL war noch aktiviert.
Diese Woche kommt ein neuer hms-2000, dann mal gucken welche Firmware drauf ist, ich vermute eine alte aus 2024. Dann muss ich wohl eine wlite S kaufen und erstmals Kontakt zu Hoymiles suchen für das 1.1.12

AndreasBoehm, 2025-04-10:
Ich würde hier erstmal zu machen, falls es Probleme mit deinem neuen Wechselrichter gibt kannst du gerne einen neuen Issue öffnen :)

closed 2025-04-10 by AndreasBoehm