hoylabs/OpenDTU-OnBattery #1829: Bei zwei Inverten im DPL wird der zweite mit abgeschaltet wenn der erste offline geht

Saarblick, 2025-04-07:
### What happened?

Mein Akku betriebener HM600 wird abgeschaltet wenn der HMT2250 offline geht.
Um 20.15 wird der HMT2250 gelb
um 20.21 wird der HM600 rot
um 20.22 wird der HMT2250 wieder grün
um 20.22 wird der HM600 wieder grün
um 20.23 wird der HMT2250 gelb
um 20.24 wird der HM600 wieder grün
um 20.28 wird der HM600 kurz grün
um 20.29 wird der HM600 kurz grün
um 20.29 bleibt der HM600 grün nachdem der HMT2250 rot ist.

### To Reproduce Bug

War gestern genau so, konnte nur kein log erstellen

[LOG DPL.txt.txt](https://github.com/user-attachments/files/19637737/LOG.DPL.txt.txt)

### Expected Behavior

Update for the next release

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

727c357

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
send by PM File was to big
```

### Anything else?

[LOG DPL.txt.txt](https://github.com/user-attachments/files/19637742/LOG.DPL.txt.txt)

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-07:
Kannst du bitte "ausführliche Protokollierung" unter DTU deaktivieren und im DPL aktivieren, anschließend einen neustart der DTU durchführen?

Leider fehlt im Log das ausführliche Protokoll des DPL, deshalb sehe ich keinen Grund warum der WR ständig aus und an geht.

Bitte ein neues Log anfertigen nach dem du die oben erwähnten Schritte durchgeführt hast.

Saarblick, 2025-04-08:
Hallo,

Einmal neu:

20.16 HMT Gelb
20.16 HM gelb
20.22 HMT kurz grün
20.23. HM Rot
20.24 HM grün
20.24 HMT grün/Gelb
20.24 HM gelb
20.25 HMT grün/gelb
20.25 HM grün/gelb
20.26 HM grün/Gelb
20.27 HMT grün/Gelb
20.28 HM grün/Gelb
20.29 HMT grün/Gelb
20.30 HM grün/ Gelb
20.31 HMT Rot
20.32 HM Grün
Danach alles gut..

[LOG DPL2.txt.txt](https://github.com/user-attachments/files/19655124/LOG.DPL2.txt.txt)

AndreasBoehm, 2025-04-08:
Die DTU ist der überzeugung das es noch Tag ist und denkt deshalb das der HMT noch Strom liefern kann.

Hast du unter Einstellungen -> NTP deinen Standort korrekt konfiguriert? Und welchen Dämmerungstyp hast du dort eingestellt?

Kannst du auch noch einen Screenshot von Info -> NTP machen?

Beispiel

[image]

AndreasBoehm, 2025-04-08:
Das ist der default standort.
Bitte passe die Koordinaten auf deinen Standort an.

> Screenshot von Info -> NTP

Nicht Settings -> NTP ;)

Saarblick, 2025-04-08:
Ok, für Tag ist es derzeit aber schon recht dunkel
[image]

AndreasBoehm, 2025-04-08:
Ja, das ist nicht gut.

bitte deine Koordinaten eintragen und zusätzlich den Dämmerungstyp auf „Standard Dämmerung“ ändern.

Saarblick, 2025-04-08:
Danke,
Nach standard Dämmerung wurde es Nacht 🤪

AndreasBoehm, 2025-04-08:
Bin gespannt ob es damit besser funktioniert.

AndreasBoehm, 2025-04-09:
> Im entsprechenden WR-Menü habe ich die Kommunikation für nachts für diese beiden Inverter deaktiviert.

Das ist richtig und wichtig.

> Jetzt bekomme ich zwar im DPL-Menü den Warnhinweis, dass es erforderlich sei, dass die Kommunikation für alle WR eingeschaltet sei, aber mir erschließt sich nicht der Sinn, weshalb nachts Anfragen empfangen/gesendet werden sollen, wenn die WR ohnehin nicht erreichbar sind.

Das ist ein Bug in der Anzeige, nehm ich als Issue auf.

> Und dann noch eine Verständnisfrage zu Solar-Passthrough

Bitte leg eine neue Diskussion oder einen neuen Issue an, das passt nicht zum aktuellen Issue.

AndreasBoehm, 2025-04-09:
> Im entsprechenden WR-Menü habe ich die Kommunikation für nachts für diese beiden Inverter deaktiviert.

@Saarblick hast du das für deinen HMT auch so konfiguriert?

AndreasBoehm, 2025-04-09:
@RealNBB welche Version nutzt du denn aktuell? Eine Test-Version von `development` oder `2025.02.10` ?

RealNBB, 2025-04-09:
> [@RealNBB](https://github.com/RealNBB) welche Version nutzt du denn aktuell? Eine Test-Version von `development` oder `2025.02.10` ?

Diese:

https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14302247009

Saarblick, 2025-04-09:
> > Im entsprechenden WR-Menü habe ich die Kommunikation für nachts für diese beiden Inverter deaktiviert.
> 
> @Saarblick hast du das für deinen HMT auch so konfiguriert?

Gerade nachgeschaut, nein habe bei beiden alles an

Soll ich das anpassen oder erst mal abwarten wie es heute klappt?

[image]

AndreasBoehm, 2025-04-09:
Mach das für den Solar betrieben HMT auf alle Fälle aus. Sonst versucht die DTU dauernd mit dem zu kommunizieren obwohl er nie antworten wird in der Nacht.

vaterlangen, 2025-04-13:
Bei mir ergibt sich ein ähnliches Problem - allerdings Morgens, wenn der PV WR erreichbar wird. Sobald er erreichbar wird (aber noch keine Energie liefern kann) wird der SmartBufferBattery (SBB) WR auf Standby gesetzt obwohl der andere WR noch nicht liefern kann. Das ganze geht dann ein paarMinuten hin und her (PV WR geht wieder aus => SBB WR übernimmt, PV WR geht wieder an => SBB WR geht auf Standby) bis der PV WR dann liefern kann. Ab dann läuft alles wieder stabil und wie gewohnt. 

Setup:
1x HMT2250 mit 2x Zendure HUB als `SmartBufferBatterie` WR
1x HMT2000 mit drei PV Modulen (>400W) als `Solarmodule` WR

Koordinaten sind korrekt eingestellt und Dämmerung ist auf 90,8°. Allerdings liegt meine PV Anlage hinter dem Haus und wird morgens Vollständig verschattet - evtl stimmt deshalb die Berechung für Tag/Nacht in meinem Fall nicht ganz. Leider konnte ich Logs und Verlauf nicht Zeitgleich aufehmen, daher von zwei Tagen - das Verhalten habe ich aber schon öfter beobachtet.

```
Websocket: [/console][1] connect
06:52:35.334 > Websocket: [/livedata][1] disconnect
06:52:35.843 > Websocket: [/solarchargerlivedata][1] disconnect
06:52:35.843 > Websocket: [/batterylivedata][1] disconnect
06:52:35.843 > [DPL] up 124 s, it is day, next inverter restart at 61800 s (set to 0)
06:52:36.868 > [DPL] targeting -10 W, base load is 275 W, power meter reads 202.2 W (valid)
06:52:36.868 > [DPL] requesting 212 W from 1 solar-powered inverter currently producing 0 W (diff 212 W, hysteresis 10 W)
06:52:36.868 > [DPL] will cover 212 W using 1 solar-powered inverter
06:52:36.868 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 0 W (diff 0 W, hysteresis 10 W)
06:52:36.868 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
06:52:36.868 > [DPL inverter SBBWR]:
06:52:36.868 >     smart-buffer-powered, standing by at 0 W, output included in power meter reading
06:52:36.883 >     lower/current/upper limit: 40/38/2250 W, output capability: 2250 W
06:52:36.883 >     sending commands enabled, reachable, eligible
06:52:36.883 >     max reduction production/standby: 0/0 W, max increase: 2250 W
06:52:36.883 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
06:52:36.883 >     MPPTs AC power: a: 0 W b: 0 W c: 0 W
06:52:36.883 > [DPL inverter PVWR]:
06:52:36.883 >     solar-powered, standing by at 0 W, output included in power meter reading
06:52:36.883 >     lower/current/upper limit: 25/216/2000 W, output capability: 2000 W
06:52:36.883 >     sending commands enabled, reachable, eligible
06:52:36.900 >     max reduction production/standby: 0/0 W, max increase: 2000 W
06:52:36.900 >     target limit/output/state: 212 W (update)/212 W/production, 3 update timeouts
06:52:36.900 >     MPPTs AC power: a: 0 W b: 0 W
06:52:36.900 > [DPL inverter PVWR]: sending limit of 10.6 % (212 W respectively), max output is 2000 W
06:52:36.900 > [DPL] waiting for a start/stop/restart/limit command to complete
06:52:36.900 > Fetch inverter: PVWR
06:52:36.900 > Queue size - NRF: 0 CMT: 2
06:52:36.900 > ZendureBattery: Now: 1744174358, LastFull: 1070513516, Diff: 1743865039
06:52:37.243 > Admin AP remaining seconds: 120 / 180
06:52:37.892 > Success
06:52:38.314 > [DPL inverter PVWR]: limit update succeeded, actual limit is 10.6 % (212 W respectively), effective 2059 ms after update started, requested were 10.6 %
06:52:38.314 > [DPL inverter PVWR]: Starting inverter...
06:52:38.314 > Fetch inverter: SBBWR
06:52:38.314 > Queue size - NRF: 0 CMT: 3
06:52:38.314 > Success
06:52:39.717 > Fetch inverter: PVWR
06:52:40.451 > Queue size - NRF: 0 CMT: 3
06:52:40.451 > Success
06:52:40.961 > Success
06:52:41.989 > Success
06:52:43.693 > [DPL inverter PVWR]: Starting inverter...
06:52:43.694 > Fetch inverter: SBBWR
06:52:43.694 > Queue size - NRF: 0 CMT: 2
06:52:43.694 > Success
06:52:44.541 > Fetch inverter: PVWR
06:52:44.541 > Queue size - NRF: 0 CMT: 2
06:52:44.541 > Success
06:52:44.550 > ZendureBattery: Logging Frame received!
06:52:45.060 > Middle missing
06:52:45.090 > Request retransmit: 3
06:52:45.090 > Middle missing
06:52:45.090 > Request retransmit: 4
06:52:45.090 > Middle missing
06:52:46.084 > Request retransmit: 5
06:52:46.084 > Success
06:52:46.084 > [DPL inverter PVWR]: Starting inverter...
06:52:46.084 > ZendureBattery: Logging Frame received!
06:52:46.084 > [DPL] waiting for a start/stop/restart/limit command to complete
06:52:47.618 > Fetch inverter: SBBWR
06:52:47.619 > Queue size - NRF: 0 CMT: 2
[... 89 lines cut ...]
06:53:12.817 > [DPL] waiting for a start/stop/restart/limit command to complete
06:53:12.817 > Fetch inverter: PVWR
06:53:13.729 > Queue size - NRF: 0 CMT: 2
06:53:13.729 > Success
06:53:13.762 > [DPL inverter PVWR]: limit update succeeded, actual limit is 10.7 % (214 W respectively), effective 2059 ms after update started, requested were 10.7 %
06:53:13.762 > [DPL inverter PVWR]: Starting inverter...
06:53:13.762 > Fetch inverter: SBBWR
06:53:14.115 > Queue size - NRF: 0 CMT: 3
06:53:14.115 > Success
06:53:15.779 > Fetch inverter: PVWR
06:53:16.188 > Queue size - NRF: 0 CMT: 3
06:53:16.188 > Success
06:53:16.802 > Success
06:53:17.315 > Success
06:53:17.827 > [DPL inverter PVWR]: Starting inverter...
06:53:17.827 > Fetch inverter: SBBWR
06:53:18.340 > Queue size - NRF: 0 CMT: 2
06:53:18.340 > Admin AP remaining seconds: 160 / 180
```

[image]

[image]

@AndreasBoehm Kann auch gerne ein eigenes Issue aufmachen - aber ich denke die Ursache ist in beiden Fällen gleich

AndreasBoehm, 2025-04-14:
Danke für den ausführlichen report @vaterlangen !
Problem hab ich denke ich soweit verstanden, hab auch schon eine Idee wo das herkommt. Fix kommt die Tage.

AndreasBoehm, 2025-04-15:
Ich hab eine neue Test-Version erstellt die das Problem lösen sollte.
Bitte Testen :)

https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14474667821

PR #1865

vaterlangen, 2025-04-15:
> Ich hab eine neue Test-Version erstellt die das Problem lösen sollte.
Bitte Testen :)

Sieht erstmal gut aus - der SBB WR ist heute bei Sonnenuntergang aktiv geblieben, obwohl der PV WR nicht mehr liefern könnte. In den ~4 Minuten war der Regelfehler etwas höher (ziemlich genau das minimum limit des PV WR - wie lt. Code zu erwarten) als sonst. Das auch zu korrigieren dürfte aber noch deutlich aufwendiger sein - irgendeine Art der Berechnung/Erkennung der "verschattungssituation" aus geforderten AC-Limit und der DC-Power der MPPTs...

Bei Sonnenuntergang ist es bei mir schonmal eine deutliche Verbesserung - Danke @AndreasBoehm !

Wie es bei Sonnenaufgang aussieht liefere ich die Tage nach.

AndreasBoehm, 2025-04-16:
Danke @vaterlangen.

> Das auch zu korrigieren dürfte aber noch deutlich aufwendiger sein - irgendeine Art der Berechnung/Erkennung der "verschattungssituation" aus geforderten AC-Limit und der DC-Power der MPPTs...

Das ist bereits implementiert. Das Problem ist das der Solar WR zwischen erreichbar und nicht-erreichbar hin und her wechselt. Wenn er kürzlich erreichbar wurde wissen wir nicht wie viel Leistung er liefern kann und gehen davon aus das der WR das minimal limit liefern kann. Der verschattungs check kommt dann im nächsten Regelzyklus und ignoriert den Solar WR.

Jetzt ist es wichtig das minimal limit bei allen WRs auf den niedrigsten Wert zu ändern der vernünftig läuft, also ~13 W pro MPPT, um den Fehler so gering wie möglich zu halten. Zusätzlich ist eben auch der Standort und Dämmerung so einzustellen das wir Solar WR auch zum richtigen moment aus dem DPL rausfliegen lassen.

@Saarblick wie siehst bei dir aus? Funktioniert das bei dir jetzt auch alles soweit?

Saarblick, 2025-04-16:
Bei mir läuft alles, bin aber unterwegs konnte das letzte update nicht aufspielen.

vaterlangen, 2025-04-16:
> Das ist bereits implementiert. Das Problem ist das der Solar WR zwischen erreichbar und nicht-erreichbar hin und her wechselt. Wenn er kürzlich erreichbar wurde wissen wir nicht wie viel Leistung er [...]

> Jetzt ist es wichtig das minimal limit bei allen WRs auf den niedrigsten Wert zu ändern der vernünftig läuft, also ~13 W pro MPPT, um den Fehler so gering wie möglich zu halten. Zusätzlich ist eben [...]

Ok, super! Beobachte ich nochmal eine Weile. Für mich ist das aktuell absolut ok - mein PV WR läuft mit 25W minimum Limit stabil. Das Thema mit dem Regelfehler interessiert mich eher akademisch 😉

Danke für die schnelle Lösung!