hoylabs/OpenDTU-OnBattery #1841: [Bug] DPL Settings shows warning for solar inverter when commands at night are disabled

AndreasBoehm, 2025-04-09:
### What happened?

When polling and sending commands for solar inverters is disabled the DPL settings show a warning, but it shouldn't.

[image]

### To Reproduce Bug

- Disable sending commands and polling for a inverter
- Open DPL settings

### Expected Behavior

Warning is only shown for non-solar inverters

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

RealNBB, 2025-04-16:
Danke...passt nun! :)

AndreasBoehm, 2025-04-17:
Danke fürs testen @RealNBB 

Ich hab mich mittlerweile von @schlimmchen belehren lassen und weiß jetzt das es nicht schlimm ist wenn Nachts senden/empfangen aktiv ist da wir einen Solar WR im DPL ignorieren sobald es 'Nacht' ist.

RealNBB, 2025-04-17:
Ahh okay! :)

Na ich lass es aber trotzdem deaktiviert bei mir! :D

PR #1868 (2025-04-16): webapp: show relevant requirements for solar powered inverters
Closes #1841 

This pull request includes updates to the localization files and improvements to the `PowerLimiterAdminView` component to handle new communication requirements for inverters. The most important changes include splitting the inverter communication hint into day and night versions and updating the logic to check for these conditions.

Localization updates:

* [`webapp/src/locales/de.json`](diffhunk://#diff-a6b5fa0b8dd386d0bb5157c44cfb0c9c6ad7422a3a3670e7e956c40fe5529e2fL688-R689): Added separate hints for day and night [... cut]