hoylabs/OpenDTU-OnBattery #1849: Energy Readout Stocks with Victron VM-3P

Saarblick, 2025-04-10:
### What happened?

With Victron SmartMeter VM-3P75CT the readout stocks compleatly when I connect paralel via Victron App the victron Smartmeter to compare the energy consumption

[LOG Victron.txt.txt](https://github.com/user-attachments/files/19689843/LOG.Victron.txt.txt)

### To Reproduce Bug

Log into the DTU check the values open the victron app and check here some values in paralel, when you go back to DTU the value of  energy consumtion are frozen.

Fix are only with DTU Restart possible

### Expected Behavior

Fix when possible

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

727c357 V4.4.7-ditry

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

[image]

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-11:
@Sorbat could you assist with your Wireshark skills to collect some logs about whats going on here?
And can you try to reproduce this issue? Thanks!

AndreasBoehm, 2025-04-11:
You got me wrong @Saarblick, i tagged Sorbat in that message. He is the one that initially requested the Victron VM integration and has it running without issues as far as i know.

Sorbat, 2025-04-11:
Das Problem von @Saarblick war bei mir am gestrigen Abend erstmalig aufgetreten. Leider blieb es das einzige Mal, trotz mehrfachen öffnen der App heute konnte ich es nicht nachstellen. 

Mit Wireshark konnte ich keine Kommunikation zwischen der DTU und dem VM sehen (obwohl der Wert aktualisiert wurde und demnach ja eine Kommunikation stattfand). 
Anfragen, die ich selbst mit Modbusmaster sendete, konnte ich sehen. Keine Ahnung 🤷🏻‍♂️

Saarblick nutzt Victronseitig den Cerbo, welcher die Daten ebenfalls über Modbus UDP abruft. Mein Cerbo ist noch nicht angeschlossen. Evtl könnte es ja auch damit zusammenhängen, wobei ich dann nicht wüsste, woher das Problem bei mir gestern Abend kommen sollte 🤷🏻‍♂️

schlimmchen, 2025-04-20:
```
20:02:38.953 > [PowerMeters::Udp::Victron] received 227 bytes:
20:02:38.953 > 00 71 00 00 00 0D 01 17 0A 22 57 00 00 00 04 D5 
20:02:38.953 > 01 00 00 00 72 00 00 00 0D 01 17 0A 22 5C 00 00 
20:02:38.953 > 00 04 73 00 00 00 DE AD 00 00 00 B7 01 03 B4 13 
20:02:38.953 > 87 00 03 00 00 0F 8F 00 00 01 92 00 00 00 23 00 
20:02:38.953 > 00 00 00 00 00 00 00 00 00 00 00 59 EC FF 41 00 
20:02:39.000 > 00 04 D9 00 00 05 91 9A 91 00 00 58 F8 00 DD 00 
20:02:39.000 > 00 0A 73 00 00 01 4A 9B E8 00 00 5A 24 00 69 00 
20:02:39.000 > 00 06 6F 00 00 00 E1 9B 95 00 00 00 00 00 00 00 
20:02:39.000 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20:02:39.000 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20:02:39.000 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20:02:39.000 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20:02:39.000 > 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E5 FF 
20:02:39.000 > FF FE 9E 00 00 01 B9 00 00 01 D5 00 00 01 F8 00 
20:02:39.000 > 00 00 73 
20:02:39.000 > [PowerMeters::Udp::Victron] invalid transaction ID: 0071
```

Joar, das sieht aber nach einem Fehler am Victron aus?! Warum schickt der uns was anderes, als wir erwaten? Die Antwort auf unsere Anfrage beginnt erst bei "DE AD 00 00 ...". Diese UDP Pakete sind unidirektional, also warum steht da mehr drin, als wir uns bestellt haben?

Interessant (ChatGPT o4-mini-high):

There are **3** Modbus‑UDP messages in that dump:

* Message 1 @ offset 0: TID=0x0071, Protocol=0, Length=0x000D (13 bytes → 19 bytes total)
* Message 2 @ offset 19: TID=0x0072, Protocol=0, Length=0x000D (13 bytes → 19 bytes total)
* Message 3 @ offset 38: TID=0xDEAD, Protocol=0, Length=0x00B7 (183 bytes → 189 bytes total)

Nagut, fair enough... Warum auch immer wir vom Victron, nur weil die App offen ist, Zeug erhalten, das wir nicht angefragt haben: Gültige Modbus-Daten müssen wir trotzdem verarbeiten können und dann eben wegwerfen, wenn wir nur an der Antwort zu unserer Anfrage interessiert sind.

Warum dann aber der Victron still bleibt... Keine Ahnung. Wir schicken munter jede `PollingIntervalMs` einen neuen Request. Jedenfalls sehe ich im Code nichts, dass das verhindern würde. Wäre aber ein Vebose Log wert...

Bitte nehmt mal die Firmware aus #1882. Da werden unbekannte modbus antworten übersprungen -- hoffentlich korrekt. Sucht nach einer Zeile "skipping message with unexpected transaction ID". Und falls dann wieder Stille ist, würde ich gerne "sent modbus request" sehen.

Sorbat, 2025-04-21:
Wir hatten damals bei Modbus Abfrage UnitID 1 verwendet. Die UnitID 1 stammte, wenn ich es richtig in Erinnerung habe, aus der modbus_udp_client.py aus #1736. Wir waren uns damals unsicher, ob das richtig ist, ich hatte ein paar andere IDs getestet, bekam da aber Timeouts, mit UnitID 1 funktionierte die Abfrage, weswegen wir diese übernommen hatten. 
Vielleicht ist die 1 aber nicht richtig und es kommt deshalb zu Problemen, wenn der Cerbo parallel abfragt?

Saarblick, 2025-04-22:
Habe die 1882 installiert, bleibt auch hängen.
Soll ich ein Log erstellen oder auch die 1869 testen?

AndreasBoehm, 2025-04-22:
Bitte erstell ein Log.

1869 ist ein alter Hut, bitte nicht installieren.

schlimmchen, 2025-04-22:
Log wäre gut, denn da steht vielleicht "packet too large" und es steht dort hoffentlich "sent modbus request" auch während es klemmt. Du kannst aber auch gerne pauschal die neue Firmware in #1882 installieren (der [entsprechende Kommentar](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1882#issuecomment-2817317582) wird gleich aktualisiert).

Saarblick, 2025-04-23:
Anscheinend ist mein Kommentar untergegangen.
Problem ist gefixt, habe den ganzen Tag die Victron app auf und zu gemacht.
Ich hatte trotzdem ein Log erstellt, bitte melden wenn es benötigt wird.

PR #1882 (2025-04-27): Fix: skip unexpected Modbus UDP answers from Victron power meter
Closes #1849.
closed 2025-04-27 by schlimmchen