hoylabs/OpenDTU-OnBattery #1852: Neustart per Curl nicht möglich

mathias172, 2025-04-11:
### What happened?

Ich muss aus Gründen meine DTU nachts neu starten. Ich versuche, einen API-Call laut https://opendtu-onbattery.net/firmware/web_api/#list-of-urls

wie folgt:
curl -X 'POST' \
  'http://192.168.1.1/api/maintenance/reboot' \
  -H 'accept: */*' \
  -H 'Authorization: Basic YWRtaW46cGFzc3dvcmQ=' \
  -d ''

Und bekomme als Fehler: 
{"type":"warning","message":"No values found!","code":1002}
Was bitte mache ich falsch? Ist doch laut Anleitung mit -d '' - als null Daten für den Neustart.

PS: Die Crendentials sind korrekt.

### To Reproduce Bug

einfach mal in Linux ne Konsole mit den Daten füttern

### Expected Behavior

Neustart

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

latest

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Anything else?

-

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-11:
Die Dokumentation stimmt leider nicht.

Payload muss `reboot: true` enthalten

mathias172, 2025-04-11:
Hi Andreas, danke für die Antwort. Ich bin selber Programmierer aber ich raffs nicht. Ich habe -d 'reboot: true' und -d 'reboot=true' versucht. Nichts.

Wie muss der Call definiert sein?

AndreasBoehm, 2025-04-11:
Wir erwarten eine json payload.

Deine zwei varianten sind kein valides json

AndreasBoehm, 2025-04-11:
Hats jetzt geklappt?
Und warum musst du überhaupt neustarten? 😂

mathias172, 2025-04-12:
Hi Andreas, klappt.
Also falls mal ein Neueinsteiger sich denkt "na wie geht dat denn nu?`"
Kann man so machen: curl -X 'POST' 'http://admin:passwort@192.168.1.1/api/maintenance/reboot'   -H 'accept: */*'   -d 'data={"reboot":true}

Ich muss erstmal als workaround opendtu Nachts neu starten, da es keine Nulleinspeisung mit meinem Speicher schafft. Da wird alles auf 100% geregelt. Ich schreibe gleich das Ticket dazu... Wird die DTU neu gestartet, klappt es wieder mit der Nulleinspeisung.

closed 2025-04-12 by mathias172