hoylabs/OpenDTU-OnBattery #1854: DPL: keine Nulleinspeisung nachts mit Speicher

mathias172, 2025-04-12:
### What happened?

Ich habe 3 BKW angeschlossen via Hoymiles HM 600 und an einem davon einen Growatt NOAH 2000 Speicher. Der Speicher wird um 0:00 Uhr aktiv (auf 600W fix eingestellt), die DTU soll Nulleinspeisung erreichen. Tagsüber ist das kein Problem jedoch ist mir die letzten Nächte aufgefallen, dass die DTU auf 100% Leistung am Wechselrichter "hängt" obwohl z.B. nur 120W benötigt werden. Die DTU zeigt den Wert vom Stromzähler korrekt an, kennt also den Verbrauch im Logfile (habe leider noch keines gespeichert) ist zu sehen, dass selbst über 15 Minuten nicht einmal versucht wird, etwas zu regeln.

Starte ich die DTU neu funktioniert alles, wie gewünscht.

### To Reproduce Bug

Tritt jede Nacht auf, Setup siehe oben.

### Expected Behavior

Nulleinspeisung

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

latest stable

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-12:
Das kommt mir sehr bekannt vor und klingt nach einem Problem das wir denke ich bereits gelöst hatten.

Versuchs mal mit der aktuellsten Test-Version https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259

mathias172, 2025-04-12:
Schon wieder danke für die schnelle Antwort ;) Ich werde die nächsten Nächte mal ein Auge drauf werfen. Danke allen beteiligten für dieses großartige Projekt! So geht aktiver Klimaschutz -> den Leuten was "zum Anfassen" reichen und nicht doof auf der Autobahn ankleben ;)

andreash-esp, 2025-04-16:
Hi, ich meine ich habe dasselbe Verhalten seit dem Umstieg auf die letzte Stable. Mein Setup ist ein Hoymiles HM-1200 mit je einem Bluepalm Speicher pro MPPT, also zwei Stück. Der DPL ist auf einen angestrebten Netzbezug von 20W eingestellt. Das lief recht stabil, ich wollte aber die SmartBattery Unterstützung testen.

Seither habe ich zwei Phänomene, die ich erstmal auf schlechten WLAN-Empfang zurückgeführt hatte. Daran liegt es aber wohl nicht.

1. Zu wenig Leistung vom Wechselrichter, wie oben beschrieben. Der WR hängt dann gefühlt an einem Limit fest, obwohl mehr Leistung abrufbar wäre. Im Log taucht dann auf: "target limit/output/state: -1 W (unchanged)/648 W/unchanged, 0 update timeouts"
Dazu habe ich das die von AndreasBoehm erwähnte Test-Version aufgespielt, das hat zumindest gestern Abend geholfen.

2. Denn der DPL eine geringe Einspeisung erreicht, regelt er nicht zurück sondern bleibt bei -15W bis -30W hängen. Dazu habe ich das Log gestern gesichert:
1:48:13.901 > [DPL] targeting 20 W, base load is 100 W, power meter reads -24.0 W (valid)
21:48:13.971 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
21:48:13.971 > [DPL inverter 4711]:
21:48:13.971 >     solar-powered, producing 648 W, output included in power meter reading
21:48:13.971 >     lower/current/upper limit: 30/640/1200 W, output capability: 1200 W
21:48:13.971 >     sending commands enabled, reachable, disqualified (nighttime)
21:48:13.971 >     max reduction production/standby: 0/0 W, max increase: 0 W
21:48:13.971 >     target limit/output/state: -1 W (unchanged)/648 W/unchanged, 0 update timeouts
21:48:13.971 >     MPPTs AC power/DC voltage: a: 324 W/45.8 V b: 324 W/45.7 V
21:48:15.011 > [DPL] targeting 20 W, base load is 100 W, power meter reads -24.0 W (valid)
21:48:15.011 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
21:48:15.011 > [DPL inverter 4711]:
21:48:15.011 >     solar-powered, producing 648 W, output included in power meter reading
21:48:15.011 >     lower/current/upper limit: 30/640/1200 W, output capability: 1200 W
21:48:15.452 >     sending commands enabled, reachable, disqualified (nighttime)
21:48:15.453 >     max reduction production/standby: 0/0 W, max increase: 0 W
21:48:15.453 >     target limit/output/state: -1 W (unchanged)/648 W/unchanged, 0 update timeouts
21:48:15.453 >     MPPTs AC power/DC voltage: a: 324 W/45.8 V b: 324 W/45.7 V
Sieht so aus, als würde er von einer Richtung aus zum angestrebten Bezug regeln, von der anderen Richtung aus nicht. Evtl. ein fehlendes "abs(Abweichung)" irgendwo?
Möglicherweise tritt das auch verstärkt auf, wenn ein Akku leer wird. Zu dem Zeitpunkt hatte ich heute nacht eine Einspeisung von 0,3kWh in einer Stunde: 
[image]
Dazu leider kein Log, da war ich nicht wach.

AndreasBoehm, 2025-04-16:
@andreash-esp 
> Das kommt mir sehr bekannt vor und klingt nach einem Problem das wir denke ich bereits gelöst hatten.
> 
> Versuchs mal mit der aktuellsten Test-Version https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259

andreash-esp, 2025-04-16:
> [@andreash-esp](https://github.com/andreash-esp)
> 
> > Das kommt mir sehr bekannt vor und klingt nach einem Problem das wir denke ich bereits gelöst hatten.
> > Versuchs mal mit der aktuellsten Test-Version https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259

Die Version habe ich gestern Abend aufgespielt, seither scheint die Regelung von oben zu klappen (Phänomen 1), wobei ich bei angstrebtem Bezug 20W einen Bezug von 100-200kWh pro Stunde nicht erklären kann bei stabilem Nachtverbraucht und ausreichend WR-Leistung und Batteriekapazität.

Phanömen 2, das nicht-Ausregeln von unten, hatte ich auch bei der Testversion beobachtet.

Sorry, das ist in meinem Eintrag nicht ganz rübergekommen.

AndreasBoehm, 2025-04-16:
> 21:48:13.971 > [DPL inverter 4711]:
> 21:48:13.971 > solar-powered, producing 648 W, output included in power meter reading
> 21:48:13.971 > lower/current/upper limit: 30/640/1200 W, output capability: 1200 W
> 21:48:13.971 > sending commands enabled, reachable, disqualified (nighttime)

Dein WR ist als **solar-powered** konfiguriert und wird deswegen nachts nicht geregelt **disqualified (nighttime)**

AndreasBoehm, 2025-04-16:
> Sorry, das ist in meinem Eintrag nicht ganz rübergekommen.

Bitte versuch in Zukunft deinen Text etwas zu formatieren, die wall of text in deinem ersten Kommentar ist schwer zu lesen, deshalb hab ich auch gar nicht gesehen das du die Test-Version schon probiert hast.

andreash-esp, 2025-04-16:
> Dein WR ist als **solar-powered** konfiguriert und wird deswegen nachts nicht geregelt **disqualified (nighttime)**

Ok, danke! Das hat sich praktisch mit der Einführung der SmartBattery geändert? Ich hatte das Feature testweise deaktiviert, um zu sehen ob das Problem daran liegt.

AndreasBoehm, 2025-04-16:
> Das hat sich praktisch mit der Einführung der SmartBattery geändert?

Das kam erst im späteren Verlauf, da uns Solar WR die spät Abends noch erreichbar sind sonst Probleme machen.

mathias172, 2025-04-16:
Auch mal Feedback von mir dazu: Hier läufts jetzt fehlerfrei. Super Arbeit!

Github-Nutzer, 2025-04-17:
Blöde Frage dazu - ich glaube irgendwo gelesen zu haben dass man bei Verwendung von mehreren WR solarmodul- und smartbuffer - betrieben nicht mischen soll? Ist das korrekt?

AndreasBoehm, 2025-04-17:
Ich bin mir sicher ob ich deine Frage richtig verstanden habe, deswegen beantworte ich das einfach mal beide Varianten.

### Solar-, Smart-Buffer- und Batteriebetrieben WR gemeinsam mit DPL betreiben (mischen)
Das ist kein Problem und auch so vorgesehen das man diese arten von WR gemeinsam mit dem DPL betreiben kann

### Solarmodule und Smart-Buffer an einem einzelnen WR anschließen
Das verwenden von unterschiedlichen Stromquellen an einem WR wird nicht unterstützt, da es für den DPL keine möglichkeit gibt hier ordentlich zu steuern. Versuchen kann man es, aber wir werden hierbei auch nicht supporten.

Github-Nutzer, 2025-04-17:
Sorry, unpräzise ausgedrückt ... Frage mit Variante 1 beantwortet - danke!

schlimmchen, 2025-04-18:
> Auch mal Feedback von mir dazu: Hier läufts jetzt fehlerfrei. Super Arbeit!

@mathias172 Sind wir hier fertig, ist dein Problem gelöst?

mathias172, 2025-04-19:
Ich habe mir jetzt mal jede Nacht die SQL's aus meiner DB angeschaut und jetzt mal verglichen. Sieht top aus, das Problem ist gelöst und eben gerade nochmal zwei Platten angebaut ;) Tausend Dank! (Ticket geschlossen)

closed 2025-04-19 by mathias172