hoylabs/OpenDTU-OnBattery #1862: Dtu hängt sich auf, wenn Inverter in Standby geht

Feuerossi, 2025-04-15:
### What happened?

Wenn mein Speicher, Marstek 2500 B, leer ist und nur in Ladung geht, also nichts durchschleift, hängt sich die Dtu auf. Der Netzbezug bleibt auf dem letzten gemessenen Wert stehen. Nach einem Neustart geht alles wieder. 

### To Reproduce Bug

Speicher liefert, alles gut. Speicher leer, Dtu hängt sich auf

### Expected Behavior

Dtu sollte sich weiter aktualisieren 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.02.10

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
[DPL] up 1664 s, NO next inverter restart at 0 s (set to -1)
10:42:31.101 > [DPL] targeting 0 W, base load is 175 W, power meter reads 69.0 W (valid)
10:42:31.110 > [DPL] inverter 114183337042 is behind power meter producing 0 W
10:42:31.117 > [DPL] skipping 114183337042 as it is not reachable
10:42:31.128 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 69 W AC
10:42:31.133 > [DPL inverter 114183337042]:
10:42:31.146 >     smart-buffer-powered, standing by at 0 W
10:42:31.149 >     lower/current/upper limit: 10/0/600 W, output capability: 0 W
10:42:31.162 >     sending commands enabled, offline, disqualified (unreachable)
10:42:31.166 >     max reduction production/standby: 0/0 W, max increase: 0 W
10:42:31.177 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
10:42:31.184 >     MPPTs AC power: a: 0 W b: 0 W
10:42:31.199 > Fetch inverter: 112182858716
10:42:31.208 > Queue size - NRF: 4 CMT: 0
10:42:31.372 > RX Period End
10:42:31.384 > All missing
10:42:31.395 > Nothing received, resend whole request
10:42:31.403 > TX AlarmData Channel: 40 --> 15 83 33 70 42 80 12 36 48 80 11 00 67 FE 1B F1 00 00 00 00 00 00 00 00 54 81 4C 
10:42:32.218 > [DPL] up 1666 s, NO next inverter restart at 0 s (set to -1)
10:42:32.247 > [DPL] targeting 0 W, base load is 175 W, power meter reads 69.0 W (valid)
10:42:32.263 > [DPL] inverter 114183337042 is behind power meter producing 0 W
10:42:32.272 > [DPL] skipping 114183337042 as it is not reachable
10:42:32.280 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 69 W AC
10:42:32.288 > [DPL inverter 114183337042]:
10:42:32.297 >     smart-buffer-powered, standing by at 0 W
10:42:32.306 >     lower/current/upper limit: 10/0/600 W, output capability: 0 W
10:42:32.325 >     sending commands enabled, offline, disqualified (unreachable)
10:42:32.344 >     max reduction production/standby: 0/0 W, max increase: 0 W
10:42:32.364 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
10:42:32.390 >     MPPTs AC power: a: 0 W b: 0 W
10:42:32.398 > RX Period End
10:42:32.413 > All missing
10:42:32.423 > Nothing received, resend whole request
10:42:32.447 > TX AlarmData Channel: 61 --> 15 83 33 70 42 80 12 36 48 80 11 00 67 FE 1B F1 00 00 00 00 00 00 00 00 54 81 4C 
10:42:32.741 > [PowerMeterHttpJson] New total: 70.00
10:42:33.037 > RX Period End
10:42:33.159 > All missing
10:42:33.191 > Nothing received, resend whole request
10:42:33.211 > TX AlarmData Channel: 75 --> 15 83 33 70 42 80 12 36 48 80 11 00 67 FE 1B F1 00 00 00 00 00 00 00 00 54 81 4C 
10:42:33.215 > Fetch inverter: 114183337042
10:42:33.234 > Request SystemConfigPara
10:42:33.243 > Queue size - NRF: 4 CMT: 0
10:42:33.260 > [DPL] up 1667 s, NO next inverter restart at 0 s (set to -1)
[... 570 lines cut ...]
10:43:00.282 > [DPL] skipping 114183337042 as it is not reachable
10:43:00.312 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 72 W AC
10:43:00.313 > [DPL inverter 114183337042]:
10:43:00.314 >     smart-buffer-powered, standing by at 0 W
10:43:00.314 >     lower/current/upper limit: 10/0/600 W, output capability: 0 W
10:43:00.347 >     sending commands enabled, offline, disqualified (unreachable)
10:43:00.348 >     max reduction production/standby: 0/0 W, max increase: 0 W
10:43:00.363 >     target limit/output/state: -1 W (unchanged)/0 W/unchanged, 0 update timeouts
10:43:00.364 >     MPPTs AC power: a: 0 W b: 0 W
10:43:00.365 > RX Period End
10:43:00.366 > All missing
10:43:00.367 > Nothing received, resend whole request
10:43:00.368 > TX RealTimeRunData Channel: 23 --> 15 83 33 70 42 80 12 36 48 80 0B 00 67 FE 1C 11 00 00 00 00 00 00 00 00 90 E2 16 
10:43:00.683 > RX Period End
10:43:00.689 > All missing
10:43:00.695 > Nothing received, resend whole request
10:43:00.761 > TX RealTimeRunData Channel: 40 - is 0/0 W
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Solarteur, 2025-04-15:
Schon einmal die aktuellste Test-Version https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14373514259 ausprobiert?

Zwischenzeitlich sind viele solcher Probleme gelöst worden - vielleicht auch deines.

Feuerossi, 2025-04-15:
Moin Ich versuche es mal heute Abend Danke 👋😊

Feuerossi, 2025-04-16:
Nachdem Speicher heute Nacht abgeschaltet hat, lief die Dtu am Morgen stabil weiter. Danke 😊👌

AndreasBoehm, 2025-04-16:
Cool! Dann mach ich hier zu?

Feuerossi, 2025-04-16:
Close 👌

closed 2025-04-16 by AndreasBoehm