hoylabs/OpenDTU-OnBattery #1875: DPL - Spannungsspitzen bei SBB führen zur Fernabschaltung des WR

mikerway, 2025-04-18:
### What happened?

Hi,
betreibe zwei BeCool (Marstek) B2500 v1 Speicher am HM-1500.
Wenn ich diesen als Smart Buffer einstelle und es kommt zu einer großen Spannungsspitze, also z.B. ein Tiefkühlschrank schaltet sich ein, dann wird der WR durch "Fernwartung" abgeschaltet. Das passiert bei mir relativ oft...

[image]

Ist der WR als Solarbetrieben konfiguriert, dann besteht das Problem nicht, nur kann man diesen jetzt nicht mehr in der Nacht betreiben...
Ich nehme an, dass Solarbetriebene sich an das Mindestlimit halten, also dies immer produzieren, auch wenn der berechnete Verbrauch tiefer liegt, während sich beim Smart Buffer der WR bei Erreichen, bzw. unterhalb des Limits, abgeschaltet wird.
Falls ich da in meiner Annahme richtig liege, lässt sich im DPL eine Funktion einbauen, dass der sich der Smart Buffer (auf Knopfdruck) ebenso verhält wie Solarbetriebene oder das der Solarbetriebene (auf Knopfdruck) auch in der Nacht betreiben lässt?

### To Reproduce Bug

Eine kurze Spannungsspitze erzeugen

### Expected Behavior

Keine Abschaltung des WR bei Spannungsspitzen 

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell
09:11:52.778 > [DPL] up 3385 s, it is day, next inverter restart at 60305 s (set to 1)
09:11:52.778 > [DPL] targeting -5 W, base load is 100 W, power meter reads -6.2 W (valid)
09:11:52.778 > [DPL] requesting 581 W from 1 smart-buffer-powered inverter currently producing 582 W (diff -1 W, hysteresis 10 W)
09:11:52.778 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:11:52.778 > [DPL inverter 116183772224]:
09:11:52.778 >     smart-buffer-powered, producing 582 W, output included in power meter reading
09:11:52.778 >     lower/current/upper limit: 80/573/1500 W, output capability: 1500 W
09:11:53.137 >     sending commands enabled, reachable, eligible
09:11:53.137 >     max reduction production/standby: 502/582 W, max increase: 918 W
09:11:53.137 >     target limit/output/state: -1 W (unchanged)/582 W/unchanged, 0 update timeouts
09:11:53.137 >     MPPTs AC power/DC voltage: a: 291 W/45.9 V b: 291 W/45.9 V
09:11:53.137 > Fetch inverter: 116183772224
09:11:53.675 > Queue size - NRF: 1 CMT: 0
09:11:53.675 > [PowerMeters::Json::Http] New total: -9.71
09:11:53.675 > Last missing
09:11:53.868 > Request retransmit: 4
09:11:53.868 > Success
09:11:53.868 > [DPL] up 3386 s, it is day, next inverter restart at 60305 s (set to 1)
09:11:53.868 > [DPL] targeting -5 W, base load is 100 W, power meter reads -9.7 W (valid)
09:11:53.868 > [DPL] requesting 577 W from 1 smart-buffer-powered inverter currently producing 582 W (diff -5 W, hysteresis 10 W)
09:11:53.868 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:11:53.868 > [DPL inverter 116183772224]:
09:11:53.868 >     smart-buffer-powered, producing 582 W, output included in power meter reading
09:11:53.868 >     lower/current/upper limit: 80/573/1500 W, output capability: 1500 W
09:11:54.134 >     sending commands enabled, reachable, eligible
09:11:54.134 >     max reduction production/standby: 502/582 W, max increase: 918 W
09:11:54.134 >     target limit/output/state: -1 W (unchanged)/582 W/unchanged, 0 update timeouts
09:11:54.134 >     MPPTs AC power/DC voltage: a: 291 W/45.9 V b: 291 W/45.9 V
09:11:54.134 > Fetch inverter: 116183772224
09:11:54.673 > Queue size - NRF: 1 CMT: 0
09:11:54.673 > [PowerMeters::Json::Http] New total: 1040.12
09:11:54.673 > Success
09:11:55.000 > [DPL] up 3387 s, it is day, next inverter restart at 60305 s (set to 1)
09:11:55.083 > [DPL] targeting -5 W, base load is 100 W, power meter reads 1040.1 W (valid)
09:11:55.083 > [DPL] requesting 1500 W from 1 smart-buffer-powered inverter currently producing 582 W (diff 918 W, hysteresis 10 W)
[... 547 lines cut ...]
09:38:07.487 > Middle missing
09:38:07.811 > Request retransmit: 2
09:38:07.811 > Middle missing
09:38:07.811 > Request retransmit: 2
09:38:07.811 > Middle missing
09:38:07.811 > Request retransmit: 3
09:38:07.811 > Success
09:38:08.229 > Fetch inverter: 116183772224
09:38:08.229 > Queue size - NRF: 1 CMT: 0
09:38:08.229 > [DPL] up 4960 s, it is day, next inverter restart at 60305 s (set to 1)
09:38:08.243 > [DPL] targeting -5 W, base load is 100 W, power meter reads -4.7 W (valid)
09:38:08.243 > [DPL] requesting 636 W from 1 smart-buffer-powered inverter currently producing 636 W (diff 0 W, hysteresis 10 W)
09:38:08.243 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:38:08.243 > [DPL inverter 116183772224]:
09:38:08.243 >     smart-buffer-powered, producing 636 W, output included in power meter reading
09:38:08.243 >     lower/current/upper limit: 80/628/1500 W, output capability: 1500 W
09:38:08.243 >     sending commands enabled, reachable, eligible
```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-18:
Was meinst du mit Spannungsspitze?

Der WR produziert ca 950 W, PowerMeter meldet -950 W, das bedeutet Verbrauch 0W.

Warum sollte der WR Strom produzieren wenn der Verbrauch bei 0 W liegt?

mikerway, 2025-04-19:
Mit Spannungsspitzen meine ich, wenn ein Gerät kurzfristig einen hohen Anlaufstrom benötigt, also z.B. der Kompressor vom Tiefkühlschrank ne Sekunde 1000 W benötigt, dann aber mit 180 W weiter arbeitet.
Aber vielleicht liegt es gar nicht daran. Habe Mal versucht mit meinen beschränken Kenntnissen den Log zur Abschaltung nachzuvollziehen:

09:37:42.627 > [PowerMeters::Json::Http] New total: 1037.57
09:37:43.076 > Fetch inverter: 116183772224
09:37:43.076 > Queue size - NRF: 1 CMT: 0
09:37:43.076 > [DPL] up 4935 s, it is day, next inverter restart at 60305 s (set to 1)
09:37:43.091 > [DPL] targeting -5 W, base load is 100 W, power meter reads 1037.6 W (valid)
09:37:43.091 > [DPL] requesting 1500 W from 1 smart-buffer-powered inverter currently producing 582 W (diff 918 W, hysteresis 10 W)
09:37:43.091 > [DPL inverter 116183772224]:
09:37:43.091 > expected AC power per MPPT 275 W
09:37:43.091 > [DPL] will cover 1500 W using 1 smart-buffer-powered inverter
09:37:43.091 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:37:43.091 > [DPL inverter 116183772224]:
09:37:43.091 > smart-buffer-powered, producing 582 W, output included in power meter reading
09:37:43.145 > lower/current/upper limit: 80/568/1500 W, output capability: 1500 W
09:37:43.145 > sending commands enabled, reachable, eligible
09:37:43.145 > max reduction production/standby: 502/582 W, max increase: 918 W
09:37:43.145 > target limit/output/state: 1500 W (update)/1500 W/production, 0 update timeouts
09:37:43.145 > MPPTs AC power/DC voltage: a: 291 W/45.8 V b: 291 W/45.8 V
09:37:43.145 > [DPL inverter 116183772224]: sending limit of 100.0 % (1500 W respectively), max output is 1500 W
09:37:43.145 > [DPL] waiting for a start/stop/restart/limit command to complete
09:37:43.145 > Success

Hier meldet der PowerMeter es fehlen 1037 W. Nun wurde dem WR befohlen, 1500 W auszugeben, also die 568 W die schon produziert wurden + 918 W = 1500 W maximal

09:37:43.588 > [PowerMeters::Json::Http] New total: 57.23
09:37:44.679 > Fetch inverter: 116183772224
09:37:44.679 > Queue size - NRF: 2 CMT: 0
09:37:44.679 > [PowerMeters::Json::Http] New total: 46.77
09:37:45.175 > Fetch inverter: 116183772224
09:37:45.175 > Queue size - NRF: 2 CMT: 0
09:37:45.175 > All missing
09:37:45.584 > Nothing received, resend whole request
09:37:45.584 > [PowerMeters::Json::Http] New total: 47.74
09:37:46.624 > Fetch inverter: 116183772224
09:37:46.624 > Queue size - NRF: 2 CMT: 0

Hier ist der WR bereits am Limit.

09:37:46.624 > [PowerMeters::Json::Http] New total: -633.91
09:37:47.180 > Fetch inverter: 116183772224
09:37:47.180 > Queue size - NRF: 2 CMT: 0
09:37:47.180 > Success
09:37:47.577 > [DPL inverter 116183772224]: limit update succeeded, actual limit is 100.0 % (1500 W respectively), effective 4088 ms after update started, requested were 100.0 %
09:37:47.577 > [DPL] waiting for sufficiently recent inverter data
09:37:47.577 > [PowerMeters::Json::Http] New total: -951.37
09:37:48.877 > Fetch inverter: 116183772224
09:37:48.877 > Queue size - NRF: 1 CMT: 0
09:37:48.877 > Success
09:37:48.877 > [DPL] waiting for sufficiently recent power meter reading
09:37:48.877 > [DPL] up 4941 s, it is day, next inverter restart at 60305 s (set to 1)
09:37:48.978 > [PowerMeters::Json::Http] New total: -954.68

Der WR produziert noch 1500 W, der PowerMeter sagt, jetzt ziehe davon 954 ab.

09:37:48.978 > [DPL] targeting -5 W, base load is 100 W, power meter reads -954.7 W (valid)
09:37:48.978 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 581 W (diff -581 W, hysteresis 10 W)

Warum requesting 0 W from 1 smart-buffer-powered und warum curretly producing 581 W, er müsste noch bei 100% sein? Es wurde doch noch kein geändertes Limit an den WR geschickt?!?

09:37:48.978 > [DPL] will cover 0 W using 1 smart-buffer-powered inverter
09:37:48.978 > [DPL] battery allowance is 0/0 W DC/AC, solar power is 0/0 W DC/AC, requested are 0 W AC
09:37:48.978 > [DPL inverter 116183772224]:
09:37:48.978 > smart-buffer-powered, producing 581 W, output included in power meter reading
09:37:48.978 > lower/current/upper limit: 80/1500/1500 W, output capability: 1500 W

Hier steht das current Limit noch auf 1500 W, trotzdem wird das target Limit auf 0 W gesetzt...

09:37:49.833 > sending commands enabled, reachable, eligible
09:37:49.833 > max reduction production/standby: 501/581 W, max increase: 0 W
09:37:49.833 > target limit/output/state: -1 W (unchanged)/0 W/standby, 0 update timeouts
09:37:49.833 > MPPTs AC power/DC voltage: a: 291 W/45.8 V b: 290 W/45.8 V
09:37:49.833 > [DPL inverter 116183772224]: Stopping inverter...

... und der Inverter stoppt... und warum zeigt MPPT je 291 W an, obwohl doch noch gar kein Befehl zur Änderung geschickt wurde?

An den Speichern kann es nicht liegen, denn als Solar-WR funktioniert es einwandfrei, jedenfalls bis die Nacht anbricht.

AndreasBoehm, 2025-04-19:
> 09:37:48.978 > [DPL] targeting -5 W, base load is 100 W, power meter reads -954.7 W (valid)
09:37:48.978 > [DPL] requesting 0 W from 1 smart-buffer-powered inverter currently producing 581 W (diff -581 W, hysteresis 10 W)

Hier liegt das Hauptproblem.

PowerMeter -954 W
WR produziert 581 W

Wo kommen die restlichen 370 W Einspeisung her?

Das was der PowerMeter meldet passt einfach nicht zu dem was der WR meldet, komisch, verstehe ich nicht.

SW-Niko, 2025-04-19:
Hallo @AndreasBoehm ,
kann es sein das wir hier ein zeitliches Problem haben? 🤔 
Die Inverter benötigen eine Zeit bis sie das neue Limit einstellen. Die Powermeter benötigen eine Zeit bis sie einem neuen Messwert liefern können. Und jetzt kommen noch die Smart-Batteries dazu, die eine zusätzliche Regelung mit Zeitverzögerung einbringen.

Wenn wir jetzt zum Beispiel zu schnell die Daten vom Powermeter für die nächste Berechnung verwenden dann kann genau das passieren.  

Wenn das nur manchmal passiert dann fällt das auch nicht auf. Außer die DPL Berechnung ergibt einen negativen oder sehr kleinen Wert, was wiederum die kurzzeitige Abschaltung des Inverters bewirkt. Und das fällt auf.

schlimmchen, 2025-04-19:
Wir warten auf aktuelle Statistiken der Inverter, und da steht drin wie viel die produzieren. Und wir bestehen auf einen PowerMeter Wert, der mind. zwei Sekunden jünger ist als die letzte Statistik.

schneemann78, 2025-04-20:
Hi,
ich habe seit letzter Woche einen Growatt Noah und das auch schon beobachtet. Neueste Firmware und auch die Testversion die ich hier gefunden habe.
Den Wechselrichter hatte ich nun auch auf neuesten Stand, keine Änderung.

Es muss aber noch was anderes dazugekommen, nur Max Leistung ist es nicht.
Ich beobachte bei mir weiter um herauszufinden wann genau es auftritt.
Als letzte Änderung habe ich den Timeout vom Stromzähler auf 500ms reduziert auch da schaue ich ob da was bringt.

Auf jeden Fall danke für euren Einsatz

SW-Niko, 2025-04-20:
Ich habe bis jetzt auch nicht herausgefunden woran es genau liegt. 🙅 

Aber ich kann sagen, das an meinem System die Abschaltung nur selten und kurzzeitig aufgetreten ist.
Vor ein paar Monaten habe ich mir eine Sperre programmiert. Eine Abschaltung wird nur ausgeführt wenn die Inverter Leistung länger als 15 Sekunden unter der minimalen Power-Schwelle liegt. Bis dahin verwende ich einfach die minimalen Power-Schwelle.
Seit dem habe ich keine kurzzeitigen Abschaltungen mehr.

Ok, das ist nur ein Pflaster für das eigentliche Problem und nicht wirklich eine allgemeine Lösung. Aber vielleicht informativ für die weitere Fehlersuche.

tappas197112, 2025-04-23:
Für mich hört sich das an, als ob das mein Problem mit dem Growatt Noah (in Ansätzen) beschreibt. Hatte ich vor einiger Zeit [hier](https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1722) ausführlich beschrieben.

Heute Morgen hatte ich dazu einige Neustarts des HMS-1600-4T, anbei ein Auszug von 06:06 Uhr:

[image]

SW-Niko, 2025-04-29:
>Das klingt interessant, um das dauernde Wechseln zu unterbinden. Wie sind deine Erfahrungen mit dieser Lösung?

Ich habe keine Probleme mehr mit Abschaltungen. Das "Pflaster" ist zu mindestens auf meinem System sehr gut.

Ich sehe das du mit 24V arbeitest. Das kann unter anderen auch zu kurzzeitigen Abschaltungen führen. 
Siehe auch #1601

Thaui57, 2025-06-02:
betreibe wie @mikerway auch zwei Marstek B2500-D v2/3 Speicher an einem HMS-1600-4T. Wenn ich diesen als Smart Buffer einstelle und es kommt zu einer Spannungsspitze  (Waschmaschine oder Ceranfeld) erhalte ich die Fehlermeldung 124 Fernabschaltung. Allerdings ist mir das in den Versionen vor dem Release 2025.04.17 nicht so aufgefallen oder der Fehler hat sich nach geraumer Zeit wieder resetet.  Gibt es schon eine Lösung für dieses Phänomen?

[image]

Squint5471, 2025-06-02:
> Ich habe keine Probleme mehr mit Abschaltungen. Das "Pflaster" ist zu mindestens auf meinem System sehr gut.
>
> Ich sehe das du mit 24V arbeitest. Das kann unter anderen auch zu kurzzeitigen [...]

Könntest du deine Lösung teilen? Ich hab das Problem noch nicht lösen können.

Thaui57, 2025-06-02:
@Squint5471 ich habe keine Lösung. Nur durch Zufall finde ich heraus das der WR off ist und starte ihn händisch neu

Squint5471, 2025-06-02:
> [@Squint5471](https://github.com/Squint5471) ich habe keine Lösung. Nur durch Zufall finde ich heraus das der WR off ist und starte ihn händisch neu

Mein letzter Beitrag ging an @SW-Niko ;)

Ich habe einige "Watchdogs" in home assistant laufen, die diese Probleme des DPL überwachen und mich informieren.

SW-Niko, 2025-06-02:
Hallo @Squint5471

>Könntest du deine Lösung teilen? Ich hab das Problem noch nicht lösen können.

Klar kein Problem. Du must dir dann aber eine eigene FW bauen und einspielen.

SW-Niko, 2025-06-02:
Ok,
Du must in der Datei "PowerLimiter.cpp" nach der Zeile:
`uint16_t inverterTotalPower = calcTargetOutput();`
folgende Zeile einfügen:
`if (inverterTotalPower < config.PowerLimiter.BaseLoadLimit) { inverterTotalPower = config.PowerLimiter.BaseLoadLimit; }`

Im Prinzip wird dadurch verhindert das Inverterleistung unter die BaseLoad fällt.

Gib bitte Bescheid ob dein Problem damit gelöst wird.