hoylabs/OpenDTU-OnBattery #1879: Pipeline Fails on PR

vaterlangen, 2025-04-19:
### What happened?

The newly added auto comments are failing the PR pipeline at https://github.com/hoylabs/OpenDTU-OnBattery/pull/1878

### To Reproduce Bug

Create PR (maybe without maintainer permissions)

### Expected Behavior

Pipeline should not fail if commenting the link is not working

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU-OnBattery?

development

### What firmware variant (PIO Environment)?

generic

### Relevant log/trace output

```Shell
Run actions/github-script@v7
Creating a comment in PR #1878
RequestError [HttpError]: Resource not accessible by integration
    at /home/runner/work/_actions/actions/github-script/v7/dist/index.js:9537:21
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async upsertComment (eval at callAsyncFunction (/home/runner/work/_actions/actions/github-script/v7/dist/index.js:35424:16), <anonymous>:19:5)
    at async eval (eval at callAsyncFunction (/home/runner/work/_actions/actions/github-script/v7/dist/index.js:35424:16), <anonymous>:52:1)
    at async main (/home/runner/work/_actions/actions/github-script/v7/dist/index.js:35522:20) {
  status: 403,
  response: {
    url: 'https://api.github.com/repos/hoylabs/OpenDTU-OnBattery/issues/1878/comments',
    status: 403,
    headers: {
      'access-control-allow-origin': '*',
      'access-control-expose-headers': 'ETag, Link, Location, Retry-After, X-GitHub-OTP, X-RateLimit-Limit, X-RateLimit-Remaining, X-RateLimit-Used, X-RateLimit-Resource, X-RateLimit-Reset, X-OAuth-Scopes, X-Accepted-OAuth-Scopes, X-Poll-Interval, X-GitHub-Media-Type, X-GitHub-SSO, X-GitHub-Request-Id, Deprecation, Sunset',
      'content-encoding': 'gzip',
      'content-security-policy': "default-src 'none'",
      'content-type': 'application/json; charset=utf-8',
      date: 'Sat, 19 Apr 2025 21:11:19 GMT',
      'referrer-policy': 'origin-when-cross-origin, strict-origin-when-cross-origin',
      server: 'github.com',
      'strict-transport-security': 'max-age=31536000; includeSubdomains; preload',
      'transfer-encoding': 'chunked',
      vary: 'Accept-Encoding, Accept, X-Requested-With',
      'x-accepted-github-permissions': 'issues=write; pull_requests=write',
      'x-content-type-options': 'nosniff',
      'x-frame-options': 'deny',
      'x-github-api-version-selected': '2022-11-28',
      'x-github-media-type': 'github.v3; format=json',
      'x-github-request-id': 'ABC2:1B3FE2:AC6C48C:15711DDA:68041177',
      'x-ratelimit-limit': '5000',
      'x-ratelimit-remaining': '4994',
      'x-ratelimit-reset': '1745100098',
      'x-ratelimit-resource': 'core',
      'x-ratelimit-used': '6',
      'x-xss-protection': '0'
    },
    data: {
      message: 'Resource not accessible by integration',
      documentation_url: 'https://docs.github.com/rest/issues/comments#create-an-issue-comment',
      status: '403'
    }
  },
  request: {
    method: 'POST',
    url: 'https://api.github.com/repos/hoylabs/OpenDTU-OnBattery/issues/1878/comments',
    headers: {
      accept: 'application/vnd.github.v3+json',
      'user-agent': 'actions/github-script octokit-core.js/5.0.1 Node.js/20.19.0 (linux; x64)',
      authorization: 'token [REDACTED]',
      'content-type': 'application/json; charset=utf-8'
    },
    body: `{"body":"<!-- bot: build-artifacts -->\\n### Build Artifacts\\n\\nFirmware built from this pull request's code:\\n\\n* [opendtu-onbattery-generic_esp32_8mb.zip](https://nightly.link/hoylabs/OpenDTU-OnBattery/actions/artifacts/2974294500.zip)\\n* [opendtu-onbattery-generic_esp32s3_usb.zip](https://nightly.link/hoylabs/OpenDTU-OnBattery/actions/artifacts/2974294944.zip)\\n* [opendtu-onbattery-generic_esp32_4mb_no_ota.zip](https://nightly.link/hoylabs/OpenDTU-OnBattery/actions/artifacts/2974295255.zip)\\n* [opendtu-onbattery-generic_esp32s3.zip](https://nightly.link/hoylabs/OpenDTU-OnBattery/actions/artifacts/2974295574.zip)\\n\\n### Notice\\n* These artifacts are ZIP files containing the factory update binary as well as the OTA update binary.\\n  Extract the binaries from the ZIP files first. Do not use the ZIP files themselves to perform an update.\\n* These links point to artifacts of the latest [**successful** build run](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14552871415).\\n* The linked artifacts were built from 5556edd62b578b1c24ce180c7ed1d407bdd50826."}`,
    request: {
      agent: [Agent],
      fetch: [Function: proxyFetch],
      hook: [Function: bound bound register]
    }
  }
}
Error: Unhandled error: HttpError: Resource not accessible by integration
```

### Anything else?

https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/14552871415/job/40826245503?pr=1878

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-20:
Thanks for creating this issue.

Very annoying that it does not work when a workflow is triggered from a fork :/

I will look into fixing this.

AndreasBoehm, 2025-04-20:
Links with potential solutions

https://github.com/live-codes/pr-comment-from-artifact

https://github.com/nyurik/auto_pr_comments_from_forks

https://github.com/orgs/community/discussions/51403

AndreasBoehm, 2025-04-21:
This has been fixed by commit https://github.com/hoylabs/OpenDTU-OnBattery/commit/4a9434f2777f28c46a8459bdb35706db2e4953b2