hoylabs/OpenDTU-OnBattery #1880: Stromzähler Einstellungen koennen nicht gespeichert werden ohne vorheriges "Konfiguration testen"

Manos1966, 2025-04-20:
### What happened?

This is NOT a real problem (I am just reporting it in case someone else has had problems).

If I do not test the Power Meter configuration before saving it, the settings are not saved.

This is probably something that always existed and it is not related to the new Firmware version.

SOLUTION: Always test your Power Meter configuration by pressing "Send HTTP(S) request(s) and process response(s)" before saving your Power Meter configuration.
(you should be doing this anyway) 😉 

### To Reproduce Bug

New Firmware installation.
First time setup of Power Meter information
Do not press "Send HTTP(S) request(s) and process response(s)" button.
Try to Save your configuration

### Expected Behavior

Pressing Save should save configuration even if you have not tested it.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32_4mb_no_ota

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-20:
Thank you for creating this issue. I will have a look.

schlimmchen, 2025-04-20:
> New Firmware installation.
First time setup of Power Meter information

This seems to be crucial, as I am unable to reproduce this by changing settings while the PowerMeter was setup before.

Manos1966, 2025-04-20:
> This seems to be crucial, as I am unable to reproduce this by changing settings while the PowerMeter was setup before.

That is what I thought , too
I was just making a new OpenDTU to replace an existing one whose NRF24 I "burned" so, I was duplicating the settings from the old to the new. I did not need to test the Power Meter settings as I knew they are correct. And stumbled upon the problem.

It is a "non-issue" but better you are aware of it.
At the end of the day, NOBODY will set up his Power Meter without pressing the Test button during the first installation.

Next time you do a Factory Firmware installation, test it.

Manos1966, 2025-05-05:
@AndreasBoehm I had no possibility to press the "Save" button and see the 'Setting saved' confirmation.