hoylabs/OpenDTU-OnBattery #1890: "Set limit persistent" isn't persistent

michihuf, 2025-04-21:
### What happened?

If I set a consistent limit for example 50% it resets to 100% on the next day. This happened in all versions I tested before as well. The same in the February built and a development built two weeks ago. I have a HMS-1600-4T.
When I get my Batteries this won't be relevant for me anymore, because with DPL everything works quite well.

### To Reproduce Bug

Turn off DPL.
Set limit to 50%.
Check the limit the next day.
Always 100% again

### Expected Behavior

The limit should be persistent.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU-OnBattery?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU-OnBattery, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

AndreasBoehm, 2025-04-21:
Which firmware version is your inverter running?

AndreasBoehm, 2025-04-21:
I would recommend 01.01.12 but i am not using a solar-powered inverter without DPL which means that i don't ever see the persistent limit being in use.

If this is a necessity for you either need to get a original DTU to adjust the limit or try a older firmware version on your inverter. Sorry.

michihuf, 2025-04-21:
For me this won't be a problem because I will also use DPL in the future. I just fell over it while testing.
I've seen there is also something going on concerning this issue in opendtu bug report section and saw your post. I flashed now your Build Artifact from db5c1b3.
Tomorrow I will see, if the setting stays persistent. First thing i can say:
With non-persistant button and percentage I get "Invalid type specified!"
With persistent/percentage, nonpersistent/Watt and persistent/watt I don't get an error.
I will report tomorrow.

AndreasBoehm, 2025-04-21:
The build that you linked to only changes the non-persistent command.

You need to reload the page or clear your browser cache to fix the error that you got when you tried to set the limit.

michihuf, 2025-04-21:
Thanks, now the error is gone.
If you want me to do some testing for the persistent button I can do that.

justinh998, 2025-04-22:
Wo liegt eigentlich dass aktuelle Problem beim setzten des persistenten Limits? Wissen wir aktuell den Befehl dafür nicht / wurde dass noch nicht mit einer originalen DTU gedumpt?

AndreasBoehm, 2025-04-22:
> Wissen wir aktuell den Befehl dafür nicht

korrekt. Also zumindest kenne ich den Befehl nicht.

michihuf, 2025-04-22:
Also mit dem Build Artifact von db5c1b3 bekomme ich bei relativ/persistant den Fehler "Invalid type specified!". Zumindest absolut/persistant scheint zu gegen. Cache hatte ich geleert 😜 Ich bin auf jeden Fall wieder auf 2025.04.17.
Ich setze also das persistente Limit erst mal mit der originalen DTU auf 800W.
Wenn ich das richtig verstanden habe, kann ich ohne zusätzliche Hardware nicht helfen den Befehl raus zubekommen, stimmt's? Danke für eure spannende Arbeit und Engagement!

AndreasBoehm, 2025-04-22:
> Also mit dem Build Artifact von https://github.com/hoylabs/OpenDTU-OnBattery/commit/db5c1b3fcd871cec43ecfd97e3d9dd0e4444aca3 bekomme ich bei relativ/persistant den Fehler "Invalid type specified!"

Du meintest zwar das du den cache geleert hast aber dieser Fehler kommt definitiv von einem mismatch zwischen Web App und ESP Code. Also bitte nochmal cache leeren und Seite neuladen bis es klappt. Für den DPL ist das allerdings irrelevant, der nutzt trotzdem das neue command.

Ich hab ehrlich gesagt keine Ahnung wie man die Befehle der original DTU sniffen kann.

knotzebom, 2025-04-22:
Könnte man nicht hier in der Diskussion um die neuen Befehle bitten?

https://github.com/tbnobody/OpenDTU/discussions/2153#discussioncomment-11891235

AndreasBoehm, 2025-04-22:
Hab die neuen Commands mittlerweile erfahren

AndreasBoehm, 2025-04-23:
Ich hab alle Commands angepasst, also auch persistent, konnte es aber noch nicht testen.
Hier findet ihr die neue Test-Version: https://github.com/hoylabs/OpenDTU-OnBattery/pull/1571#issuecomment-2817877687

michihuf, 2025-04-25:
Scheint nun persistent mit Percentage setting zu sein. Teste aber noch weiter...

michihuf, 2025-04-27:
Nun sind alle Limits persistent, auch die nonpersistent Limits.

AndreasBoehm, 2025-04-27:
Das ist nicht gut. Geht mal lieber auf die letzte release version zurück.

michihuf, 2025-04-28:
Ich bin jetzt zurück auf den aktuellen Release. 
Als Randinfo: Das Limit das durch die Testversion gesetzt wurde ist offensichtlich persistent gespeichert und wird am nächsten morgen auch korrekt angezeigt und ausgegeben. Der Befehl funktioniert also grundsätzlich.

schlimmchen, 2025-05-03:
Mit dem aktuellen Release startet der WR mit dem zuletzt eingestellten Limit. Und irgendwann (diese 4 Minuten?) ist er dann auf dem über die cloud eingestellten Limit unterwegs. Mehr hab ich noch nicht versucht.

AndreasBoehm, 2025-05-04:
@schlimmchen Mit firmware 1.1.12?
Das 4 Minuten Problem besteht eigentlich nur mit firmware 2.0.4.

schlimmchen, 2025-05-06:
Also ein vom DPL gesetztes Limit wird offensichtlich während des Betriebs gehalten vom WR, auch wenn er länger kein neues non-persistent Limit erhält.

Wenn ich ein bestimmtes persistent (relative, 6.6%) Limit setze und abwarte (gut 10 Minuten), wird dieses Limit gehalten.

Schicke ich dann das Kommando zum Neustarten, stellt sich heraus, dass das Limit nicht so recht persistent ist, denn es wird bei 25% produziert (was ich zuletzt über die Cloud eingestellt hatte). Die Anzeige des Limits bleibt beim alten Wert für ein paar Minuten (genau hab ich jetzt nicht geschaut), dann wird das tatsächlich aktive Limit angezeigt.

Das heißt: Oben hab ich Unsinn geschrieben. Nach dem Setzen eines persistenten Limits und einem Neustart produziert der WR nicht wie das zuletzt angeblich persistent gesetzte Limit es vorschreibst, sondern er produziert nach dem zuletzt aus der Cloud gesetzten Limit. Nach einer Weile korrigiert sich auch die Anzeige des aktuellen Limits in der Live-Ansicht.

Frittenstaebchen, 2025-05-20:
Ich hab das selbe Problem, wenn der Wechselrichter gestartet wird in der Früh, wird das Limit wieder auf 100% gesetzt. Was mir aufgefall ist, wenn er händisch neugestartet wird, gibt er mir in OpenDTU einen Wert von 0% 0W retour und nach einer gewissen Zeit stellt er es wieder auf 100% 1600W, möchte aber permanent mit 800W betreiben

Hoymiles HMS-1600-4T
Hardware-Version: 01.10
Firrmware Version: 1.1.12
DPL supported Yes