hoylabs/OpenDTU-OnBattery #1894: [Request] Zendure - "read  only" option

lpg-elek, 2025-04-22:
### Is your feature request related to a problem? Please describe.

As I'm testing `hoylabs/OpenDTU-OnBattery/pr1878-202504212115 | 4f9e66e` it would be a nice feature if openDTU gets an "Zendure read only" mode, if any home automation software (other, than openDTU) is controling the Zendure device, the actual firmware is overwrinting some settings as minSOC  / maxSOC....  from time to time
e.g. (screenshot from ioBroker / Zendure adapter): 

[image]

### Describe the solution you'd like

It wold be ok, to send out 
`{"properties":["getAll","getInfo"]} `
at the selected update time, but do not control any value if "read only / display only" mode is selected. 

### Describe alternatives you've considered

nothing

AndreasBoehm, 2025-04-22:
Pleas let me know if this is a request that you want to consider @vaterlangen

vaterlangen, 2025-04-22:
> Pleas let me know if this is a request that you want to consider [@vaterlangen](https://github.com/vaterlangen)

As this may also help for users who are facing zero power issues, I already implemented this in a feature branch

closed 2025-05-05 by schlimmchen